ObKo/stm32-cmake #10: Make Error 2 @ Win 8.1

klaus-liebler, 2015-03-03:
Dear ObKo,

thank you for this work to support cmake & stm32. I have tried to use your scripts in win8.1 with gnu arm gcc 4.8.

I adapted:
- in stdperiph/CMakeLists the path STM32F1_StdPeriphLib_DIR
- in cmsis/CMakeLists the path STM32F1_StdPeriphLib_DIR
- in cmsis-3.0/CMakeLists the path STM32F1_StdPeriphLib_DIR
- in gcc_stm32.cmake
*TOOLCHAIN_PREFIX
*SET(TOOLCHAIN_BIN_DIR ${TOOLCHAIN_PREFIX}/${TARGET_TRIPLET}/bin) (directory structure in Windows seems to be different that in Linux)
*CMAKE_FORCE_C_COMPILER(${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-gcc.exe GNU) (added .exe)
*CMAKE_FORCE_CXX_COMPILER(${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-g++.exe GNU) (added .exe)
*SET(CMAKE_ASM_COMPILER ${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-gcc.exe )(added .exe)
*SET(CMAKE_OBJCOPY ${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-objcopy.exe CACHE INTERNAL "objcopy tool")(added .exe)
*SET(CMAKE_OBJDUMP ${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-objdump.exe CACHE INTERNAL "objdump tool")(added .exe)

Futhermore I copied/renamed some executables in the toolchain directory in order to fulfil the expected structure.

Then i issued the following command...and got an error
==========================================
 c:\stm32tc\stm32-cmake-master\cmsis> cmake -DCMAKE_TOOLCHAIN_FILE=../gcc_stm32.c
make -DSTM32_FAMILY=F1 -DCMAKE_INSTALL_PREFIX=C:\stm32tc\gcc48_2014q3\arm-none-e
abi/ -DCMAKE_BUILD_TYPE=Release -G "Unix Makefiles"
-- No TOOLCHAIN_PREFIX specified, using default: C:/stm32tc/gcc48_2014q3
-- No TARGET_TRIPLET specified, using default: arm-none-eabi
-- The ASM compiler identification is GNU
-- Found assembler: C:/stm32tc/gcc48_2014q3/arm-none-eabi/bin/arm-none-eabi-gcc.
exe
-- No STM32F1_StdPeriphLib_DIR specified, using default: C:/stm32tc/STM32F10x_St
dPeriph_Lib_V3.5.0
-- Configuring done
-- Generating done
-- Build files have been written to: C:/stm32tc/stm32-cmake-master/cmsis

c:\stm32tc\stm32-cmake-master\cmsis>make
Scanning dependencies of target cmsis_f1_cl
[ 12%] Building C object CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeri
ph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.
obj
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[2]: *** [CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5
.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj] Fehler
 1
make[1]: *** [CMakeFiles/cmsis_f1_cl.dir/all] Fehler 2
make: *** [all] Fehler 2
==========================================

In fact, the directory C:\stm32tc\stm32-cmake-master\cmsis\CMakeFiles\cmsis_f1_cl.dir\C_\stm32tc\STM32F10x_StdPeriph_Lib_V3.5.0\Libraries\CMSIS\CM3\DeviceSupport\ST\STM32F10x is empty...

What can I do?

Thank you in advance!

Regards,

Klaus

ObKo, 2015-03-03:
> arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory

Try to add ${TOOLCHAIN_PREFIX}/${TARGET_TRIPLET}/bin to PATH environment variable before executing make, i.e
```
set PATH=C:/stm32tc/gcc48_2014q3/arm-none-eabi/bin/;%PATH%
```
or using Control Panel -> System -> Advanced -> Environment Variables

ObKo, 2015-03-03:
oops

klaus-liebler, 2015-03-03:
No, unfortunately this is not the solution. The compiler is definitively in the path as I can start it "manually" from the command line:
...
    c:\stm32tc\stm32-cmake-master\cmsis>arm-none-eabi-gcc.exe
    arm-none-eabi-gcc.exe: fatal error: no input files
    compilation terminated.
...

ObKo, 2015-03-04:
* Try to compile some file manually: ``` arm-none-eabi-gcc.exe <some_real_c_file> ```
* Run make with cmake's verbose flag: ``` make VERBOSE=1 ```, and post log here.

klaus-liebler, 2015-03-04:
```
c:\stm32tc\stm32-cmake-master\stm32-blinky>arm-none-eabi-gcc main.c
main.c: In function 'initGPIO':
main.c:9:28: error: 'RCC_APB1Periph_TIM3' undeclared (first use in this function
)
     RCC_APB1PeriphClockCmd(RCC_APB1Periph_TIM3, ENABLE);
                            ^
main.c:9:28: note: each undeclared identifier is reported only once for each fun
ction it appears in
main.c:9:49: error: 'ENABLE' undeclared (first use in this function)
     RCC_APB1PeriphClockCmd(RCC_APB1Periph_TIM3, ENABLE);
                                                 ^
main.c:11:5: error: unknown type name 'GPIO_InitTypeDef'
     GPIO_InitTypeDef GPIO_Config;
     ^
main.c: In function 'initTimers':
main.c:39:5: error: unknown type name 'TIM_TimeBaseInitTypeDef'
     TIM_TimeBaseInitTypeDef TIM_BaseConfig;
     ^
main.c:40:5: error: unknown type name 'TIM_OCInitTypeDef'
     TIM_OCInitTypeDef TIM_OCConfig;
     ^
main.c:43:19: error: request for member 'TIM_Prescaler' in something not a struc
ture or union
     TIM_BaseConfig.TIM_Prescaler = (uint16_t)(SystemCoreClock / 10000) - 1;
                   ^
main.c:43:37: error: 'uint16_t' undeclared (first use in this function)
     TIM_BaseConfig.TIM_Prescaler = (uint16_t)(SystemCoreClock / 10000) - 1;
                                     ^
main.c:43:47: error: 'SystemCoreClock' undeclared (first use in this function)
     TIM_BaseConfig.TIM_Prescaler = (uint16_t)(SystemCoreClock / 10000) - 1;
                                               ^
main.c:45:19: error: request for member 'TIM_Period' in something not a structur
e or union
     TIM_BaseConfig.TIM_Period = 5000;
                   ^
main.c:46:19: error: request for member 'TIM_ClockDivision' in something not a s
tructure or union
     TIM_BaseConfig.TIM_ClockDivision = 0;
                   ^
main.c:47:19: error: request for member 'TIM_CounterMode' in something not a str
ucture or union
     TIM_BaseConfig.TIM_CounterMode = TIM_CounterMode_Up;
                   ^
main.c:47:38: error: 'TIM_CounterMode_Up' undeclared (first use in this function
)
     TIM_BaseConfig.TIM_CounterMode = TIM_CounterMode_Up;
                                      ^
main.c:49:17: error: request for member 'TIM_OCMode' in something not a structur
e or union
     TIM_OCConfig.TIM_OCMode = TIM_OCMode_PWM1;
                 ^
main.c:49:31: error: 'TIM_OCMode_PWM1' undeclared (first use in this function)
     TIM_OCConfig.TIM_OCMode = TIM_OCMode_PWM1;
                               ^
main.c:50:17: error: request for member 'TIM_OutputState' in something not a str
ucture or union
     TIM_OCConfig.TIM_OutputState = TIM_OutputState_Enable;
                 ^
main.c:50:36: error: 'TIM_OutputState_Enable' undeclared (first use in this func
tion)
     TIM_OCConfig.TIM_OutputState = TIM_OutputState_Enable;
                                    ^
main.c:52:17: error: request for member 'TIM_Pulse' in something not a structure
 or union
     TIM_OCConfig.TIM_Pulse = 2499;
                 ^
main.c:53:17: error: request for member 'TIM_OCPolarity' in something not a stru
cture or union
     TIM_OCConfig.TIM_OCPolarity = TIM_OCPolarity_High;
[... 35 lines cut ...]
aster/cmsis/CMakeFiles 1
[ 12%] Building C object CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeri
ph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.
obj
C:/stm32tc/gcc48_2014q3/arm-none-eabi/bin/arm-none-eabi-gcc.exe  -DSTM32F1 -DSTM
32F10X_CL -mthumb -fno-builtin -mcpu=cortex-m3 -Wall -std=gnu99 -ffunction-secti
ons -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-ma
th -ftree-vectorize -Os -flto -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Librar
ies/CMSIS/CM3/DeviceSupport/ST/STM32F10x -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3
.5.0/Libraries/CMSIS/CM3/CoreSupport -IC:/stm32tc/stm32-cmake-master/cmsis    -o
 CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/
CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj   -c C:/stm32tc/STM3
2F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system
_stm32f10x.c
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[2]: *** [CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5
.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj] Fehler
 1
make[2]: Leaving directory `c:/stm32tc/stm32-cmake-master/cmsis'
make[1]: *** [CMakeFiles/cmsis_f1_cl.dir/all] Fehler 2
make[1]: Leaving directory `c:/stm32tc/stm32-cmake-master/cmsis'
make: *** [all] Fehler 2

c:\stm32tc\stm32-cmake-master\cmsis>
```

ObKo, 2015-03-05:
try to run 
```
C:/stm32tc/gcc48_2014q3/arm-none-eabi/bin/arm-none-eabi-gcc.exe -v -DSTM32F1 -DSTM
32F10X_CL -mthumb -fno-builtin -mcpu=cortex-m3 -Wall -std=gnu99 -ffunction-secti
ons -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-ma
th -ftree-vectorize -Os -flto -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Librar
ies/CMSIS/CM3/DeviceSupport/ST/STM32F10x -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3
.5.0/Libraries/CMSIS/CM3/CoreSupport -IC:/stm32tc/stm32-cmake-master/cmsis    -o
 CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/
CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj   -c C:/stm32tc/STM3
2F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system
_stm32f10x.c
```
manually from command prompt (note, there is ```-v``` flag).

It seems that this isuue isn't related to cmake. Googling shows me, that this problem is quite common for MinGW and other GCC compilers on windows:

http://stackoverflow.com/questions/3848357/createprocess-no-such-file-or-directory

So, I suggest checking PATH variable carefully - remove pathes to any other GNU toolchains (such as cygwin, mingw or other vendor tools), and adding <path_to_toolchain>\libexec\gcc\arm-none-eabi\<gcc_version>\ (try to add all directories inside <path_to_toolchain> that contains *.dll, that should be enough :) )

klaus-liebler, 2015-03-05:
Dear ObKo,

adding "C:\stm32tc\gcc48_2014q3\lib\gcc\arm-none-eabi\4.8.4", to the path, where the one single dll-file "liblto_plugin-0.dll" is located, helped. I got one step further. Now, the compiler complains about a missing "stdint.h"-File.

This file is located in C:/stm32tc/gcc48_2014q3/arm-none-eabi/include, as stated in gcc_stm32.cmake with
```
SET(TOOLCHAIN_PREFIX "C:/stm32tc/gcc48_2014q3")
[...]
SET(TARGET_TRIPLET "arm-none-eabi")
[...]
SET(TOOLCHAIN_INC_DIR ${TOOLCHAIN_PREFIX}/${TARGET_TRIPLET}/include)
```
The "make VERBOSE=1"-output is
```
c:\stm32tc\stm32-cmake-master\cmsis>make VERBOSE=1
C:/stm32tc/CMake/bin/cmake.exe -HC:/stm32tc/stm32-cmake-master/cmsis -BC:/stm32t
c/stm32-cmake-master/cmsis --check-build-system CMakeFiles/Makefile.cmake 0
C:/stm32tc/CMake/bin/cmake.exe -E cmake_progress_start C:/stm32tc/stm32-cmake-ma
ster/cmsis/CMakeFiles C:/stm32tc/stm32-cmake-master/cmsis/CMakeFiles/progress.ma
rks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `c:/stm32tc/stm32-cmake-master/cmsis'
make -f CMakeFiles/cmsis_f1_cl.dir/build.make CMakeFiles/cmsis_f1_cl.dir/depend
make[2]: Entering directory `c:/stm32tc/stm32-cmake-master/cmsis'
C:/stm32tc/CMake/bin/cmake.exe -E cmake_depends "Unix Makefiles" C:/stm32tc/stm3
2-cmake-master/cmsis C:/stm32tc/stm32-cmake-master/cmsis C:/stm32tc/stm32-cmake-
master/cmsis C:/stm32tc/stm32-cmake-master/cmsis C:/stm32tc/stm32-cmake-master/c
msis/CMakeFiles/cmsis_f1_cl.dir/DependInfo.cmake --color=
make[2]: Leaving directory `c:/stm32tc/stm32-cmake-master/cmsis'
make -f CMakeFiles/cmsis_f1_cl.dir/build.make CMakeFiles/cmsis_f1_cl.dir/build
make[2]: Entering directory `c:/stm32tc/stm32-cmake-master/cmsis'
C:/stm32tc/CMake/bin/cmake.exe -E cmake_progress_report C:/stm32tc/stm32-cmake-m
aster/cmsis/CMakeFiles 1
[ 12%] Building C object CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeri
ph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.
obj
C:/stm32tc/gcc48_2014q3/arm-none-eabi/bin/arm-none-eabi-gcc.exe  -DSTM32F1 -DSTM
32F10X_CL -mthumb -fno-builtin -mcpu=cortex-m3 -Wall -std=gnu99 -ffunction-secti
ons -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-ma
th -ftree-vectorize -Os -flto -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Librar
ies/CMSIS/CM3/DeviceSupport/ST/STM32F10x -IC:/stm32tc/STM32F10x_StdPeriph_Lib_V3
.5.0/Libraries/CMSIS/CM3/CoreSupport -IC:/stm32tc/stm32-cmake-master/cmsis    -o
 CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/
CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj   -c C:/stm32tc/STM3
2F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system
_stm32f10x.c
In file included from C:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/
CM3/DeviceSupport/ST/STM32F10x/stm32f10x.h:478:0,
                 from C:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/
CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c:65:
C:/stm32tc/STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/CoreSupport/core_c
m3.h:90:75: fatal error: stdint.h: No such file or directory
 #include <stdint.h>                           /* Include standard types */
                                                                           ^
compilation terminated.
make[2]: *** [CMakeFiles/cmsis_f1_cl.dir/C_/stm32tc/STM32F10x_StdPeriph_Lib_V3.5
.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c.obj] Fehler
 1
make[2]: Leaving directory `c:/stm32tc/stm32-cmake-master/cmsis'
make[1]: *** [CMakeFiles/cmsis_f1_cl.dir/all] Fehler 2
make[1]: Leaving directory `c:/stm32tc/stm32-cmake-master/cmsis'
make: *** [all] Fehler 2

c:\stm32tc\stm32-cmake-master\cmsis>
```

Thank you so much for your help!

ObKo, 2015-03-05:
That is weird, GCC should always use default include pathes (C:/stm32tc/gcc48_2014q3/arm-none-eabi/include). Do you build toolchain by yourself or using some prebuilt toolchain, like CodeSourcery G++ Lite?

Can you show the output of:
```arm-none-eabi-gcc.exe -Wp,-v -E -```
?

klaus-liebler, 2015-03-05:
```
C:\Users\klaus>arm-none-eabi-gcc.exe -Wp,-v -E -
ignoring nonexistent directory "c:\stm32tc\gcc48_2014q3\arm-none-eabi\bin\../lib
/gcc/arm-none-eabi/4.8.4/include"
ignoring nonexistent directory "c:\stm32tc\gcc48_2014q3\arm-none-eabi\bin\../lib
/gcc/arm-none-eabi/4.8.4/include-fixed"
ignoring nonexistent directory "c:\stm32tc\gcc48_2014q3\arm-none-eabi\bin\../lib
/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/include"
ignoring nonexistent directory "c:/stm32tc/gcc48_2014q3/arm-none-eabi/lib/../lib
/gcc/arm-none-eabi/4.8.4/include"
ignoring nonexistent directory "/home/build/work/GCC-4-8-build/install-mingw/arm
-none-eabi/home/build/work/GCC-4-8-build/install-mingw/lib/gcc/arm-none-eabi/4.8
.4/../../../../include"
ignoring nonexistent directory "c:/stm32tc/gcc48_2014q3/arm-none-eabi/lib/../lib
/gcc/arm-none-eabi/4.8.4/include-fixed"
ignoring nonexistent directory "c:/stm32tc/gcc48_2014q3/arm-none-eabi/lib/../lib
/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/include"
ignoring nonexistent directory "/home/build/work/GCC-4-8-build/install-mingw/arm
-none-eabi/usr/include"
#include "..." search starts here:
#include <...> search starts here:
End of search list.
```

ObKo, 2015-03-05:
``` ignoring nonexistent directory "c:\stm32tc\gcc48_2014q3\arm-none-eabi\bin\../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/include" ```
Seems your GCC can't translate such crappy pathes. I don't know why it happens, but this is indeed a toolchain issue. As a quick and dirty hack, you can add ```SET(CMAKE_C_FLAGS "-I${TOOLCHAIN_INC_DIR} ${CMAKE_C_FLAGS}")``` (same for ..._CXX_FLAGS) into gcc_stm32.cmake, or add  ${TOOLCHAIN_INC_DIR} using INCLUDE_DIRECTORIES command in CMakeLists.txt

ObKo, 2015-03-05:
Also you can try to use C_INCLUDE_PATH and CPLUS_INCLUDE_PATH environment vars

klaus-liebler, 2015-03-05:
Hello ObKo,

I installed a fresh new version of the ARM GNU GCC (4.9_2014q4) on my system and everything works fine now. It was just necessary to adjust the top-level paths (Toolchain-Prefix and Path to StdPeriphLibrary)...

Thank you a lot for your help!

closed 2015-03-06 by ObKo