ObKo/stm32-cmake #11: Debug Support in Eclipse CDT

klaus-liebler, 2015-03-07:
Dear ObKo,

thank you again for helping me to get things running. Compilation works fine with makefiles and I can flash the resulting bin-Files to my board using the ST Link Utility. Great!

Next thing I tried is to generate an Eclipse project, import the project in my workspace, compile it using the make target "stm32-blinky.bin" an flash the resulting bin file to my test board using the ST-Link Utility.  Great, this works as well! There is only one drawback:The indexer does not work properly. Many code elements (e.g. all integer variable types) are underlined in red. This makes development in Eclipse a bit hard as I normally  rely on these live code checks...

Up to now, I have used the "GNU ARM Eclipse"-Plugin, which enables convenient compiling, flashing and debugging STM32 code within Eclipse. After having configured cmake to create an Eclipse CDT Project with CMAKE_BUILD_TYPE=DEBUG, I tried to create a new debug configuration for the  stm32-blinky-Debug@stm32-blinky-sources" project. After having started the debugger I got the message 

**Launching stm32-blicnky-Debug_stm32-blinky-sources Configuration has encountered a problem. Error while launching command: gdb --version**

My question is now: Do stm32-cmake projects generally support debugging in Eclipse? If yes, what do I need to do?

Thank you in advance for your answer!

Klaus

ObKo, 2015-03-08:
The eclipse CDT generator of cmake is long-outdated and new versions of CDT can't import it correctly. For source editing I recommend to generate unix makefiles, then import it into eclipse:
[image]
[image]
Then, after import, you'll see that eclipse can't see any imports/symbols and mark it red. To fix this, you'll need to add search pathes and macro definitions manually. Go to Project proprties -> C/C++ General -> Preprocessor include path, Macros, etc -> Providers. There, you need to move 
CDT Cross-GCC Built-in Compiler Settings entry to top of the list, and replace ${COMMAND} macro with absolute path to GCC:
[image]
Next, go to Paths and Symbols entry. There you must add main include dir:
[image]
and some macros:
[image]
(of course, use your MCU defs). This settings can be saved and loaded into other projects.
I don't know why, but eclipse won't find header ```stm32f10x_conf.h``` which is inside project dir and included in stm32f10x.h, so add ```#include "stm32f10x_conf.h``` to main.c 
This is the result:
[image]

ObKo, 2015-03-08:
About debbuging.

I'm using GDB Hardware Debugging Plugin, which can be installed from eclipe repos:

[image]

Next, in Project's debugging configuration, you'll need to add new launch configuration:
[image]
(It should find your binary, but I didn't build project, so it can't find any binaries)
set path to your debugger:
[image]
Remote target settings depends on GDB server you're using. This settings is for OpenOCD.

After that, you can start debugging on target by pressing debug button.

klaus-liebler, 2015-03-10:
Dear ObKo,

thank you again for this great and complete description, that actually helped me to start a debugging session with my STM32F103!!! Unfortunately I was not able to "teach" eclipse where to find all the header files. So, the compiler in the backgroud worked fine, but the indexer could not find almost all symbols. I tried different settings and path definition, but I did not succeed. After some hours I gave up. With all these red underlines everywhere, I cannot code efficiently :-(

As you obviously use a linux based environment I even gave a fresh Ubuntu 14.10 installation a try. There I ended at the step "cmake -DCMAKE_TOOLCHAIN_FILE=../gcc_stm32.cmake -DCMAKE_MODULE_PATH=<path_to_cmake_folder_of_this_project>/Modules -DSTM32_FAMILY=F1 -DCMAKE_INSTALL_PREFIX=<path_to_toolchain>/arm-none-eabi/ -DCMAKE_BUILD_TYPE=Release" with error CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find CMSIS (missing: CMSIS_LIBRARIES CMSIS_INCLUDE_DIR
  CMSIS_LINKER_SCRIPT CMSIS_SOURCES).

I was a quite unsure about the <path_to_toolchain> placeholder. There is the path "/usr/arm-none-eabi" on my system with subdirectories bin, include, lib and share. So, is [...] -DCMAKE_INSTALL_PREFIX=/usr/arm-none-eabi/ [...] correct?

closed 2017-03-21 by ObKo