ObKo/stm32-cmake #20: incorrect chip define for f0

onepixel0, 2016-01-25:
for example in case of STM32_CHIP=stm32f030f4
ram and flash set correctly, but define is STM32F030xC (should be STM32F030x6)

closed 2016-01-26 by ObKo

onepixel0, 2016-01-26:
gcc_stm32f0.cmake
line 9
typo 30x8  ( maybe 030x8 ? )

ObKo, 2016-01-26:
Thanks for notice, fixed in 43b885aa69712afb9b48f6be416a889d0a528cb6

onepixel0, 2016-01-26:
Hey )
in STM32_GET_CHIP_PARAMETERS macro I see some mistakes now
and I believe fragment of code below would more straightforward

IF(${TYPE} STREQUAL 030x6)
            set(RAM "4K")
    ELSEIF(${TYPE} STREQUAL 030x8)
            set(RAM "8K")
    ELSEIF(${TYPE} STREQUAL 030xC)
            set(RAM "32K")
    ELSEIF(${TYPE} STREQUAL 031x6)
            SET(RAM "4K")
    ELSEIF(${TYPE} STREQUAL 038xx)
            SET(RAM "4K")
    ELSEIF(${TYPE} STREQUAL 042x6)
            SET(RAM "6K")
    ELSEIF(${TYPE} STREQUAL 048x6)
            SET(RAM "6K")
    ELSEIF(${TYPE} STREQUAL 051x8)
            SET(RAM "8K")
    ELSEIF(${TYPE} STREQUAL 058xx)
            SET(RAM "8K")
    ELSEIF(${TYPE} STREQUAL 070x6)
            set(RAM "6K")
    ELSEIF(${TYPE} STREQUAL 070xB)
            set(RAM "16K")
    ELSEIF(${TYPE} STREQUAL 071xB)
            SET(RAM "16K")
    ELSEIF(${TYPE} STREQUAL 072xB)
            SET(RAM "16K")
    ELSEIF(${TYPE} STREQUAL 078xx)
            SET(RAM "16K")
    ELSEIF(${TYPE} STREQUAL 091xC)
            SET(RAM "32K")
    ELSEIF(${TYPE} STREQUAL 098xx)
            SET(RAM "32K")
    ENDIF()

ObKo, 2016-01-27:
Fixed in e45f4323d4833881613efa1afded534d040d97bb

commit 0ac996f4 (2016-01-26): Fix #20.