ObKo/stm32-cmake #24: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

DrTon, 2016-02-29:
My plan is to compile ChibiOS 3 using cmake.
I'm trying to run cmake for stm32-chibios and getting following:

```
ton@aircat:/Users/ton/Copter/chibios3test/build% cmake -DSTM32_CHIP=STM32F407IG -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc_stm32.cmake -DCMAKE_MODULE_PATH=../cmake -DCHIBIOS_ROOT=../ChibiOS -DCMAKE_BUILD_TYPE=Debug ../stm32-chibios
-- No TOOLCHAIN_PREFIX specified, using default: /usr/local
-- No TARGET_TRIPLET specified, using default: arm-none-eabi
-- No STM32_FAMILY specified, trying to get it from STM32_CHIP
-- Selected STM32 family: F4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Chibios version:
-- No CHIBIOS_PROCESS_STACK_SIZE specified, using default: 0x400
-- No CHIBIOS_MAIN_STACK_SIZE specified, using default: 0x400
-- Found ChibiOS: CHIBIOS_nil_crt1.c-NOTFOUND;CHIBIOS_nil_vectors.c-NOTFOUND;CHIBIOS_nil_nilcore.c-NOTFOUND;CHIBIOS_nil_nil.c-NOTFOUND;CHIBIOS_nil_crt0_v7m.s-NOTFOUND;CHIBIOS_nil_nilcore_v7m.c-NOTFOUND;CHIBIOS_nil_nilcoreasm_v7m.s-NOTFOUND;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/src/hal.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/src/hal_queues.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/osal/nil/osal.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/STM32/STM32F4xx/hal_lld.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/STM32/LLD/DMAv2/stm32_dma.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/common/ARMCMx/nvic.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/src/pal.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/STM32/LLD/GPIOv2/pal_lld.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/src/serial.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/STM32/LLD/USARTv1/serial_lld.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/src/st.c;/Users/ton/Copter/chibios3test/ChibiOS/os/hal/ports/STM32/LLD/TIMv1/st_lld.c  
-- STM32F407IG has 1024KiB of flash memory and 128KiB of RAM
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CHIBIOS_nil_PLATFORM_cmparams.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
CHIBIOS_nil_PLATFORM_core_cm4.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
CHIBIOS_nil_PLATFORM_stm32f4xx.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
CHIBIOS_nil_nil.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
CHIBIOS_nil_nilcore.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
CHIBIOS_nil_niltypes.h_INCLUDE_DIR
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios
   used as include directory in directory /Users/ton/Copter/chibios3test/stm32-chibios

-- Configuring incomplete, errors occurred!
See also "/Users/ton/Copter/chibios3test/build/CMakeFiles/CMakeOutput.log".
```
What I'm doing wrong?

DrTon, 2016-02-29:
Ooops, just checked again ChibiOS branch, and it was master, not stable_3.0.x. Now it works. :)

closed 2016-02-29 by DrTon

ObKo, 2016-03-01:
ChibiOS had changed version numbering and file layout about 2 month ago. I'm planning to adopt that changes soon.