ObKo/stm32-cmake #25: Allow user to modify compiler flags in CMake Cache

krtkr, 2016-03-03:
Hi everyone!

Any reasons to make all compiler, linker etc. flags CACHE INTERNAL instead of CACHE STRING?

Well, if I would like to build my project using your stm32-cmake library with -mfloat-abi=hard instead of -mfloat-abi=softfp I have to modify gcc_stm32f4.cmake file, which is something unexpected for me. Maybe I did something stupid and missed correct way to modify the flags?

My guess your library should help users to get everything compiling but not forcing them to use only your flags. It would be pretty nice to have possibility to change flags and play around optimization options without modifying build scripts at the stm32-cmake.

Making there flags CACHE STRING instead of INTERNAL should make the day. What do you think?

mgiaco, 2017-09-27:
any news here?

mgiaco, 2017-12-14:
@ObKo Do you know any other way to change the flags?

ObKo, 2020-05-17:
stm32-cmake rework in progress: https://github.com/ObKo/stm32-cmake/pull/117

It will fix that issue.

closed 2020-05-17 by ObKo