ObKo/stm32-cmake #30: cmsis?

mgiaco, 2016-07-07:
Hello, 
I came back to this project today.  As I found out now a lot of stuff has changed. So for a first try I use a ubuntu VM to set up everything. So I think the first thing someone should do is build the packages folder. But as I found out there is no CMSIS download feature within this version. As I remember some time ago the scripts also handle the download of CMSIS. So do you have a good place to download them and where to place them in the file system?`

thx

ObKo, 2016-07-10:
> So I think the first thing someone should do is build the packages folder. But as I found out there is no CMSIS download feature within this version. As I remember some time ago the scripts also [...]
I'm not familiar with those scripts - they were written by @mikepurvis and bit outdated for now. Maybe, I'll find some time to update them...

stm32-cmake is focused on STM32CubeMX packages now, which contains CMSIS inside.

mgiaco, 2016-07-10:
Okay thanks, So the package folder should not be used any more?

thx

closed 2017-01-11 by mgiaco