ObKo/stm32-cmake #35: CMake performs test-compiles with wrong CC

CL-os-cillation, 2017-01-09:
Hi,
I'm trying to compile the "blinky" example project for the STM32F407VG, however running
`cmake -DSTM32_CHIP=STM32F407VG -DCMAKE_TOOLCHAIN_FILE=../stm32-cmake/cmake/gcc_stm32.cmake -DCMAKE_BUILD_TYPE=Debug -G "Eclipse CDT4 - Unix Makefiles" .`
leads to the following errors:
`-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Eclipse version is set to 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.7/Modules/CMakeTestCCompiler.cmake:51 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

  It fails with the following output:

   Change Dir: /home/christian5/Desktop/share/projekt-blinky/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_08b95/fast"

  make: Warning: File 'Makefile' has modification time 1.4 s in the future

  /usr/bin/make -f CMakeFiles/cmTC_08b95.dir/build.make
  CMakeFiles/cmTC_08b95.dir/build

  make[1]: Entering directory
  '/home/christian5/Desktop/share/projekt-blinky/CMakeFiles/CMakeTmp'

  make[1]: Warning: File 'CMakeFiles/cmTC_08b95.dir/flags.make' has
  modification time 1.3 s in the future

  Building C object CMakeFiles/cmTC_08b95.dir/testCCompiler.c.o

  /usr/bin/cc -mthumb -fno-builtin -mcpu=cortex-m4 -mfpu=fpv4-sp-d16
  -mfloat-abi=softfp -Wall -std=gnu99 -ffunction-sections -fdata-sections
  -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-math
  -ftree-vectorize -o CMakeFiles/cmTC_08b95.dir/testCCompiler.c.o -c
  /home/christian5/Desktop/share/projekt-blinky/CMakeFiles/CMakeTmp/testCCompiler.c

  cc: warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead

  cc: error: unrecognized argument in option '-mabi=aapcs'                                                                                  
                                                                                                                                            
  cc: note: valid arguments to '-mabi=' are: ms sysv                                                                                        
                                                                                                                                            
  cc: error: unrecognized command line option '-mthumb'; did you mean                                                                       
  '-mtbm'?                                                                                                                                  
                                                                                                                                            
  cc: error: unrecognized command line option '-mfpu=fpv4-sp-d16'                                                                           
                                                                                                                                            
  cc: error: unrecognized command line option '-mfloat-abi=softfp'                                                                          

  make[1]: *** [CMakeFiles/cmTC_08b95.dir/build.make:66:
  CMakeFiles/cmTC_08b95.dir/testCCompiler.c.o] Error 1

  make[1]: Leaving directory
  '/home/christian5/Desktop/share/projekt-blinky/CMakeFiles/CMakeTmp'

  make: *** [Makefile:126: cmTC_08b95/fast] Error 2

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:1 (PROJECT)

-- Configuring incomplete, errors occurred!`

It appears that CMake is trying to use the flags for the ARM compiler with the standard CC (x86). My CMake version is 3.7.1

mgiaco, 2017-01-11:
Are you using windows or linux? If windows try my latest pull request.

ObKo, 2017-01-12:
> `-DCMAKE_TOOLCHAIN_FILE=../stm32-cmake/cmake/gcc_stm32.cmake`

Please, use absolute file path to toolchain file... 

And add 
`-DCMAKE_MODULE_PATH=<path>`
with path to cmake/ folder of this project (or just copy contents of cmake/ folder to system cmake module path...)

closed 2017-03-17 by ObKo