ObKo/stm32-cmake #39: Linking: HAL functions aren't accessible for internal libs

ildar, 2017-03-17:
I have a project like this:
|--CMakeLists.txt
|--appmain
|  |--main.c
|  `--CMakeLists.txt
`--libapp
   |--libfunc.c
   `--CMakeLists.txt

libfunc.c uses some HAL functions that aren't used from the main.c
During the build libapp.a and main.o are built fine.
But link stage fails, complaining that HAL functions above aren't resolved.

ObKo, 2017-03-17:
Can you give more information? At least, contents of your CMakeLists.txt ...

ildar, 2017-03-17:
[cmk.zip](https://github.com/ObKo/stm32-cmake/files/850387/cmk.zip)

ildar, 2017-03-17:
Linking C executable testmain.elf
../modules/adc/libadc.a(analogconv.c.obj): In function `initADC':
/home/ildar/Projects/0embedded/ucu/modules/adc/analogconv.c:105: undefined
reference to `HAL_ADC_Init'
/home/ildar/Projects/0embedded/ucu/modules/adc/analogconv.c:113: undefined
reference to `HAL_ADCEx_MultiModeConfigChannel'
../modules/adc/libadc.a(analogconv.c.obj): In function `initChannel':
/home/ildar/Projects/0embedded/ucu/modules/adc/analogconv.c:137: undefined
reference to `HAL_ADC_ConfigChannel'
../modules/adc/libadc.a(analogconv.c.obj): In function `deinitADCUCU':
/home/ildar/Projects/0embedded/ucu/modules/adc/analogconv.c:342: undefined
reference to `HAL_ADC_DeInit'
collect2: error: ld returned 1 exit status
make[2]: *** [testmain/testmain.elf] Ошибка 1
make[1]: *** [testmain/CMakeFiles/testmain.elf.dir/all] Ошибка 2
make: *** [all] Ошибка 2

пт, 17 мар. 2017 г. в 16:00, Konstantin Oblaukhov <notifications@github.com

>:

> Can you give more information? At least, contents of your CMakeLists.txt
> ...
-- 
Ildar Mulyukov,
child of God

email: ildar.mulyukov@gmail.com
GoogleTalk: ildar.mulyukov@gmail.com
blog: http://johan-notes.blogspot.com/

ildar, 2017-03-17:
tried `-Wl,--no-as-needed`, didn't help

ObKo, 2017-03-17:
Looks like gcc link order issue. http://stackoverflow.com/questions/45135/why-does-the-order-in-which-libraries-are-linked-sometimes-cause-errors-in-gcc

Can you try build HAL as library:
`add_library(hal STATIC ${CMSIS_SOURCES} ${STM32HAL_SOURCES})`
and link it like this
`target_link_libraries(testmain.elf adc em embUnit hal m c nosys)`
?

ildar, 2017-03-17:
If it helps (I think it should), will there a general solution for such
layouts? Or maybe it's worth to document as a FAQ..

ildar, 2017-03-18:
I am sorry. This isn't the stm32-cmake's bug, that was my bug.
I had HAL_ADC_MODULE_ENABLED disabled in stm32f1xx_hal_conf.h. Hence HAL
part was built without ADC and so link errors.
After the .h file fix, both regular way and libhal.a work fine. Thanks a
lot!

closed 2017-03-18 by ildar