ObKo/stm32-cmake #42: CMakeForceCompiler is required: "The C compiler is not able to compile a simple test program"

renaudcerrato, 2017-04-01:
Since you moved away from `CMAKE_FORCE_XXX_COMPILER` (https://github.com/ObKo/stm32-cmake/commit/1910bc584ebd9258761897ca397d0b550b93921d#diff-ed993e724594a8cbe4fc49c78ad27d97), I'm running into this issue:
 
http://cmake.3232098.n2.nabble.com/Toolchains-and-CMAKE-FORCE-C-COMPILER-in-3-5-td7592757.html

Basically, CMAKE is unable to validate my toolchain because of linkage error. This is somehow correct, as I compile for bare metal and I am providing the _exit function in my code:
 
```
-- No TOOLCHAIN_PREFIX specified, using default: /usr
-- No TARGET_TRIPLET specified, using default: arm-none-eabi
-- No STM32_FAMILY specified, trying to get it from STM32_CHIP
-- Selected STM32 family: F4
-- The C compiler identification is GNU 6.2.1
-- The CXX compiler identification is GNU 6.2.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/arm-none-eabi-gcc" is not able to compile a simple
  test program.

  It fails with the following output:

   Change Dir: /home/renaud/Workspace/test/build/CMakeFiles/CMakeTmp

  Run Build Command:"/usr/bin/make" "cmTC_17f06/fast"

  /usr/bin/make -f CMakeFiles/cmTC_17f06.dir/build.make
  CMakeFiles/cmTC_17f06.dir/build

  make[1]: Entering directory
  '/home/renaud/Workspace/test/build/CMakeFiles/CMakeTmp'

  Building C object CMakeFiles/cmTC_17f06.dir/testCCompiler.c.obj

  /usr/bin/arm-none-eabi-gcc -mthumb -fno-builtin -mcpu=cortex-m4
  -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -Wall -std=gnu99 -ffunction-sections
  -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops
  -ffast-math -ftree-vectorize -o
  CMakeFiles/cmTC_17f06.dir/testCCompiler.c.obj -c
  /home/renaud/Workspace/test/build/CMakeFiles/CMakeTmp/testCCompiler.c

  Linking C executable cmTC_17f06

  /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_17f06.dir/link.txt
  --verbose=1

  /usr/bin/arm-none-eabi-gcc -mthumb -fno-builtin -mcpu=cortex-m4
  -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -Wall -std=gnu99 -ffunction-sections
  -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops
  -ffast-math -ftree-vectorize -Wl,--gc-sections -mthumb -mcpu=cortex-m3
  -mabi=aapcs CMakeFiles/cmTC_17f06.dir/testCCompiler.c.obj -o cmTC_17f06

  /usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o):
  In function `exit':

  exit.c:(.text.exit+0x2c): undefined reference to `_exit'

  collect2: error: ld returned 1 exit status

  CMakeFiles/cmTC_17f06.dir/build.make:97: recipe for target 'cmTC_17f06'
  failed

  make[1]: *** [cmTC_17f06] Error 1

  make[1]: Leaving directory
  '/home/renaud/Workspace/test/build/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_17f06/fast' failed

  make: *** [cmTC_17f06/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:6 (PROJECT)

-- Configuring incomplete, errors occurred!
```

As a quick fix, I had to re-enable CMAKE_FORCE_XXX_COMPILER:

```
diff --git a/cmake/gcc_stm32.cmake b/cmake/gcc_stm32.cmake
index be637e2..27a9f44 100644
--- a/cmake/gcc_stm32.cmake
+++ b/cmake/gcc_stm32.cmake
@@ -1,3 +1,5 @@
+INCLUDE(CMakeForceCompiler)
+
 GET_FILENAME_COMPONENT(STM32_CMAKE_DIR ${CMAKE_CURRENT_LIST_FILE} DIRECTORY)
 SET(CMAKE_MODULE_PATH ${STM32_CMAKE_DIR} ${CMAKE_MODULE_PATH})
 
@@ -44,6 +46,9 @@ ELSE()
     SET(TOOL_EXECUTABLE_SUFFIX "")
 ENDIF()
 
+CMAKE_FORCE_C_COMPILER(${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-gcc${TOOL_EXECUTABLE_SUFFIX} GNU)
+CMAKE_FORCE_CXX_COMPILER(${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-g++${TOOL_EXECUTABLE_SUFFIX} GNU)
+
 SET(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
 SET(CMAKE_C_COMPILER ${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-gcc${TOOL_EXECUTABLE_SUFFIX})
 SET(CMAKE_CXX_COMPILER ${TOOLCHAIN_BIN_DIR}/${TARGET_TRIPLET}-g++${TOOL_EXECUTABLE_SUFFIX})
```

Any comments on this?

renaudcerrato, 2017-04-01:
It seems that CMake 3.6 now support [`CMAKE_TRY_COMPILE_TARGET_TYPE`](https://cmake.org/cmake/help/v3.6/variable/CMAKE_TRY_COMPILE_TARGET_TYPE.html#variable:CMAKE_TRY_COMPILE_TARGET_TYPE) to workaround this very specific issue by setting:

```
SET(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
```

But as of today, the latest Ubuntu [ships with CMake 3.5.x](http://packages.ubuntu.com/yakkety/cmake).

ildar, 2017-04-01:
This is duplicate of https://github.com/ObKo/stm32-cmake/issues/41

ObKo, 2017-04-02:
Fixed in d33a0eaf0a0dcb05685a1927e77eb9e1bf9a34b8

closed 2017-04-02 by ObKo