ObKo/stm32-cmake #47: CMake Error at FindCMSIS.cmake:3 (STM32_GET_CHIP_TYPE)

sledgeh, 2017-04-13:
Hello. I have met a problem:
```
[uuy@pc stm32-blinky]$ cmake -DCMAKE_BUILD_TYPE=Debug -DSTM32_CHIP=STM32F407VG -DCMAKE_TOOLCHAIN_FILE=gcc_stm32.cmake  ./
CMake Error at ...../stm32-cmake/cmake/FindCMSIS.cmake:3 (STM32_GET_CHIP_TYPE):
  Unknown CMake command "STM32_GET_CHIP_TYPE".
```
Please help to resolve.

ildar, 2017-04-14:
You shouldn't copy `gcc_stm32.cmake` into your project's directory. Instead
use it like so:
> cmake -DSTM32_CHIP=STM32F407VG
-DCMAKE_TOOLCHAIN_FILE=/path/to/stm32-cmake/cmake/gcc_stm32.cmake .

sledgeh, 2017-04-14:
@ildar, thank you for respond. I also tried this way with the same result. I updated 1st post.

ildar, 2017-04-14:
There must be something wrong with CMakeLists.txt . Please post it here. Or
you can try mine: https://github.com/ildar/stm32-tests/tree/Blink . The
project must build and blink.

sledgeh, 2017-04-17:
Thank you, @ildar. Looking at your project I understood that I failed to set all mandatory variables.

closed 2017-04-17 by sledgeh