ObKo/stm32-cmake #56: Compile under Windows - undefined reference to `_exit'

sstickel, 2017-06-23:
I recently set up a template project for our future stm32 projects. Under Linux everything works like a charm but under Windows I cannot get it to work correctly. 

When running cmake the compiler check fails with:
```
exit.c:(.text.exit+0x18): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status

```
While testing I compiled a simple hello World using arm-none-eabi-gcc This was successful after adding `--specs=nosys.specs` to the command. Without it threw the same error as cmake. 

I cannot figure out what I have to change in cmake to make this work.  I tried to set CMAKE_EXE_LINKER_FLAGS, even in the toolchain file for F0, did not work.

Not sure if it's a bug or my inability...

Versions used:
Windows 7
CMake 3.8.2
arm gcc 6-2017-q1
CubeFW_F0 1,8.0

sstickel, 2017-07-03:
No reaction? Am I the only one who tried to use this under windows?

sstickel, 2017-07-03:
Okay, I got rid of this error when setting the deprecated 
```
CMAKE_FORCE_C_COMPILER(arm-none-eabi-gcc GNU)
CMAKE_FORCE_CXX_COMPILER(arm-none-eabi-g++ GNU)
SET(STM32_CHIP STM32F030R8)
```
in my cmake file.

This way cmake executes cleanly, but now make fails with a `Makefile:37: *** missing separator.  Stop.`
This is the snippet of the make file, it dies at `!ELSE`
```
!IF "$(OS)" == "Windows_NT"
NULL =
!ELSE
NULL = nul
!ENDIF
SHELL = cmd.exe
```
Is anybody using stm32-cmake successfully under Windows 7?
I have no idea whats happening here.

mgiaco, 2017-07-03:
Hey, Yes I using it under windows So I do not have any problem. Use a build.settings file and call it that way.
 
Look here. https://github.com/liebherrnenzing/stm32f4_canpie-fd

cheers
mathias

sstickel, 2017-07-04:
Hi,
okay, this was a stupid problem. Never thought that setting up a development environment under windows would be such a pain in the ***. It's so easy under linux. The problem was that I used make.exe from MSYS not mingw-make.exe from mingw. Compiling now works, but I still need this in my makefile. 

```
CMAKE_FORCE_C_COMPILER(arm-none-eabi-gcc GNU)
CMAKE_FORCE_CXX_COMPILER(arm-none-eabi-g++ GNU)
SET(STM32_CHIP STM32F030R8)
```

Any help about getting rid of this is appreciated. This is my Cmake File currently working under windows. The Force Compiler part ist the only thing I changed since it was working under linux:
```
CMAKE_MINIMUM_REQUIRED(VERSION 2.8)

SET(STM32Cube_DIR $ENV{STM32_CMAKE_CUBEDIR})
SET(TOOLCHAIN_PREFIX $ENV{STM32_CMAKE_TC_PREFIX})
SET(CMAKE_TOOLCHAIN_FILE $ENV{STM32_CMAKE_TC_FILE})
SET(CMAKE_MODULE_PATH $ENV{STM32_CMAKE_MODULE_PATH})
SET(CMAKE_VERBOSE_MAKEFILE:BOOL ON)

include(CMakeForceCompiler)
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR STM32F030R8)
CMAKE_FORCE_C_COMPILER(arm-none-eabi-gcc GNU)
CMAKE_FORCE_CXX_COMPILER(arm-none-eabi-g++ GNU)

SET(STM32_CHIP STM32F030R8)
PROJECT(stm32-cubemx-cmake)

# Delete some folders and files generated by CubeMX.
# We do not need them when we use cmake because cmake finds
# all these files in ${STM32Cube_DIR}
file(REMOVE_RECURSE "Drivers")
file(REMOVE_RECURSE "Middlewares")
file(REMOVE "Src/system_stm32f0xx.c")

ENABLE_LANGUAGE(ASM)
FIND_PACKAGE(CMSIS REQUIRED)
FIND_PACKAGE(STM32HAL REQUIRED)

INCLUDE_DIRECTORIES(
    ${CMAKE_CURRENT_SOURCE_DIR}
    ${CMSIS_INCLUDE_DIRS}
    ${STM32HAL_INCLUDE_DIR}
    Inc
)

FILE(GLOB PROJECT_SOURCES
    Src/*.c
    Src/*.cpp
)

ADD_EXECUTABLE(${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES} ${CMSIS_SOURCES} ${STM32HAL_SOURCES})
target_link_libraries(${CMAKE_PROJECT_NAME} rdimon)
STM32_SET_TARGET_PROPERTIES(${CMAKE_PROJECT_NAME})
STM32_ADD_HEX_BIN_TARGETS(${CMAKE_PROJECT_NAME})
```

mgiaco, 2017-07-04:
You always need some variables to build on windows. That´s live. :-). And for that a build.setting is the right choice I think.
This is mine - and I do not need any modification by hand.

```cmake
SET(TOOLCHAIN_PREFIX "c:/Program Files (x86)/GNU Tools ARM Embedded/4.9 2015q1" CACHE PATH "toolchain prefix")
SET(CMAKE_MODULE_PATH "e:/github_c/stm32-cmake/cmake" CACHE PATH "cmake module path")
SET(CMAKE_TOOLCHAIN_FILE "e:/github_c/stm32-cmake/cmake/gcc_stm32.cmake" CACHE FILEPATH "toolchain file")
SET(STM32_CHIP "STM32F407VGT6" CACHE STRING "stm32 chip")
SET(STM32Cube_DIR "e:/STM32Cube/Repository/STM32Cube_FW_F4_V1.14.0" CACHE PATH "cube path")
```

sstickel, 2017-07-04:
Okay, I found the problem. 

That `undefined reference to _exit` error was completely misleading. 

One must not use the default Makefile generator under Windows, but use `cmake -G "MinGW Makefiles"`

This way cmake and mingw32-make work as they should and *without* CMakeForceCompiler.

closed 2017-07-04 by sstickel