ObKo/stm32-cmake #57: make C++ optional

ildar, 2017-08-18:
some tollchains may be without C++ compiler. Can stm32-cmake handle such situations?

ccpp, 2020-11-30:
Yes this is an old thread, but as there is no answer I'll put my 5 cents here.

Using C-only projects actually works, you need to use LANGUAGES in [project()](https://cmake.org/cmake/help/latest/command/project.html):
```
project(myproject LANGUAGES C ASM)
```
stm32-cmake will still search for all compilers (C, C++, ASM) but CMake will only pick the relevant ones.

Note though that this is slightly connected to #69 (see my comment there). When a compiler is not found, a host compiler is picked without any warning.

atsju, 2020-12-19:
@ObKo  I think this can be closed as it is really outdated.
As @ccpp explained all examples are given with `project(myproject LANGUAGES C ASM)` which doesn't trigger the search for C++ compiler.

closed 2021-06-21 by atsju