ObKo/stm32-cmake #66: Could NOT find CMSIS (missing: CMSIS_SOURCES)

vveliev, 2017-12-24:
I'm new to cmaka and CL building. So maybe I'm missing something. Thanks!

OS: macOS 10.13.2
cmake: 3.10.0

```
/usr/local/opt/cmake/bin/cmake -DCMAKE_C_COMPILER=/usr/local/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=/usr/local/bin/arm-none-eabi-g++ -DCMAKE_ECLIPSE_VERSION=4.7 -DSTM32_CHIP=STM32F411VE -DTOOLCHAIN_PREFIX=/usr/local/bin/  -DSTM32Cube_DIR=~/STM32Cube -DCMAKE_MODULE_PATH=~/opt/cmake/ -DCMAKE_TOOLCHAIN_FILE=~/opt/cmake/gcc_stm32f4.cmake -DCMAKE_BUILD_TYPE=Debug .
-- The C compiler identification is GNU 6.3.1
-- The CXX compiler identification is GNU 6.3.1
-- Eclipse version is set to 4.7. Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - no
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- STM32F411VE is 411xE device
CMake Error at /usr/local/Cellar/cmake/3.10.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CMSIS (missing: CMSIS_SOURCES)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.10.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /Users/vagif/opt/cmake/FindCMSIS.cmake:152 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:6 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/Users/vagif/workspace/eclipse/cmaketest/CMakeFiles/CMakeOutput.log".
See also "/Users/vagif/workspace/eclipse/cmaketest/CMakeFiles/CMakeError.log".
```

timurey, 2018-02-28:
`FindCMSIS.cmake` need full path to firmware package.
You should change STM32Cube_DIR to ~/STM32Cube/Repository/STM32Cube_FW_F4_Vx.x.x/

vveliev, 2018-03-04:
no luck 
```
cmake -DSTM32_CHIP=411xE -DCMAKE_C_COMPILER=/usr/local/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=/usr/local/bin/arm-none-eabi-g++ -DCMAKE_TOOLCHAIN_FILE=/Users/vagif/workspace/stm32/stm32-cmake/cmake/gcc_stm32f4.cmake -DCMAKE_MODULE_PATH=/Users/vagif/workspace/stm32/stm32-cmake/cmake -DTOOLCHAIN_PREFIX=/usr/local/bin/ -DSTM32Cube_DIR=~/STM32Cube/Repository/STM32Cube_FW_F4_V1.18.0/
-- 411xE is 411xE device
CMake Error at /usr/local/Cellar/cmake/3.10.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CMSIS (missing: CMSIS_SOURCES)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.10.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /Users/vagif/workspace/stm32/stm32-cmake/cmake/FindCMSIS.cmake:152 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:6 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
```

timurey, 2018-03-04:
Did you clear cache by `rm CMakeCache.txt` before building?

`cmake -DSTM32_CHIP=STM32F411xE -DCMAKE_TOOLCHAIN_FILE=~/workspace/libs/stm32-cmake/cmake/gcc_stm32.cmake -DCMAKE_MODULE_PATH=~/workspace/libs/stm32-cmake/cmake -DTOOLCHAIN_PREFIX=~/opt/gcc-arm-none-eabi -DSTM32Cube_DIR=~/STM32Cube/Repository/STM32Cube_FW_F4_V1.18.0/` works fine for me.

satrapes, 2020-02-22:
I have the same issue. Any updates?

closed 2020-05-17 by ObKo