ObKo/stm32-cmake #69: Wonder the correct usage of stm32-cmake

eeyrw, 2018-01-27:
I installed arm-gcc toolchain and add to path under Windows. However I tried to follow the instructions and finally get some weird result.

```
PS F:\Users\yuan\Desktop\stm32-cmake\stm32-blinky> cmake -DSTM32_CHIP=STM32F103C8 -DCMAKE_TOOLCHAIN_FILE=F:/Users/yuan/D
esktop/stm32-cmake/cmake/gcc_stm32f1.cmake -DCMAKE_BUILD_TYPE=Debug .
-- Building for: Visual Studio 15 2017
-- The C compiler identification is MSVC 19.10.25019.0
-- The CXX compiler identification is MSVC 19.10.25019.0
-- Check for working C compiler: E:/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x86/cl.exe
-- Check for working C compiler: E:/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: E:/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x86/cl.exe
-- Check for working CXX compiler: E:/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is MSVC
-- Found assembler: E:/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x86/cl.exe
-- Warning: Did not find file Compiler/MSVC-ASM
CMake Error at CMakeLists.txt:6 (FIND_PACKAGE):
  By not providing "FindCMSIS.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "CMSIS", but
  CMake did not find one.

  Could not find a package configuration file provided by "CMSIS" with any of
  the following names:

    CMSISConfig.cmake
    cmsis-config.cmake

  Add the installation prefix of "CMSIS" to CMAKE_PREFIX_PATH or set
  "CMSIS_DIR" to a directory containing one of the above files.  If "CMSIS"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "F:/Users/yuan/Desktop/stm32-cmake/stm32-blinky/CMakeFiles/CMakeOutput.log".
PS F:\Users\yuan\Desktop\stm32-cmake\stm32-blinky>
```
I am a newbie of cmake and want to know how to get it work.

szechyjs, 2018-01-27:
You need to set `TOOLCHAIN_PREFIX` to the location of your downloaded arm-gcc toolchain.

patrickelectric, 2018-03-12:
I have the same error.
`cmake .. -DTOOLCHAIN_PREFIX=/usr/bin`

closed 2020-05-17 by ObKo

ccpp, 2020-11-30:
Hi, I don't think this should be closed. There is a usability issue here:

If stm32-cmake does not find the right compiler (e.g., /usr/bin/arm-non-eabi-gcc) then CMake will just continue with the default host compiler (/usr/bin/cc, or MSVC as for @eeyrw) **without any error message**:

```
cmake \
        -DCMAKE_TOOLCHAIN_FILE="$BUILDDIR/stm32-cmake/cmake/stm32_gcc.cmake" \
        -DSTM32_TOOLCHAIN_PATH="/usr" \
        -DSTM32_TARGET_TRIPLET="arm-none-eabi" \
        $PROJECT_SOURCE || exit 1
(...)
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
```

I would expect CMake to not find any C compiler instead

Hish15, 2020-11-30:
Hi @ccpp,

Do not hesitate to write a new ticket, referencing this one instead.