ObKo/stm32-cmake #71: Run time error on L476 due to wrong RAM size.

PhSc, 2018-02-09:
On startup, the STM32L476RG (and surely some others would) ends up in hard fault handler on first linked branch.

g-radam, 2018-02-20:
Encountered same issue for my nucleo-STM32L476RG, Rough fix - by replacing:

```
    ELSEIF(${TYPE} STREQUAL "476xx")
        SET(RAM "128K")
```
with 
```
    ELSEIF(${TYPE} STREQUAL "476xx")
        SET(RAM "96K")
```
in file gcc_stm32l4.cmake

In my search, I found that the ChibiOS's .ld for the L476 specifies 96K for general ram, and 32K for hardware parity check ram. Datasheet says "Up to 128 KB of SRAM **_including_** 32 KB _with_ hardware parity check"

ChibiOS's .ld for L476 (for reference)
```
MEMORY
{
    flash : org = 0x08000000, len = 1M
    ram0  : org = 0x20000000, len = 96k
    ram1  : org = 0x00000000, len = 0
    ram2  : org = 0x00000000, len = 0
    ram3  : org = 0x00000000, len = 0
    ram4  : org = 0x10000000, len = 32k
    ram5  : org = 0x00000000, len = 0
    ram6  : org = 0x00000000, len = 0
    ram7  : org = 0x00000000, len = 0
}

/* RAM region to be used for Main stack. This stack accommodates the processing
   of all exceptions and interrupts*/
REGION_ALIAS("MAIN_STACK_RAM", ram0);

/* RAM region to be used for the process stack. This is the stack used by
   the main() function.*/
REGION_ALIAS("PROCESS_STACK_RAM", ram0);

/* RAM region to be used for data segment.*/
REGION_ALIAS("DATA_RAM", ram0);

/* RAM region to be used for BSS segment.*/
REGION_ALIAS("BSS_RAM", ram0);

/* RAM region to be used for the default heap.*/
REGION_ALIAS("HEAP_RAM", ram0);

INCLUDE rules.ld
```

ObKo, 2020-11-01:
Closing due to https://github.com/ObKo/stm32-cmake/pull/117 being merged and made this issue outdated.

closed 2020-11-01 by ObKo