ObKo/stm32-cmake #73: Could NOT find STM32HAL (missing: STM32HAL_SOURCES)

CreFroD, 2018-03-01:
I use STM32Cube F4 version 1.19.0 on Mac OS. There are 4 "NOT_FOUND" files at the end of STM32HAL_SOURCES: 
```
...
HAL_stm32f4xx_hal_fmc_c_FILE-NOTFOUND
HAL_stm32f4xx_hal_fsmc_c_FILE-NOTFOUND
HAL_stm32f4xx_hal_sdmmc_c_FILE-NOTFOUND
HAL_stm32f4xx_hal_usb_c_FILE-NOTFOUND
```
That is why `FIND_PACKAGE_HANDLE_STANDARD_ARGS` at the end of `FindSTM32HAL.cmake` throws the error.

KKoovalsky, 2018-03-05:
For a temporary workaround use:  
`FIND_PACKAGE(STM32HAL COMPONENTS tim gpio ${AND_OTHER_COMPONENTS_YOU_NEED} REQUIRED)`

pymaximus, 2018-07-26:
I am getting this error now on Fedora 28 with latest F4 libs. It is complaining about 

`/home/frank/mnt/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sdram.cHAL_stm32f4xx_hal_fmc_c_FILE-NOTFOUND.`

Indeed, the  file stm32f4xx_hal_fmc.c does not exist, although stm32f4xx_ll_fmc.c does exist. Any guidance on what the issue is, and how to fix it ? I tried the ${fmc} workaround shown above. That allows cmake to run to completion, but  linking fails later. I can see that hal_fmc does not get compiled at all.

The strange thing is that this same project compiles without issue on MacOSX, and I can see that
the ll_ fmc gets found/compiled without issue.

`[ 32%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c.obj
[ 34%] Building ASM object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f429xx.s.obj
[ 37%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c.obj
[ 39%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
[ 41%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.c.obj
[ 44%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr_ex.c.obj
[ 46%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj
[ 48%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc_ex.c.obj
[ 51%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c.obj
[ 53%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim.c.obj
[ 55%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim_ex.c.obj
[ 58%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc.c.obj
[ 60%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc_ex.c.obj
[ 62%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sdram.c.obj
[ 65%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_fmc.c.obj
[ 67%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_spi.c.obj
[ 69%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_ltdc.c.obj
[ 72%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c.obj
[ 74%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma_ex.c.obj
[ 76%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma2d.c.obj
[ 79%] Building C object CMakeFiles/stm32f4_example_project.dir/Users/frank/STM32Cube/Repository/STM32Cube_FW_F4_V1.21.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rng.c.obj`

pymaximus, 2018-07-27:
Created pull request #81 to address fmc missing for F4 target.

closed 2020-05-17 by ObKo