ObKo/stm32-cmake #74: Unknown CMake command "STM32_GET_CHIP_TYPE"

kisielk, 2018-03-08:
Just trying to compile one of the examples, I get this:

```
$ cmake . -DSTM32_CHIP=STM32F407IG -DCMAKE_MODULE_PATH=../cmake
CMake Error at /Users/kamil/intellijel/eurorack/ARM/stm32-cmake/cmake/FindCMSIS.cmake:3 (STM32_GET_CHIP_TYPE):
  Unknown CMake command "STM32_GET_CHIP_TYPE".
Call Stack (most recent call first):
  CMakeLists.txt:6 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/Users/kamil/intellijel/eurorack/ARM/stm32-cmake/stm32-blinky/CMakeFiles/CMakeOutput.log".
```

KKoovalsky, 2018-03-12:
You have forgotten to specify -DCMAKE_TOOLCHAIN_FILE

kisielk, 2018-03-12:
```
 cmake . -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc_stm32.cmake -DCMAKE_MODULE_PATH=../cmake -DSTM32_CHIP=STM32F407IG
CMake Error at /Users/kamil/intellijel/eurorack/ARM/stm32-cmake/cmake/FindCMSIS.cmake:3 (STM32_GET_CHIP_TYPE):
  Unknown CMake command "STM32_GET_CHIP_TYPE".
Call Stack (most recent call first):
  CMakeLists.txt:6 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/Users/kamil/intellijel/eurorack/ARM/stm32-cmake/stm32-blinky/CMakeFiles/CMakeOutput.log".
```

KKoovalsky, 2018-03-15:
Whats Your directory tree? Maybe there is a mistake in setting relative directory? I can see that You're invoking `cmake .` from current directory and setting `CMAKE_MODULE_PATH` to `../cmake`. Try to use absolute paths.

kisielk, 2018-03-15:
You can see the directory tree in the error messages. I'm executing cmake from within the `stm32-blinky` subdirectory, so the relative path of `../cmake` should work. 

For the actual project I'm going to be using this with I need to use relative paths, so I need to get it to work using this method.

KKoovalsky, 2018-03-18:
Please, try it with absolute path first to check if the relative path is correct.

> the relative path of ../cmake should work

so You're not really sure about that.

I can't see Your directory tree from logs.

kisielk, 2018-03-20:
Ok I'm not sure what was wrong, but I deleted the repository and checked it out again, now when I copy and paste the code from my original comment it works (after also adding `-DTOOLCHAIN_PREFIX` and `-DSTM32Cube_DIR`)

closed 2018-03-20 by kisielk