ObKo/stm32-cmake #75: cmake error with GNU Tools ARM Embedded\7 2017-q4-major

gbmont, 2018-04-27:
Hi, 
I do have some problems with gcc 7.21.

```

-- No TARGET_TRIPLET specified, using default: arm-none-eabi
-- No STM32_FAMILY specified, trying to get it from STM32_CHIP
-- Selected STM32 family: F4
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: c:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: c:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe -- broken
CMake Error at C:/Program Files/CMake/share/cmake-3.11/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "c:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: E:/LWN/github_c/stm32f4_canpie-fd/Debug/CMakeFiles/CMakeTmp

    Run Build Command:"C:/PROGRA~2/GNUARM~1/BUILDT~1/2DBD0~1.8-2/bin/make.exe" "cmTC_c6f86/fast"
    C:/PROGRA~2/GNUARM~1/BUILDT~1/2DBD0~1.8-2/bin/make.exe -f CMakeFiles/cmTC_c6f86.dir/build.make CMakeFiles/cmTC_c6f86.dir/build
    make.exe[1]: Entering directory 'E:/LWN/github_c/stm32f4_canpie-fd/Debug/CMakeFiles/CMakeTmp'
    c:/program files (x86)/gnu arm eclipse/build tools/2.8-201611221915/bin/sh: /C/Program Files/CMake/bin/cmake.exe: not found
    CMakeFiles/cmTC_c6f86.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_c6f86.dir/testCCompiler.c.obj' failed
    make.exe[1]: *** [CMakeFiles/cmTC_c6f86.dir/testCCompiler.c.obj] Error 127
    make.exe[1]: Leaving directory 'E:/LWN/github_c/stm32f4_canpie-fd/Debug/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_c6f86/fast' failed
    make.exe: *** [cmTC_c6f86/fast] Error 2
```

I do not find out whats the problem.

Many thanks
cheers
geri

TurhanOz, 2018-07-05:
Hi @gbmont, have you found a solution ?
Cheers

jayaramanrp, 2019-06-12:
anyone got solution? facing same problem

g-radam, 2019-06-12:
This issue looks like nothing to do with this project. The cmake error is saying your compiler is failing it's self-diagnostics. (it builds a small test application and tries to run it to see if the compiler works).

I would search around on google to find out how to fix your compiler on windows.

I use Linux, so these issues aren't present.

closed 2020-05-17 by ObKo

Dragon-66, 2022-03-06:
I have the same problem

atsju, 2022-03-06:
@Dragon-66 

This issue is 4Y old and closed for 2Y. Conclusion was that this is not linket to stm32-cmake repository.

If you experience an issue and believe stm32-cmake repo should fix it then consider opening a new issue and provide all relevant information.

Dragon-66, 2022-03-06:
> @Dragon-66
> 
> This issue is 4Y old and closed for 2Y. Conclusion was that this is not linket to stm32-cmake repository.
> 
> If you experience an issue and believe stm32-cmake repo should fix it then consider opening a new issue and provide all relevant information.

I don't quite understand “that this is not linket to stm32-cmake repository.”

atsju, 2022-03-06:
> I don't quite understand “that this is not linket to stm32-cmake repository.”

I only quoted last comment. 

> This issue looks like nothing to do with this project. 

Again, this issue is closed and has been for a long time.
Please open a new issue and provide all relevant information. You might link to this issue in your futur issue if you think it's relevant. 
I repeat **we need a fresh issue to be opened**. 
Indeed, since 4Y ago, the code has changed a lot and you will see a bug template requesting all information needed to help us solve the issue. We can not solve issues with just one log and no answers to additional questions. So kindly ensure you provide as much information as possible/needed and answer any additional answer we might have. Failing to do so will likely make it impossible for us to help.