ObKo/stm32-cmake #82: Is modern cmake would be possible in embedded world ?

davidtazy, 2018-08-23:
Hi,
cmake >3.0 invite us to see packages  as "TARGET"

1. would it be possible to have CMSIS, HAL... as target and be able to write:
```
...
add_executable(MyProject main.c)
target_link_libraries(MyProject PRIVATE Stm32::CMSIS) 
```
i think it would be more "readable"

2. as a general question, why embedded world encourage us to **not** create libraries but build all files ?

excuse my english...

free-easy, 2018-12-12:
1. It's possible to use the target based approach. See my pull request.
It's quite trivial, I just define some IMPORTED target at the end of the Find module, where I gather all the variables.

2. I don't think there's any reason for the embedded world to encourage this.
Actually IMHO the embedded world is quite weird. 
Each manufacturer build their custom Eclipse-based IDE, the library they supply is often only accessible through their website, when now days it would be much useful to have those libraries (which are given with the source code) on platforms like GitHub, with another benefit that issue tracking is much easier compared to their ugly forum.
Therefore, I think the problem is more that the embedded world (in many cases) just ignores tools like CMake.

ObKo, 2020-05-12:
If you're still interested, please take a look at modern-cmake branch: https://github.com/ObKo/stm32-cmake/tree/modern-cmake

I've implemented support for CMSIS and HAL using imported targets and modern cmake features, such as target_sources and INTERFACE scope.

Looks pretty good for me, but I need more opinions and suggestions.

Will make that master after implementing some missing features.

davidtazy, 2020-05-12:
Still interested!
Thx you for that.  I will look at it!

closed 2020-11-01 by ObKo