ObKo/stm32-cmake #90: Add support for STM32H7 family

Hish15, 2019-03-22:
Check out [this monster](https://www.youtube.com/watch?v=LAx2iqlu9ts), we have to have it.

The STM32F7 is already handled so I think there is not much work to do.
If I'm not wrong it's just the cmake/gcc_stm32H7.cmake to write right?

liarokapisv, 2019-09-05:
Hello @Hish15 , I am also interested in support for this family. Did you ever get to that PR?

Hish15, 2019-09-05:
Hi,
I did work here :
https://github.com/Hish15/stm32-cmake/tree/HandlingH7Family  but no PR.I
droped the idea to use it and made it work on my own from scratch.
I wanted to learn :)
it took a lot of time but it's not hard at all!

I wiil try to make the PR actually :)

Greetings

Hish15, 2020-05-18:
closed by #104

closed 2020-05-18 by Hish15