ObKo/stm32-cmake #92: [Question] How to exclude STM32Cube from stm32-cmake?

ivakyb, 2019-04-01:
I don't use STM32Cube in my project, I use older stm32 standard peripheral driver. Some functions has conflicts. 

### How to disable STM32Cube?

pan-mroku, 2019-04-02:
Do you mean that you want to use SPL instead of HAL/LL? In that case I think you need to convert/rewrite cmake `Find*.cmake` files to match SPL. [This document](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/5e/65/b0/94/13/e4/42/ca/DM00405316/files/DM00405316.pdf/jcr:content/translations/en.DM00405316.pdf) shows how to migrate code from SPL to LL so you need to look what the differences are between those libraries.

I would gamble that you mostly need to change file search paths but that's just my guess.

ivakyb, 2019-04-02:
Yes.
Thank you.

ucayalifish, 2019-04-02:
I use stm32-cmake in a big "legacy" project. Firstyou need is to set a few variables. Here is an example:
`SET(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "xxxxx")
SET(CMSIS_DIR "${CMAKE_CURRENT_SOURCE_DIR}/xxxxx")
SET(STM32_LINKER_SCRIPT ${CMAKE_CURRENT_SOURCE_DIR}/xxxxx/combined.ld)
SET(STM32_CHIP "STM32F437ZI")
SET(TOOLCHAIN_PREFIX "xxx/gcc-arm-none-eabi-7")
SET(CMAKE_TOOLCHAIN_FILE "xxxx/cmake/gcc_stm32.cmake")
SET(STM32_SPL_DIR "xxxx/STM32F4xx_STANDARD_PERIPHERAL_DRIVERS/STM32F4xx_StdPeriph_Driver")
`
This is just an example.
Then include the sources that you need like that:
`set(CMSIS_SOURCES
        ${CMSIS_DIR}/Device/ST/STM32F4xx/Source/Templates/gcc_ride7/startup_stm32f427_437xx.s
        )

file(GLOB TEST_SOURCES
        ${CMAKE_CURRENT_SOURCE_DIR}/test/*.c
        )

file(GLOB READER_SOURCES
        ${CMAKE_CURRENT_SOURCE_DIR}/EXTI_reader/*.c
        )

file(GLOB STM32_SPL_SOURCES
        ${STM32_SPL_DIR}/src/*.c
        )
`
Set include path, definitions and options. Add you project files and finalize like that:
`ADD_EXECUTABLE(${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES})

STM32_SET_TARGET_PROPERTIES(${CMAKE_PROJECT_NAME})
STM32_ADD_HEX_BIN_TARGETS(${CMAKE_PROJECT_NAME})
STM32_PRINT_SIZE_OF_TARGETS(${CMAKE_PROJECT_NAME})
`
Definetely not the best possible way but it works. stm32-cmake provides reasonable and adapted to specific MCU toolchain files and custom targets.

ivakyb, 2019-04-05:
@ucayalifish , Thank you for response. 

You may want to edit you answer as follows:

I use stm32-cmake in a big "legacy" project. First you need is to set a few variables. Here is an example:
```
SET(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "xxxxx") 
SET(CMSIS_DIR "${CMAKE_CURRENT_SOURCE_DIR}/xxxxx") 
SET(STM32_LINKER_SCRIPT ${CMAKE_CURRENT_SOURCE_DIR}/xxxxx/combined.ld)
SET(STM32_CHIP "STM32F437ZI") 
SET(TOOLCHAIN_PREFIX "xxx/gcc-arm-none-eabi-7") 
SET(CMAKE_TOOLCHAIN_FILE "xxxx/cmake/gcc_stm32.cmake") 
SET(STM32_SPL_DIR "xxxx/STM32F4xx_STANDARD_PERIPHERAL_DRIVERS/STM32F4xx_StdPeriph_Driver")
```

This is just an example.
Then include the sources that you need like that:
```
set(CMSIS_SOURCES
${CMSIS_DIR}/Device/ST/STM32F4xx/Source/Templates/gcc_ride7/startup_stm32f427_437xx.s
)

file(GLOB TEST_SOURCES
${CMAKE_CURRENT_SOURCE_DIR}/test/*.c
)

file(GLOB READER_SOURCES
${CMAKE_CURRENT_SOURCE_DIR}/EXTI_reader/*.c
)

file(GLOB STM32_SPL_SOURCES
${STM32_SPL_DIR}/src/*.c
)
```

Set include path, definitions and options. Add you project files and finalize like that:
```
ADD_EXECUTABLE(${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES})
```
```
STM32_SET_TARGET_PROPERTIES(${CMAKE_PROJECT_NAME})
STM32_ADD_HEX_BIN_TARGETS(${CMAKE_PROJECT_NAME})
STM32_PRINT_SIZE_OF_TARGETS(${CMAKE_PROJECT_NAME})
```
Definetely not the best possible way but it works. stm32-cmake provides reasonable and adapted to specific MCU toolchain files and custom targets.

ivakyb, 2019-04-06:
SET(CMAKE_TOOLCHAIN_FILE "xxxx/cmake/gcc_stm32.cmake") 

Must appear before `project()`

closed 2019-04-06 by ivakyb