ObKo/stm32-cmake #95: building for stm32f071x8

W-M-D, 2019-06-27:
hi i can't get any code to run with the stm32f071x8 any one else had issues ?

g-radam, 2019-06-27:
Is this a problem with your code, or the build platform? Have you had success running any other code using this build platform?

W-M-D, 2019-06-27:
No :/ I have tried both arm-none-eabi from the Debian repos and gcc downloaded from linearo. I am able to upload hex files and debug. But it seems like the clock isn't initializing correctly and it ends in a hard fault ( in the bootloader or something) with a corrupt stack. I am able to toggle some GPIOS in gdb but not others. I'm using a 12Mhz crystal on this board and have configured the variables in hal config. Any tips for debugging?

W-M-D, 2019-06-28:
Ok I have been able to output the clock and run the device using gdb and load. I'm using the internal OSC But I can't get it to boot on its own. I'm not too sure why as the boot loader is set to load to system memory...  The interrupts aren't incrementing the timers while running with gdb. This most likely isn't an issue with the toolchain. I apologize.

W-M-D, 2019-06-29:
ok it turns out i'm just disabled. I was able to get it working. I will give myself 1000 lashes for creating an issue without confirming it wasn't my fault

g-radam, 2019-06-29:
Haha well glad to hear you got it working. The cmake build platform generates the code relatively the same across all stm chips (and ive tested many), so my suspicion was that it had to be your code.

closed 2020-11-01 by ObKo