ObKo/stm32-cmake #111: Flashed program doesn't run on STM32F401RE

disposedtrolley, 2020-03-16:
Hello!

I'm trying to build the `stm32-blinky` project to run on my STM32F401RE Nucleo board. I've successfully generated a CMake project and can build and debug the program, but the program doesn't run if I flash it and restart the board (the LED stays on and nothing happens).

My CMake flags:

```
-DSTM32_CHIP=STM32F401RE
-DCMAKE_MODULE_PATH=../../cmake
-DCMAKE_TOOLCHAIN_FILE=../../cmake/gcc_stm32.cmake
-DTOOLCHAIN_PREFIX=/usr/local/opt/gcc-arm-none-eabi-9-2019-q4-major -DSTM32Cube_DIR=/usr/local/opt/STM32Cube_FW_F4_V1.25.0
-DCMAKE_BUILD_TYPE=Debug
```

Hish15, 2020-05-18:
What do you mean by "can build **and debug**"?

disposedtrolley, 2020-05-18:
I mean that the project builds successfully, and the code will execute on the MCU while it’s being debugged (it’s a simple blinky project, so I can set a breakpoint on main and step into the loop and observe the LED blinking). If I unplug it from my computer, connect it to power, and reset it, nothing is executed.

Hish15, 2020-05-19:
Ok, that's weird for sure...
Do you have the expected blinking behavior while debuging? (1Hz blinking).

On nucleo board you have a jumper to select your power source. I suspect that you are not powering the board through the ST-LINK, so the processor is not even powered... To check that you can try to plug the board to a usb charger using the same usb port you use for debug.

disposedtrolley, 2020-05-23:
> Do you have the expected blinking behavior while debuging? (1Hz blinking).

Yep it's all working while debugging. I can step through the infinite loop in `main` and toggle the LED too.

> I suspect that you are not powering the board through the ST-LINK, so the processor is not even powered...

Nah it's definitely powered! The board works fine as-is when loading in a STM32Cube project. I think it _may_ have something to do with the memory map and how the linker is managing that.

ObKo, 2020-11-01:
Closing due to https://github.com/ObKo/stm32-cmake/pull/117 being merged and made this issue outdated.

closed 2020-11-01 by ObKo