ObKo/stm32-cmake #118: Question- Using CubeMX-Generated BSP?

MBetters, 2020-05-19:
I'm using STM32CubeMX to generate my STM32H743BIT6 BSP. Is there a way to use BSP instead of pointing `-DSTM32_CUBE_H7_PATH=...` to the generic STM32CubeH7?

Hish15, 2020-05-19:
Actually the BSP is searched in the folder ${STM32Cube_DIR}.

With STM32CubeMX you can copy the files you need to your folder location, but you can also just point to your "standard" path to you STM32Cube_FW_XXX folder (it is a generation option) 
If you do ask it to generate in the project folder, try setting ${STM32Cube_DIR} accordingly :)

MBetters, 2020-05-19:
@Hish15 Sorry, I should've clarified. I'm using the `STM32_CUBE_${FAMILY}_PATH` variable in the `modern-cmake` branch. However, one thing I've noticed is that when I change `STM32_CUBE_${FAMILY}_PATH` to point to my generated BSP directory, I get the following:

```
CMake Error at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find CMSIS (missing: CMSIS_SOURCES STM32H743ZI) (found version
  "0.0.0")
```

Hish15, 2020-05-20:
`modern-cmake` is WIP. It does not handle the BSP at all.
Why do you want to use the generated BSP? isn't it just a copy of the content of you CubeDir?
 
Using the branch `master` BSP files will be searched here :
```
HINTS ${STM32Cube_DIR}/Drivers/BSP/${STM_BOARD} ${STM32Cube_DIR}/Drivers/BSP/Components
```

MBetters, 2020-05-20:
I'm using the generated BSP because my team has made custom pin assignments and patches to the BSP code. I was able to get the generated BSP working with the `modern-cmake` branch with a couple minor tweaks to some `find_path(...)` search paths. I might do a PR to handle the case where users of this cmake repo wish to optionally point to a generated BSP instead of a full CubeDir. Do you think there's a better approach?

Hish15, 2020-05-20:
That's a great news. 
Since you did the work, I think you should make the PR, we can discuss the matter in there.

atsju, 2021-07-07:
Hi @MBetters, 
The modern-cmake branch has been merged last year and the repo is now (since few weeks) actively maintained.

> I might do a PR to handle the case where users of this cmake repo wish to optionally point to a generated BSP instead of a full CubeDir

This makes sense. Please do not hesitate to do an update on this issue and of course we would be more than happy to integrate some changes if you do a PR.