ObKo/stm32-cmake #134: STM32F401 / F411: it doesn't have a CCRAM, but CCRAM is set to 64K

t123yh, 2020-06-30:
From the reference manual these models doesn't have a built-in CCRAM, but it's set to 64K. Probably this should be fixed?

Hish15, 2020-06-30:
You are speaking of CCMRAM I suppose?
https://github.com/ObKo/stm32-cmake/blob/5638bf9141d9d670fa7d9a2a04f07fa26b74dc7a/cmake/gcc_stm32f4.cmake#L70
This is actually CortexM4 related. 
Why do you think that STM32F401 / F411 does not have it ?

t123yh, 2020-06-30:
Because you don't find them in datasheet.

In STM32F405 datasheet you can see the CCM data RAM is at 0x10000000,
[image]

But in STM32F401/F411 it's neither found in memory map nor mentioned in the datasheet at all.
[image]

Hish15, 2020-07-03:
👍 This is confirmed [here](https://community.st.com/s/question/0D53W00000BqROfSAN/do-stm32f401-and-f411-have-a-ccram-)

ObKo, 2020-11-01:
Closing due to https://github.com/ObKo/stm32-cmake/pull/117 being merged and made this issue outdated.

closed 2020-11-01 by ObKo