ObKo/stm32-cmake #135: Redefinition of linker file sections while using stm32_add_linker_script

barafael, 2020-07-06:
I am on the modern_cmake branch. My target is the stm32g474.
I changed the example from here:
https://github.com/ObKo/stm32-cmake/tree/modern-cmake/examples/custom-linker-script
to use stm32g4 with my custom script. That works fine with no linking warnings.

Then I changed my own project to use stm32_add_linker_script. I must be doing something wrong, because there are redefinition warnings for all my linker file sections in flash:

```
/usr/bin/arm-none-eabi-g++    -T /home/USER/PRODUCT_NAME/STM32G474VEHX_FLASH.ld --sysroot=\"/usr/arm-none-eabi\" -mthumb -mabi=aapcs -Wl,--gc-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard /home/USER/PRODUCT_NAME/build/G474VE.ld --specs=nosys.specs CMakeFiles/PRODUCT_NAME.elf.dir/Src/main.c.obj CMakeFiles/PRODUCT_NAME.elf.dir/Src/Application.cpp.obj ... ... CMakeFiles/PRODUCT_NAME.elf.dir/home/USER/STM32Cube/Repository/STM32Cube_FW_G4_V1.2.0/Drivers/CMSIS/Device/ST/STM32G4xx/Source/Templates/gcc/startup_stm32g474xx.s.obj  -o PRODUCT_NAME.elf  LIBRARY1.a LIBRARY2.a LIBRARY1.a 

/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld:/home/ra/PRODUCT_NAME/build/G474VE.ld:9: warning: redeclaration of memory region `FLASH'
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld:/home/ra/PRODUCT_NAME/build/G474VE.ld:10: warning: redeclaration of memory region `RAM'
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld:/home/ra/PRODUCT_NAME/build/G474VE.ld:11: warning: redeclaration of memory region `CCMRAM'
```
So it seems the file /home/USER/PRODUCT_NAME/build/G474VE.ld is generated and included anyway even though I called stm32_add_linker_file. But in the simple example, that does not seem to happen. I should mention that this was a clean build with no cached files.

What am I doing wrong here?

barafael, 2020-07-06:
For reference, this was the problem:

target_link_libraries (
    PRODUCT_NAME
    CMSIS::STM32::G474VE
)

which caused the linker file generation. I removed the line, now I don't get this warning any more.

However, now I get another warning
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol Reset_Handler; defaulting to 0000000008000000

I fixed that warning by including the CubeMX-generated assembler startup file. I hope that this does not conflict with other parts of stm32-cmake.

ObKo, 2020-11-01:
CMSIS::STM32::G474VE includes both startup for G474xx devices and linker script for G474VE.

You can link CMSIS::STM32::G474xx for startup only, and add script using stm32_add_linker_script

closed 2020-11-01 by ObKo