ObKo/stm32-cmake #146: Dependency on gcc_stm32f1.cmake existing

ekliptik, 2020-10-09:
Currently, my H7 project requires the presence of also the F1 .cmake file. I can't only have the .cmakes I need, since in compiler detection, the gcc_stm32.cmake is run without the STM32_CHIP variable.
```log
[cmake] -- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
[cmake] CMake Error at /home/emil/esc/sandbox/SW/TRIX-LSCOM-MCU/src/cmake-modules/gcc_stm32.cmake:114 (INCLUDE):
[cmake]   INCLUDE could not find load file:
[cmake] 
[cmake]     gcc_stm32f1
[cmake] Call Stack (most recent call first):
[cmake]   /home/emil/esc/sandbox/SW/TRIX-LSCOM-MCU/build/CMakeFiles/3.16.3/CMakeSystem.cmake:6 (include)
[cmake]   /home/emil/esc/sandbox/SW/TRIX-LSCOM-MCU/build/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)
```
```cmake
# https://github.com/ObKo/stm32-cmake/blob/master/stm32-template/CMakeLists.txt
set (STM32_CHIP STM32H743ZI)
project (trix-lscom-mcu) # this causes the first run of gcc_stm32.cmake

cmake_minimum_required (VERSION 3.0)
enable_language (ASM)

find_package (CMSIS REQUIRED)
# We are assuming firmware packages 1.8, lwIP 2.0
# we are not using eth from HAL, because source required adding __DMB() where OWN bit is set or after it is read
# links to lwIP pitfalls and bugs in README.md
find_package (STM32HAL COMPONENTS tim gpio cortex dma i2c tim flash rcc pcd gpio pwr uart spi REQUIRED) # mdma exti

file (STRINGS "includes.txt" INCLUDES)
file (STRINGS "sources.txt"  PROJECT_SOURCES)

include_directories (
    ${CMAKE_CURRENT_SOURCE_DIR}
    ${CMSIS_INCLUDE_DIRS}
    ${STM32HAL_INCLUDE_DIR}
    ${INCLUDES}
    )
    
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} --specs=nosys.specs")
set (STM32_LINKER_SCRIPT "modified_STM32H743ZITx_FLASH.ld")

add_executable (${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES} ${CMSIS_SOURCES} ${STM32HAL_SOURCES})

stm32_set_target_properties (${CMAKE_PROJECT_NAME})
stm32_add_hex_bin_targets (${CMAKE_PROJECT_NAME})
stm32_print_size_of_targets (${CMAKE_PROJECT_NAME})
```
Is this expected behavior? Is my configuration wrong in setting the variable in CMakeLists or in the way it is set? Or is this a bug?

plumewind, 2020-10-12:
try this?
`# define chip used in this project, this set must define before project definition
# for this project dont use cmake commandline option -DSTM32_CHIP=<chip>
SET(STM32_CHIP STM32F401xC)
SET(CMAKE_TOOLCHAIN_FILE ~/stm32-cmake/cmake/gcc_stm32.cmake)`

ekliptik, 2020-10-15:
I will never use an absolute path in my CMakeLists.txt, so I specified a relative path like
```cmake
set (STM32_CHIP STM32H743ZI)
set (CMAKE_TOOLCHAIN_FILE ./cmake-modules/gcc_stm32.cmake)
```
with my CMake args:
```json
        "CMAKE_BUILD_TYPE":"Debug",
        "STM32Cube_DIR":"~/STM32Cube/Repository/STM32Cube_FW_H7_V1.8.0/",
```

After removing the gcc_stm32f1.cmake file from cmake-modules, the issue persisted. My build setup is still dependant on the F1 package that I don't use. 
```
[cmake] -- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
[cmake] CMake Error at /home/emil/sandbox/SW/projectname/src/cmake-modules/gcc_stm32.cmake:114 (INCLUDE):
[cmake]   INCLUDE could not find load file:
[cmake] 
[cmake]     gcc_stm32f1
[cmake] Call Stack (most recent call first):
[cmake]   /home/emil/sandbox/SW/projectname/build/CMakeFiles/3.16.3/CMakeSystem.cmake:6 (include)
[cmake]   /home/emil/sandbox/SW/projectname/build/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)
```

Hish15, 2020-10-15:
@ekliptik, can you try again using the modern_cmake branch please?

ekliptik, 2020-10-15:
Wow! The modern_cmake branch depends on [every single stm32 family file instead!](https://github.com/ObKo/stm32-cmake/blob/a986efaed7ab6a9e7a696d545b115f45b262999f/cmake/stm32/common.cmake#L224)

The typical firmware development usecase is multiple people with one build setup, with differing paths to libraries, but with the same firmware library version, developing for one single device. Or maybe multiple targets for multiple devices. With the CMake modules in the path of the repository - so requiring all the modules to be present in said directory seems to be a bad idea. I'm into modern CMake, but not like this.

ekliptik, 2020-10-15:
I'm struggling to port my test project to modern_cmake, since I can't explicitly list the components I want. I can't use ETH driver since the latest firmware package from ST has multiple errors (in LwIP unbuildable code, in ETH drivers missing memory barriers in some ownership bit access, Piranha on the STM forums documented this well). For this reason I need control over what gets built from the repository.

ObKo, 2020-10-18:
> Wow! The modern_cmake branch depends on [every single stm32 family file instead!](https://github.com/ObKo/stm32-cmake/blob/a986efaed7ab6a9e7a696d545b115f45b262999f/cmake/stm32/common.cmake#L224) [...]

So, I don't get it a bit... 

You're keeping .cmake module for one family and removing all other? For what? stm32-cmake will search for CMSIS/HAL sources only for families specifiied in COMPONENTS.

Only one problem I can see with current implementation is some slowdown  because of all cmake modules being parsed.

> I'm struggling to port my test project to modern_cmake, since I can't explicitly list the components I want.

Why? In modern_cmake you're linking to drivers explicitly. Like
```
target_link_libraries(stm32-blinky-f4 
    HAL::STM32::F4::RCC
    HAL::STM32::F4::GPIO
    HAL::STM32::F4::CORTEX
    CMSIS::STM32::F407VG
    STM32::NoSys 
)
```
only RCC, GPIO and CORTEX drivers will be built and linked. Together with CMSIS + linker script for STM32F407VG.

Or do you using own reduced version of HAL, with drivers being physically deleted from HAL tree? So stm32-cmake can't find it and complains about that? 

Maybe I should let users to limit drivers being searched.

ObKo, 2020-11-01:
Added support for limited HAL search: https://github.com/ObKo/stm32-cmake/pull/117/commits/386cb3b85d1a781fcc9fa8139ed5a47688f4ef39.

In your case it should looks like:

```
cmake_minimum_required (VERSION 3.13)
project(trix-lscom-mcu C ASM)

find_package (CMSIS COMPONENTS STM32H743ZI REQUIRED)
find_package (STM32HAL COMPONENTS STM32F4 tim gpio cortex dma i2c tim flash rcc pcd gpio pwr uart spi REQUIRED) # mdma exti

file (STRINGS "includes.txt" INCLUDES)
file (STRINGS "sources.txt"  PROJECT_SOURCES)

set(CMAKE_INCLUDE_CURRENT_DIR TRUE)
include_directories (${INCLUDES})

add_executable(${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES})
target_link_libraries(${CMAKE_PROJECT_NAME} 
    HAL::STM32::F4::RCC
    HAL::STM32::F4::GPIO
    HAL::STM32::F4::CORTEX
    HAL::STM32::F4::TIM
    HAL::STM32::F4::DMA
    HAL::STM32::F4::I2C
    HAL::STM32::F4::FLASH
    HAL::STM32::F4::PCD
    HAL::STM32::F4::PWR
    HAL::STM32::F4::UART
    HAL::STM32::F4::SPI
    CMSIS::STM32::F407xx 
    STM32::NoSys
)
stm32_add_linker_script(${CMAKE_PROJECT_NAME} PRIVATE modified_STM32H743ZITx_FLASH.ld)
```

closed 2020-11-01 by ObKo