ObKo/stm32-cmake #147: use MicroLIB

plumewind, 2020-10-12:
hello,

first ,thanks a lot, I can develop stm32 in ubuntu.
But, a question: how to link microlib in cmakelist? is it in ubuntu?

I know it is in keil, it is private in keil. And how to use it in ubuntu?

Anyone can help me ? thank you very much.

Hish15, 2020-10-15:
Hi,
You may read the licence provided with microlib, but from my understanding you do not have access to it outside arm's toolchain...
If you do use armcc as compiler then no problemo :
```
armcc --library_type=microlib -c main.c
```
In Cmake land this is adding a `target_compile_options()` to the top target of your project (and selecting armcc as compiler).

ucayalifish, 2020-10-15:
Sorry for intervening, but is it really possible to use stm32-cmake with armcc, version 5.x in particular?

Hish15, 2020-10-15:
@ucayalifish, you are 100% right, stm32-cmake works with arm-gcc.
I don't now if there is a lot of tweaking to make to make it work in stm32-cmake
Here is the entry point if you want to try changing it and see what breaks.
https://github.com/ObKo/stm32-cmake/blob/modern-cmake/cmake/stm32_gcc.cmake

plumewind, 2020-11-17:
Thank you very much. Although I have implemented my original requirements through other methods (I don't need to rely on microlib Library), I still try your method. It seems that I can't write it right?

`## add project components
set(ELF_TARGET ${CMAKE_PROJECT_NAME}.elf)
ADD_EXECUTABLE(${ELF_TARGET} ${PROJECT_SOURCES} ${STM32HAL_SOURCES})
target_link_libraries(${ELF_TARGET} PUBLIC  -lc -lm -lnosys  --library_type=microlib)`

and get a error:
`arm-none-eabi-gcc: error: unrecognized command line option '--library_type=microlib'; did you mean '--library-directory'?`

Hish15, 2020-11-17:
Hi @plumewind, If you don't change the content on : 
https://github.com/ObKo/stm32-cmake/blob/aad75627d39f51753e00ffc9add0f03f95d2a211/cmake/stm32_gcc.cmake#L7
You will be stuck with arm-gcc.

What are you trying to do precisely?

plumewind, 2020-12-07:
This seems to have taken a detour.
In fact, I want to use the "printf()" function. In keil, it is implemented through microlib, but in the cmake project, he needs to implement it himself.

Now, I wrote a "my_printf()" myself, but it still doesn't work to print floating point numbers.

plumewind, 2020-12-07:
You are great, you have updated a lot of places!

I still use the "V1" version, which feels pretty good. I don’t know the current version. Does the supported library-newlib (libc) mean that printf can be used? How to use it?
What kind of links need to be added to cmakelist? What additional code needs to be added to main.c?

Thank you very much, it was a very happy thing!

atsju, 2020-12-19:
Hi @plumewind 
Have you fixed the issue ?
Could you please share what compiler you are using (armcc or arm_none-eabi-gcc) ?
Describe what exactly is not working and maybe share some minimal project on a public github so that we can have a look.
Also please switch to [V2](https://github.com/ObKo/stm32-cmake/releases/tag/v2.0) as your issue may simply be linked to #72.
Normally printf of float is working if the newlib library is linked (no floats by default with newlib-nano). There is still some bit of code to be added to direct the flow to correct output (you will find on google) but no need for `my_printf`.

plumewind, 2020-12-21:
THANKS,
what I say is this :
[image]

I now know that this library is unique to keil and not available in Ubuntu.

But my fundamental goal is to implement printf ().

I will try to using printf by V2, right now

atsju, 2020-12-21:
You didn't say if you were using arm-none-eabi-gcc or armcc from keil. Anyway if you are using arm-none-eabi-gcc following should help you. 

I just tried on latest develop modifying blinky.

```
#include <stdio.h>
printf("%f",4.2f);
```
this compiles without any error but we do not know where printf ends (this is normal)...
So you have to enhance with example from `GIT\STM32CubeF4\Projects\STM32F412ZG-Nucleo\Examples\UART\UART_Printf\` for example. This printf thing is always a pain in embedded so I generally prefer sprintf + calling my own sending function but it some cases with a real dedicated output channel (screen or UART) printf is fine.

I have no board to test but basically a good start would be :
implement into your main.c (or other)
```
int __io_putchar(int ch)
{
    // your send function here
    return ch;
}
```
copy `syscall.c` from example to your folder and ensure it is compiled by adding it into your cmakelist.

I do not really see any issue related to cmake-stm32 here.

plumewind, 2020-12-24:
Thank you very much！

At first, I  using arm-none-eabi-gcc in  cmake project, and using C51 in keil project. 

This is a certain fact, printf is often not easy to use on boards. Moreover, I now use sprintf + printf (% s ", STR) instead, but I always feel that this way is not a direct solution. Maybe this is me as a programmer, a stubborn bar.

I see. Thank you ！

closed 2020-12-24 by plumewind