ObKo/stm32-cmake #151: Missing Include Dir after Fetching HAL?

DuftendMystischerTapir, 2020-11-04:
Hi,

first of all: thanks! Great project. Just started a small project and was up and running within minutes. :)

One question though. When I tried using this on an existing project, `make` was not able to find `stm32f4xx_hal.h` included in one of my source files. Is it possible that the Fetch action is not adding the HAL (and CMSIS?) `Inc` folders to the `include_directories`?
When I add `${CMAKE_CURRENT_BINARY_DIR}/_deps/stm32-hal-f4-src/Inc`, it compiles but I think this is not the way it should work, I mean, manually refering to the build dir. ... But my CMake Foo is weak. ;)

Well, furthermore it does compiles then but not link. So this gives me another hunch that I am on the wron track here.

Can you see where the problem is?

Thanks.

ObKo, 2020-11-06:
Can you provide your CMakeLists.txt and cmake output log?

stm32_fetch_cmsis/stm32_fetch_hal should set `STM32_CMSIS_${FAMILY}_PATH`/`STM32_HAL_${FAMILY}_PATH`and CMSIS/HAL packages should use them.

DuftendMystischerTapir, 2020-11-06:
Hi,

thanks for your answer.

Here is the CMakeLists.txt

```
cmake_minimum_required(VERSION 3.13)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/cmake/stm32_gcc.cmake)

project(ClusterManager CXX C ASM)

stm32_fetch_cmsis(F4)
stm32_fetch_hal(F4)

find_package(CMSIS COMPONENTS STM32F412CE REQUIRED)
find_package(HAL COMPONENTS STM32F4 REQUIRED)

include_directories(
    App/Inc
    Core/Inc
)

set(PROJECT_SOURCES
    startup/startup_stm32f412cx.s
    ### Core/ is code generated from CubeMX
    Core/Src/main.c
    Core/Src/gpio.c
    Core/Src/adc.c
    Core/Src/can.c
    Core/Src/dma.c
    Core/Src/iwdg.c
    Core/Src/rtc.c
    Core/Src/tim.c
    Core/Src/stm32f4xx_it.c
    Core/Src/stm32f4xx_hal_msp.c
    Core/Src/system_stm32f4xx.c
)

add_executable(ClusterManager.elf ${PROJECT_SOURCES})

target_compile_definitions(
    ClusterManager.elf PUBLIC 
    USE_HAL_DRIVER=1
    STM32F412Cx=1
)

target_link_libraries(ClusterManager.elf CMSIS::STM32::F412CE STM32::NoSys)
stm32_add_linker_script(ClusterManager.elf PRIVATE STM32F412CEUx_FLASH.ld)
```

The cmake output looks like this ... which sems alright:
```
$ cmake -G"MinGW Makefiles" ..
-- No STM32_TOOLCHAIN_PATH specified, using default: /usr
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin/arm-none-eabi-gcc.exe
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 73 lines cut ...]
-- Generating done
-- Build files have been written to: C:/workspace_clustermanager/clustermanager/build-cmake
```

Then running `make` fails:
```
$ mingw32-make.exe
Scanning dependencies of target CMSIS_LD_F412CE
[  6%] Generating F412CE.ld
[  6%] Built target CMSIS_LD_F412CE
Scanning dependencies of target ClusterManager.elf
[ 12%] Building ASM object CMakeFiles/ClusterManager.elf.dir/startup/startup_stm32f412cx.s.obj
[ 18%] Building C object CMakeFiles/ClusterManager.elf.dir/Core/Src/main.c.obj
In file included from C:\workspace_clustermanager\clustermanager\Core\Src\main.c:22:
C:\workspace_clustermanager\clustermanager\Core\Inc/main.h:31:10: fatal error: stm32f4xx_hal.h: No such file or directory
   31 | #include "stm32f4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles\ClusterManager.elf.dir\build.make:85: recipe for target 'CMakeFiles/ClusterManager.elf.dir/Core/Src/main.c.obj' failed
mingw32-make[2]: *** [CMakeFiles/ClusterManager.elf.dir/Core/Src/main.c.obj] Error 1
CMakeFiles\Makefile2:121: recipe for target 'CMakeFiles/ClusterManager.elf.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/ClusterManager.elf.dir/all] Error 2
Makefile:101: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```

ObKo, 2020-11-06:
> `target_link_libraries(ClusterManager.elf CMSIS::STM32::F412CE STM32::NoSys)`

You should also link to HAL drivers you're using, e.g.

```
target_link_libraries(ClusterManager.elf 
    HAL::STM32::F4::RCC
    HAL::STM32::F4::GPIO
    HAL::STM32::F4::CORTEX
    CMSIS::STM32::F412CE 
    STM32::NoSys
)
```

DuftendMystischerTapir, 2020-11-06:
Hi,

thanks for the hint. It works now. :)

I now also saw this in the blinky example. Futhermore there you list the HAL modules in `find_package(HAL ...` call. Is that also needed?

Cheers

Hish15, 2020-11-14:
Hi @DuftendMystischerTapir, if everything works fine now, do not hesitate to close the ticket 😄

closed 2020-11-16 by ObKo