ObKo/stm32-cmake #157: Compile and link options don't get applied

IboKhaled, 2020-11-19:
I'm trying to set up a cross compilation toolchain for a `STM32F446RE`, using blinky for testing. I removed the `CMAKE_TOOLCHAIN_FILE` path from the *CMakeLists.txt* and pass it from the build script.

My build script looks like this:
```
#!/bin/bash

cmake . -DSTM32_CUBE_F4_PATH=/opt/STM32CubeF4/ -DCMAKE_BUILD_TYPE=Debug -DSTM32_TARGET_TRIPLET=arm-none-eabi -DSTM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/ -DCMAKE_TOOLCHAIN_FILE=/opt/stm32-cmake/cmake/stm32_gcc.cmake -DCMAKE_C_COMPILER=/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -B ./build/ &&\
cd build &&\
make &&\
cd .. &&\
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-objcopy -O binary stm32-blinky-f4 stm32-blinky-f4.bin
```
The *CMakeOutput.log* shows, that the gcc-options are not applied and compilation is done for the wrong target:
```
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_58312.dir/CMakeCCompilerABI.c.obj' '-c' '-mcpu=arm7tdmi' '-mfloat-abi=soft' '-marm' '-march=armv4t'
```
Also an error can be found in *CMakeError.log*, which probably arises out of the above:
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc 
Build flags: 
Id flags:  

The output was:
1
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```
A cmake trace showed, that *f4.cmake* is at least read.

The `TOOLCHAIN_PREFIX` option mentioned in the ReadMe triggers a warning, so I removed it from my CMake call:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    TOOLCHAIN_PREFIX
```
Compilation and linking works without errors shown in the terminal.
I suspect, I'm missing one flag in the CMake call, such as `-DTARGET=STM32F446RE`, but I don't know which one it is.

Any help would be greatly appreciated!

Hish15, 2020-11-19:
Hi @IboKhaled,
You must not define some of those var in the CLI, and some are wrong. Can you try with : 
```
cmake . -DSTM32_CUBE_F4_PATH=/opt/STM32CubeF4/ -DCMAKE_BUILD_TYPE=Debug -DSTM32_TARGET_TRIPLET=arm-none-eabi -DSTM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update -DCMAKE_TOOLCHAIN_FILE=/opt/stm32-cmake/cmake/stm32_gcc.cmake -B ./build/
```

I assume that you changed the blinky CMakeLists.txt to look like so :

```
cmake_minimum_required(VERSION 3.13)

project(stm32-blinky C ASM)
set(CMAKE_INCLUDE_CURRENT_DIR TRUE)

find_package(CMSIS COMPONENTS STM32F4 REQUIRED)
find_package(HAL COMPONENTS STM32F4 RCC GPIO CORTEX REQUIRED)
# Find all drivers:
#find_package(HAL COMPONENTS STM32L0 STM32F1 STM32F4 REQUIRED)
# Find drivers for all families:
#find_package(HAL COMPONENTS RCC GPIO CORTEX REQUIRED)
# Find LL driver:
#find_package(HAL COMPONENTS LL_GPIO REQUIRED)
# Find everything:
#find_package(HAL REQUIRED)

# STM32F4-Discovery
add_executable(stm32-blinky-f4 blinky.c stm32f4xx_hal_conf.h)
target_link_libraries(stm32-blinky-f4
    HAL::STM32::F4::RCC
    HAL::STM32::F4::GPIO
    HAL::STM32::F4::CORTEX
    CMSIS::STM32::F407VG
    STM32::NoSys
)
```

IboKhaled, 2020-11-20:
Hi @Hish15,
thank you for your quick response.

Using your CMake call, the error still persists, so no changes in the *CMakeOutput.log* or *CMakeError.log*, the build diectory was cleaned before rebuilding.

I modified the *CMakeLists.txt* to look like this:
```
cmake_minimum_required(VERSION 3.13)

project(stm32-blinky C ASM)
set(CMAKE_INCLUDE_CURRENT_DIR TRUE)

find_package(CMSIS COMPONENTS STM32F446RE REQUIRED)
find_package(HAL COMPONENTS STM32F446RE REQUIRED)

add_executable(stm32-blinky-f4 blinky.c stm32f4xx_hal_conf.h)
target_link_libraries(stm32-blinky-f4 
    HAL::STM32::F4::RCC
    HAL::STM32::F4::GPIO
    HAL::STM32::F4::CORTEX
    CMSIS::STM32::F446RE
    STM32::NoSys 
)

set_target_properties(stm32-blinky-f4 PROPERTIES RUNTIME_OUTPUT_DIRECTORY ./..)
```
I also changed the top section of *blinky.c* like so:
```
#include <stm32f4xx_hal.h>

// STM32F4-Discovery green led - PD12 - corrected to PA5 on STM32F446RE
#define LED_PORT                GPIOA
#define LED_PIN                 GPIO_PIN_5
#define LED_PORT_CLK_ENABLE     __HAL_RCC_GPIOA_CLK_ENABLE

void SysTick_Handler(void)
{
    HAL_IncTick();
...
```

Hish15, 2020-11-20:
Can you start from the **clean** sha aad75627d39f51753e00ffc9add0f03f95d2a211,
perform the command bellow, and past the output here please ?
```
cmake -Bbuild . -DSTM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update && make --directory=build
```
This assume that you download cube L0, F1 and F4 in opt/
If it is a problem for you, you can remove L0 and F1 stuff from the CMakeList.txt (nothing more)

IboKhaled, 2020-11-20:
This is the output on the terminal:
```
blinky (master)$ cmake -Bbuild . -DSTM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update && make --directory=build

-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Neither STM32_CUBE_L0_PATH nor STM32_CMSIS_L0_PATH specified using default  STM32_CUBE_L0_PATH: /opt/STM32CubeL0
-- Neither STM32_CUBE_F1_PATH nor STM32_CMSIS_F1_PATH specified using default  STM32_CUBE_F1_PATH: /opt/STM32CubeF1
-- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default  STM32_CUBE_F4_PATH: /opt/STM32CubeF4
-- Found CMSIS: /opt/STM32CubeL0/Drivers/CMSIS/Include;/opt/STM32CubeL0/Drivers/CMSIS/Device/ST/STM32L0xx/Include;/opt/STM32CubeF1/Drivers/CMSIS/Include;/opt/STM32CubeF1/Drivers/CMSIS/Device/ST/STM32F1xx/Include;/opt/STM32CubeF4/Drivers/CMSIS/Include;/opt/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Include (found version "5.4.0") found components: STM32L0 STM32F1 STM32F4 
-- Search for HAL families: STM32L0;STM32F1;STM32F4
-- Search for HAL drivers: RCC;GPIO;CORTEX
-- Search for HAL LL drivers: 
-- Found HAL: /opt/STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Inc;/opt/STM32CubeF1/Drivers/STM32F1xx_HAL_Driver/Inc;/opt/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Inc  found components: STM32L0 STM32F1 STM32F4 RCC GPIO CORTEX 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build
make: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
make[1]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
Scanning dependencies of target CMSIS_LD_L053C8
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
[ 11%] Generating L053C8.ld
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
[ 11%] Built target CMSIS_LD_L053C8
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
Scanning dependencies of target stm32-blinky-l0
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
[ 22%] Building C object CMakeFiles/stm32-blinky-l0.dir/blinky.c.obj
[ 22%] Building C object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc.c.obj
[ 22%] Building C object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal.c.obj
[ 22%] Building C object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/CMSIS/Device/ST/STM32L0xx/Source/Templates/system_stm32l0xx.c.obj
[ 33%] Building C object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_gpio.c.obj
[ 33%] Building C object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_cortex.c.obj
[ 33%] Building ASM object CMakeFiles/stm32-blinky-l0.dir/opt/STM32CubeL0/Drivers/CMSIS/Device/ST/STM32L0xx/Source/Templates/gcc/startup_stm32l053xx.s.obj
[ 44%] Linking C executable stm32-blinky-l0
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
[ 44%] Built target stm32-blinky-l0
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
Scanning dependencies of target CMSIS_LD_F407VG
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
[ 44%] Generating F407VG.ld
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
[ 44%] Built target CMSIS_LD_F407VG
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
Scanning dependencies of target stm32-blinky-f4
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird betreten
[ 44%] Building C object CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj
[ 55%] Building C object CMakeFiles/stm32-blinky-f4.dir/opt/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj
[ 55%] Building C object CMakeFiles/stm32-blinky-f4.dir/opt/STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c.obj
[ 55%] Building C object CMakeFiles/stm32-blinky-f4.dir/opt/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c.obj
[... cut ...]
make[2]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
[100%] Built target stm32-blinky-f1
make[1]: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
make: Verzeichnis „/home/ibrahim/projects/stm32-cmake-fresh/examples/blinky/build“ wird verlassen
```
Here are *CMakeOutput.log* and *CMakeError.log*:
[CMakeError.log](https://github.com/ObKo/stm32-cmake/files/5572999/CMakeError.log)
[CMakeOutput.log](https://github.com/ObKo/stm32-cmake/files/5573000/CMakeOutput.log)

Hish15, 2020-11-20:
So it worked.
in the build directory you have stm32-blinky-f1  stm32-blinky-f4  stm32-blinky-l0, those are the .elf to use.
The error in the Error.log is related to the fact that you are crosbuilding, no big deal.

IboKhaled, 2020-11-20:
Building works, true, but the code is build for the wrong target.

*CMakeOutput.log* shows, that gcc is compling using these (and more) arguments:
```
 '-mcpu=arm7tdmi' '-mfloat-abi=soft' '-marm' '-march=armv4t'
```
In *f4.cmake* is says, that these
```
-mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard
```
should be the ones used.

When running the code through a debugger, you can tell, that the processor throws an *UsageFault* Interrupt, which is caused by an "undefined instruction" (according to datasheet), which could (speculating) be caused by an instruction, which is supported by arm7tdmi, but unsupported by cortex-m4 processors.

Hish15, 2020-11-20:
What about the file `CMakeFiles/stm32-blinky-f4.dir/flags.make`
Mine shows : 
``` 
C_FLAGS =   --sysroot=\"/opt/gcc_aos/arm-none-eabi\" -mthumb -mabi=aapcs -Wall -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -ffast-math -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nosys.specs

C_DEFINES = -DSTM32F4 -DSTM32F407xx
``` 

cmake version?

IboKhaled, 2020-11-23:
You're absolutely right, mine also shows:
```
C_FLAGS =   --sysroot=\"/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi\" -mthumb -mabi=aapcs -Wall -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -ffast-math -Og -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard --specs=nosys.specs

C_DEFINES = -DSTM32F4 -DSTM32F446xx
```

I found the problem, it wasn't a problem with *stm32-cmake*, it was a problem with the debugging server. I used the *st-util* one, now tried *openOCD* and that one works fine.

I'm sorry for wasting your time and thank you for your help and patience!

closed 2020-11-23 by IboKhaled