ObKo/stm32-cmake #159: modern cmake - ex modules, how to use?

andyinno, 2020-11-25:
I am to setup a project using base code generated by STM32CodeIde,

while adding all the sources and start compiling the project I removed the duplicated source file that could be found via imported modules.
At the moment I have:

    ./Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim_ex.c

in sources. If I remove it gcc complains that:

    STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim.c:3314: undefined reference to `HAL_TIMEx_BreakCallback'

Following the README documentation I was tempted to add something like:

    HAL::STM32::F4::TIMEX

to the link libraries command but TIMEX is not found.

Can you please provide the right syntax for adding the EXT part?

What bothers me is that when cmake starts it output:

    Search for HAL families:...
    Search for HAL drivers...
    Search for HAL LL drivers...

And I don't see any EX module in there...

thank you in advance

Hish15, 2020-11-25:
Hello @andyinno ,

Assuming you are handling this as a compenent of HAL 
```
find_package(HAL COMPONENTS STM32F4 TIM)
```
Then the target name is `HAL::STM32::F4::TIMEx`
You where close with `EX` :smile:
I will make a PR to clarify this case!

andyinno, 2020-11-25:
Great, 

thank you. It works.
I think the problem is that in a lot of places cmake does not make distinction from lower case and upper case.
Also the first output produced by cmake listing all modules is lower case:

    Search for HAL drivers: adc;can;cec;comp;cortex;crc;dac;dma;exti;flash;gpio;i2c;i2s;irda;iwdg;pcd;pwr;rcc;rtc;smartcard;smbus;spi;tim;tsc;uart;usart;wwdg;eth;hcd;mmc;nand;nor;pccard;sd;sram;cryp;dcmi;hash;rng;hrtim;opamp;sdadc;dfsdm;dma2d;dsi;flash_ramfunc;fmpi2c;lptim;ltdc;qspi;sai;sdram;spdifrx;jpeg;mdios;cordic;fdcan;fmac;dts;gfxmmu;hsem;mdma;ospi;otfdec;pssi;ramecc;swpmi;firewall;lcd;pka

Therefore I did not pay attention to the case.
In the README it is spelled correctly.

PS: I don't know if I need to open a new issue or not, anyway... here a quick question: cmsis_os2.c file is used by the code generator for the FreeRTOSv2 module. It is correct that I need to import it manually or there is something missing?

Hish15, 2020-11-25:
@andyinno, I'm actually making changes to handle CMSIS_RTOS and CMSIS_RTOS2 as CMSIS components 😉. 
For now you have to handle it as outer code : create the target, add the sources and link the target to your executable's target (just a few line really 😃 )!

andyinno, 2020-11-26:
Thank you for your reply.
No problem with the few lines, this is what I currently did.

I was just worried to have misunderstood something in the documentation and therefore I was unsure about which path to follow.
Thank you!
Andrea

closed 2020-11-26 by andyinno