ObKo/stm32-cmake #164: CCRAM handling

atsju, 2020-12-14:
This issue is linked to #162 and #71.

The original problem is the linker script absolutely not working with STM32L476RG but it goes further.

**Description of the problem for L4:**
Depending on L4 series there may be 2 or 3 RAM banks from were one is CCRAM. By default the SRAM2 (CCRAM) is mapped at `0x1000 0000` and SRAM1 at `0x2000 0000`. SRAM3 exists on some L4+ devices but forget it for now as my proposal fits to this also. 

On current implementation we have: 
- the size of SRAM1+SRAM2 is put to `RAM` section at`0x2000 0000`
- the size of SRAM2 is put to `CCRAM` section at`0x1000 0000`

This is is working on most L4 because SRAM2 is also remapped to be contiguous to SRAM1 meaning it can be accessed independently from 2 addresses. BUT if someone tries to put variables to CCRAM section then chances are he/she will overwrite his/her own variable that is in RAM.

This is not working on some devices like STM32L476 because SRAM2 is NOT remapped to be contiguous. So the stack is initialized at an address that doesn't exist on device.

Note about SRAM3: this section exists on devices where SRAM2 is remapped and is SRAM1/2/3 can all be considered contiguous.

**Proposed resolution:**
As most users want to compile with all RAM available without putting explicitly variables to section CCRAM in the code, I recommend that CCRAM section is only defined on devices where SRAM1 and SRAM2 are not contiguous. On other devices the generated linker script would put only one big RAM section and thus be easy to use 100% of the RAM for the "basic" applications.
People that explicitly need separate CCRAM section for their application would then need to write some custom linker script. This seems fair as I don't think it's most widespread case.

Note I discussed with @Hish15 about this and it seems to be best compromise. Note also this is how ST handles the thing in their examples.

**Warning this should also be checked for all other families.**
 I plan to do a PR implementing this proposal.

atsju, 2020-12-17:
After reading the datasheets of [F3](https://www.st.com/en/microcontrollers-microprocessors/stm32f3-series.html) devices and according to [this document](https://www.st.com/resource/en/application_note/dm00083249-use-stm32f3stm32g4-ccm-sram-with-iar-ewarm-keil-mdkarm-and-gnubased-toolchains-stmicroelectronics.pdf) there are similar issues with F303 F3x4 and F3x8 where CCM SRAM cannot be aliased and thus the total "easily accessible" RAM is less than stated by just the header of datasheet.

atsju, 2020-12-17:
On F4 series, F413 and F423 work the same way and CCM SRAM memory can be aliased address or not. I corrected to be same handling as original proposal.

Hish15, 2020-12-19:
I agree with you @atsju, the idea is to have as much RAM  as possible available by default.
All cases where SRAM2/3,  is contiguous or remapped to contiguous memory with SRAM1, should be handled as a single big RAM.
Other cases, might be handled by advanced user with a specific linker script.

PR #165 (2021-01-31): manage CCRAM section
Change CCRAM management to be like explained in #164 

- [x] F0 - reviewed
- [x] F1 - reviewed
- [x] F2 - reviewed
- [x] F3 - commited correction
- [x] F4 - commited correction
- [x] F7 - reviewed
- [x] G0 - reviewed and added future devices
- [x] G4 - commited correction and added new/future device support
- [x] H7 - commited correction and added new/future device support
- [x] L0 - reviewed
- [x] L1 - reviewed and minor refactor
- [x] L4 - commited correction

closes #164
closed 2021-01-31 by ObKo