ObKo/stm32-cmake #168: Unsupported series STM32L5 and STM32WB

atsju, 2020-12-17:


ObKo, 2020-12-20:
I'd took a look at stm32wb during implementation of modern-cmake branch and it was look quite tricky to support, so I'd decided to skip it for future.

Problem here is same as in H7 family - non-homogeneous multicore. 
As I understood, for now, only CM4 core is used for user application in most cases, RF core usually runs ST"s firmware for BLE/ZigBee/Thread.

atsju, 2020-12-20:
Yes it's exactly what I though, it could be managed like H7 family.
From what I discussed with ST most users will use ST library for the second core but user may still build their own if needed. So both cores should be supported. I did't have a deep look into this for the moment, only used (only a little) their examples.

PR #235 (2021-08-06): Add support for WB and WL families
- Add messages with different level for easier debugging (by default low levels won't be shown to user)
- Add support for WB and WL families

I plan to add working blinky examples in a different PR

**Some things are missing for the moment**
- [x] project will configure (this is a great step) but not build due to special linker script requirements (however it builds with custom linker script). Needs to be investigated => Need to tweek WB linker script generation
- [x] WL family is not fully implemented
- [x] actions for WB and WL have not been added
- [x] the [... cut]
closed 2021-08-06 by atsju