ObKo/stm32-cmake #169: non homogeneous RAM size management

atsju, 2020-12-17:
https://github.com/ObKo/stm32-cmake/blob/aad75627d39f51753e00ffc9add0f03f95d2a211/cmake/stm32/f1.cmake#L9

L4 VS L1 is a good example. In L4, all RAM size are managed by 

```
set(STM32_L4_RAM_SIZES 
     40K  40K  64K  64K  64K  64K
     64K 160K 160K 160K  96K  96K
     96K  96K  96K 320K 320K 320K 
    320K 640K 640K 640K 640K 640K
    640K
)
```

whereas in F1 families, it's like  this followed by a buch of if else to set the correct RMA size

```
set(STM32_F1_RAM_SIZES 
     0K  0K  0K  0K  0K  0K  0K  0K
     0K  0K  0K  0K  0K  0K
)
```
I think having one method for all increases maintainability and readability.
I would be happy to help and do the modification but I do not know which one is the recommended way to do.

atsju, 2020-12-17:
also many memory management could be optimized for example `"F412C." "F412R." "F412V." "F412Z."` could be reduced to just 
 `"F412.."` like other in same family. It's strange to have different handling even inside same family, especially here where complexity is not needed.

atsju, 2020-12-19:
@ObKo  I suppose the `STM32_F1_TYPES` correspond to files in `STM32CubeF1\Drivers\CMSIS\Device\ST\STM32F1xx\Include` ?
If this is the case, Only some documentation may be needed.

ObKo, 2020-12-20:
> 
> 
> @ObKo I suppose the `STM32_F1_TYPES` correspond to files in `STM32CubeF1\Drivers\CMSIS\Device\ST\STM32F1xx\Include` ?
> If this is the case, Only some documentation may be needed.

Yes, that TYPES comes from ST definitions in their headers and reflects peripherals available. 
Often, device with same type has same RAM configuration.

atsju, 2020-12-20:
Thanks. This ticket can be closed as current architecture makes good sense.

closed 2020-12-20 by atsju