ObKo/stm32-cmake #170: Unsupported series STM32MP

atsju, 2020-12-19:
I think this could be integrated. At least I create the ticket for STM32 [MPU](https://www.st.com/en/microcontrollers-microprocessors/stm32-arm-cortex-mpus.html). 
Maybe someone will need it and do some pull request

ObKo, 2020-12-20:
We are talking about the built-in CM4 core, right? I don't think stm32-cmake makes sense for main A7 core.

atsju, 2020-12-20:
At first at though A7 and M4 but I had a deeper look and you are right. I don't think the A7 part make much sense in this repo.

Hish15, 2020-12-23:
Do not hesitate to close the ticket @atsju

atsju, 2020-12-23:
@Hish the CM4 core of MP1 still has (some) sense to be added into this repo. However I do not know A7 users habits and it they would be interested into this repo for CM4 co-processor

Hish15, 2020-12-24:
Ok!
For me two points :

- Things will get confusing with the way we manage multicore.
- Having just one part of a device handled by stm32-cmake seems weird

This is my 2 cents, not a definite answer. Keeping the ticket open wont cost much anyway.

sergeyblagodarnyy, 2021-04-13:
I've started this job in [my fork](https://github.com/Faramax/stm32-cmake.git) in the branch mp1.

atsju, 2022-02-24:
#283

closed 2022-02-24 by atsju