ObKo/stm32-cmake #171: Read RAM Information from CMSIS folder

Hish15, 2020-12-19:
I realized that the memory organization if available for all devices in
`STM32Cube<Family>/Drivers/CMSIS/Device/ST/STM32<Family>xx/Include`
Contains *.h file with SRAM1/2/3 base address and size.

Even more interesting : 
`STM32Cube<Family>/Drivers/CMSIS/Device/ST/STM32<Family>xx/Source/Template/iar/linker`

Contains  *flash.icf file with the stop and start address of full ram.

**This could be used to extract a default RAM configuration using the one provided by ST for IART.**

Hish15, 2020-12-19:
This is related to the work done on the PR #165

ObKo, 2020-12-20:
Are you talking about RAM information extracting during cmake run?
If so - I've tried to make core part of stm32-cmake independent from CMSIS/STM32Cube, because there is cases when CMSIS isn't used, like ChibiOS or libopencm3 and that feature will tie stm32-cmake to CMSIS.

Hish15, 2020-12-20:
***F0***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f030x6|4|0
f030x8|8|0
f030xc|32|0
f031x6|4|0
f038xx|4|0
f042x6|6|0
f048xx|6|0
f051x8|8|0
f058xx|8|0
f070x6|6|0
f070xb|16|0
f071xb|16|0
f072xb|16|0
f078xx|16|0
f091xc|32|0
f098xx|32|0

***F1***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f100xb|8|0
f100xe|32|0
f101x6|6|0
f101xb|16|0
f101xe|48|0
f101xg|80|0
f102x6|6|0
f102xb|16|0
f103x6|10|0
f103xb|20|0
f103xe|64|0
f103xg|96|0
f105xc|64|0
f107xc|64|0

***F2***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f205xx|128|0
f207xx|128|0
f215xx|128|0
f217xx|128|0

***F3***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f301x8|16|0
f302x8|16|0
f302xc|40|0
f302xe|64|0
f303x8|12|4
f303xc|40|8
f303xe|64|16
f318xx|16|0
f328xx|12|4
f334x8|12|4
f358xx|40|8
f373xc|32|0
f378xx|32|0
f398xx|64|16

***F4***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f401xc|64|0
f401xe|96|0
f405xx|128|64
f407xx|128|64
f410cx|32|0
f410rx|32|0
f410tx|32|0
f411xe|128|0
f412cx|256|0
f412rx|256|0
f412vx|256|0
f412zx|256|0
f413xx|320|0
f415xx|128|64
f417xx|128|64
f423xx|320|0
f427xx|192|64
f429xx|192|64
f437xx|192|64
f439xx|192|64
f446xx|128|0
f469xx|320|64
f479xx|320|64

***F7***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
f722xx|256|0
f723xx|256|0
f730xx|256|0
f732xx|256|0
f733xx|256|0
f745xx|320|0
f746xx|320|0
f750xx|320|0
f756xx|320|0
f765xx|512|0
f767xx|512|0
f769xx|512|0
f777xx|512|0
f779xx|512|0

Hish15, 2020-12-20:
***G0***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
g030xx|8|0
g031xx|8|0
g041xx|8|0
g050xx|16|0
g051xx|16|0
g061xx|16|0
g070xx|36|0
g071xx|36|0
g081xx|36|0
g0b0xx|144|0
g0b1xx|144|0
g0c1xx|144|0
***G4***
| Device | Ram size | CCRAM size |
| --- | --- | ---|
g431xx|32|0
g441xx|32|0
g471xx|128|0
g473xx|128|0
g474xx|128|0
g481xx|128|0
g483xx|128|0
g484xx|128|0
g491xx|128|0
g4a1xx|128|0
gbk1cb|32|0

Hish15, 2020-12-20:
| Device | Ram size | CCRAM size |
| --- | --- | ---|
l010x4|2|0
l010x6|8|0
l010x8|8|0
l010xb|20|0
l011xx|2|0
l021xx|2|0
l031xx|8|0
l041xx|8|0
l051xx|8|0
l052xx|8|0
l053xx|8|0
l061xx|8|0
l062xx|8|0
l063xx|8|0
l071xx|20|0
l072xx|20|0
l073xx|20|0
l081xx|20|0
l082xx|20|0
l083xx|20|0
***L1***
l100xba|16|0
l100xb|10|0
l100xc|16|0
l151xba|32|0
l151xb|16|0
l151xca|32|0
l151xc|32|0
l151xdx|80|0
l151xd|48|0
l151xe|80|0
l152xba|32|0
l152xb|16|0
l152xca|32|0
l152xc|32|0
l152xdx|80|0
l152xd|48|0
l152xe|80|0
l162xca|32|0
l162xc|32|0
l162xdx|80|0
l162xd|48|0
l162xe|80|0
***L4***
l412xx|40|0
l422xx|40|0
l431xx|64|0
l432xx|64|0
l433xx|64|0
l442xx|64|0
l443xx|64|0
l451xx|160|0
l452xx|160|0
l462xx|160|0
l471xx|96|0
l475xx|96|0
l476xx|96|0
l485xx|96|0
l486xx|96|0
l496xx|320|0
l4a6xx|320|0
l4p5xx|320|0
l4q5xx|320|0
l4r5xx|640|0
l4r7xx|640|0
l4r9xx|640|0
l4s5xx|640|0
l4s7xx|640|0
l4s9xx|640|0
***L5***
l552xc_flash|96|0
l552xc_flash|96|0
l552xc|192|0
l552xe_flash|96|0
l552xe_flash|96|0
l552xe|192|0
l562xe_flash|96|0
l562xe_flash|96|0
l562xe|192|0

Hish15, 2020-12-20:
| Device | Ram size | CCRAM size |
| --- | --- | ---|
h723xe_flash_rw|320|0
h723xe_flash_rw|320|0
h723xe_flash_rw|32|0
h723xe_flash_rw|32|0
h723xe|128|0
h723xx_flash_rw|320|0
h723xx_flash_rw|320|0
h723xx_flash_rw|32|0
h723xx_flash_rw|32|0
h723xx|128|0
h725xe_flash_rw|320|0
h725xe_flash_rw|320|0
h725xe_flash_rw|32|0
h725xe_flash_rw|32|0
h725xe|128|0
h725xx_flash_rw|320|0
h725xx_flash_rw|320|0
h725xx_flash_rw|32|0
h725xx_flash_rw|32|0
h725xx|128|0
h730xx_flash_rw|320|0
h730xx_flash_rw|320|0
h730xx_flash_rw|32|0
h730xx_flash_rw|32|0
h730xxq_flash_rw|320|0
h730xxq_flash_rw|320|0
h730xxq_flash_rw|32|0
h730xxq_flash_rw|32|0
h730xxq|128|0
h730xx|128|0
h733xx_flash_rw|320|0
h733xx_flash_rw|320|0
h733xx_flash_rw|32|0
h733xx_flash_rw|32|0
h733xx|128|0
h735xx_flash_rw|320|0
h735xx_flash_rw|320|0
h735xx_flash_rw|32|0
h735xx_flash_rw|32|0
h735xx|128|0
h742xg_flash_rw|32|0
h742xg_flash_rw|384|0
h742xg|128|0
h742xx_flash_rw|32|0
h742xx_flash_rw|384|0
h742xx|128|0
h743xg_flash_rw|288|0
h743xg_flash_rw|512|0
h743xg|128|0
h743xx_flash_rw|288|0
h743xx_flash_rw|512|0
h743xx|128|0
h745xg_flash|128|0
h745xg_flash|288|0
h745xx_flash_rw_sram1|512|0
h745xx_flash_rw_sram2|288|0
h745xx_flash|128|0
h745xx_flash|288|0
h747xg_flash|128|0
h747xg_flash|288|0
h747xx_flash_rw_sram1|512|0
h747xx_flash_rw_sram2|288|0
h747xx_flash|128|0
h747xx_flash|288|0
h750xx_flash_rw|288|0
h750xx_flash_rw|512|0
h750xx|128|0
h753xx_flash_rw|288|0
h753xx_flash_rw|512|0
h753xx|128|0
h755xx_flash_rw_sram1|512|0
h755xx_flash_rw_sram2|288|0
h755xx_flash|128|0
h755xx_flash|288|0
h757xx_flash_rw_sram1|512|0
h757xx_flash_rw_sram2|288|0
h757xx_flash|128|0
h757xx_flash|288|0
h7a3xg_flash_rw|1024|0
h7a3xg_flash_rw|128|0
h7a3xgq_flash_rw|1024|0
h7a3xgq_flash_rw|128|0
h7a3xgq|128|0
h7a3xg|128|0
h7a3xx_flash_rw|1024|0
h7a3xx_flash_rw|128|0
h7a3xxq_flash_rw|1024|0
h7a3xxq_flash_rw|128|0
h7a3xxq|128|0
h7a3xx|128|0
h7b0xx_flash_rw|1024|0
h7b0xx_flash_rw|128|0
h7b0xxq_flash_rw|1024|0
h7b0xxq_flash_rw|128|0
h7b0xxq|128|0
h7b0xx|128|0
h7b3xx_flash_rw|1024|0
h7b3xx_flash_rw|128|0
h7b3xxq_flash_rw|1024|0
h7b3xxq_flash_rw|128|0
h7b3xxq|128|0
h7b3xx|128|0

Hish15, 2020-12-20:
@ObKo, I did not that ChibiOS did not use STM32 HAL.
I did run a script on HAL folder anyway to have a base value.

closed 2021-01-31 by ObKo