ObKo/stm32-cmake #176: Include Paths Error: Invalid Argument Error

Danaozhong, 2020-12-25:
I faced an issue when updating the STM32 environment in my project to the current master of this project. I am not sure what is the actual root cause, but I had to change the cmake scripts to get in running.
I am building on Windows 10, and get a several build error because of an invalid include paths:

cc1.exe: error: "D:/Programs/STM32CubeIDE/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.win32_1.0.0.201904181610/tools/arm-none-eabi"/build/gnu-tools-for-stm32_7-2018-q2-update.20190328-1800/install-mingw/lib/gcc/arm-none-eabi/7.3.1/../../../../include: Invalid argument
[build_output_error.txt](https://github.com/ObKo/stm32-cmake/files/5742402/build_output_error.txt)
Full build log is attached.
[build_output_error.txt](https://github.com/ObKo/stm32-cmake/files/5742404/build_output_error.txt)

I managed to get rid of this error by changing the lines in utilities.cmake:
From 
`--sysroot="${TOOLCHAIN_SYSROOT}"`
to
`--sysroot=${TOOLCHAIN_SYSROOT}`

However, I don't think this is a good idea, if the path contains spaces. Do you have a better suggestions on how to fix this?

This is how I run cmake on my side:
`DIR_STM32_GNU_TOOLCHAIN="D:/Programs/STM32CubeIDE/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.win32_1.0.0.201904181610/tools/"
DIR_STM32_CUBE="C:/Users/Clemens/STM32Cube/Repository/STM32CubeF3"

cmake -DSTM32_CUBE_F3_PATH=$DIR_STM32_CUBE -DSTM32_TOOLCHAIN_PATH=$DIR_STM32_GNU_TOOLCHAIN -DTARGET_TRIPLET=arm-none-eabi -DCMAKE_BUILD_TYPE=Debug -G "Eclipse CDT4 - Unix Makefiles" Src
`

Many thanks in advance!

Additional information: I am using FreeRTOS, but not the variant provided by this CMake environment. I use my own one, and link it as a library. This worked well with a previous version of stm32-cmake (several month old), but with the new version I got problems. The error message indicates a problem with the \" characters in combination with the

atsju, 2021-01-31:
Hi @Danaozhong 
thank you for this feedback. I think you found a problem in path management.
Can you please try with a different generator like ninja or Unix Makefiles and tell if the issue is the same ?
Do you think you could provide a minimal example that reproduces the issue ? I feel like I could fix this if I am able to reproduce.

atsju, 2021-01-31:
@ObKo if it's OK to force 3.20 version of cmake I think it would be great to use [CMAKE_PATH](https://cmake.org/cmake/help/git-master/command/cmake_path.html?highlight=cmake_path) instead of SET to avoid any such issue. What do you think ?

Hish15, 2021-01-31:
Hi,
3.20 is not even a release yet 😄 
Any way, major distributions such as ubuntu-20.04 ship only cmake-3.16.3.
I suggest not going any further than that.

atsju, 2021-07-07:
Not reproduced, no activity. Closing.

closed 2021-07-07 by atsju