ObKo/stm32-cmake #180: Better "Usage" section documentation

KKoovalsky, 2021-01-17:
Hello!

Firstly, I would like to point out that it's really great that this project defines the compile targets in a proper way being a CMake targets. This is very convenient to use.

To the point: before I was able to call that without errors triggered by CMake:

```
find_package(CMSIS COMPONENTS STM32L432KC REQUIRED)
```

I had to write some CMake boilerplate which is:

```
# Set variables used internally by ObKo/stm32-cmake
set(toolchain_path path/to/the/toolchain)
set(TOOLCHAIN_PREFIX ${toolchain_path})
set(STM32_TOOLCHAIN_PATH ${toolchain_path})

# Download and populate ObKo/stm32-cmake
include(FetchContent)
FetchContent_Declare(
    obko_stm32cmake
    GIT_REPOSITORY https://github.com/ObKo/stm32-cmake.git
    GIT_TAG 49f8f8a6e38592e072ba8165007444331d49b642
)
FetchContent_MakeAvailable(obko_stm32cmake)

# Give CMake access to "Find*" modules from ObKo/stm32-cmake
FetchContent_GetProperties(obko_stm32cmake SOURCE_DIR obko_stm32cmake_source_dir)
list(APPEND CMAKE_MODULE_PATH ${obko_stm32cmake_source_dir}/cmake)

# More variable setting used inside ObKo/stm32-cmake 
set(STM32_CMAKE_DIR ${obko_stm32cmake_source_dir}/cmake)

# Without that call to "FindPackage(CMSIS ..." will not work
include(${obko_stm32cmake_source_dir}/cmake/stm32/common.cmake)

# Utilities to be used explicitly
include(${obko_stm32cmake_source_dir}/cmake/stm32/utilities.cmake)

# Download core CMSIS and set related variables for ObKo/stm32-cmake

include(FetchContent)
FetchContent_Declare(
    stm32_cmsis_core
    GIT_REPOSITORY https://github.com/STMicroelectronics/cmsis_core.git
    GIT_TAG v5.6.0
)
FetchContent_MakeAvailable(stm32_cmsis_core)
FetchContent_GetProperties(stm32_cmsis_core SOURCE_DIR stm32_cmsis_core_source_dir)

set(STM32_CMSIS_PATH ${stm32_cmsis_core_source_dir})

# Download CMSIS for STM32L4 and set related variables for ObKo/stm32-cmake 

include(FetchContent)
FetchContent_Declare(
    stm32l4_cmsis
    GIT_REPOSITORY https://github.com/STMicroelectronics/cmsis_device_l4.git
    GIT_TAG v1.7.0
)
FetchContent_MakeAvailable(stm32l4_cmsis)
FetchContent_GetProperties(stm32l4_cmsis SOURCE_DIR stm32l4_cmsis_source_dir)

set(STM32_CMSIS_L4_PATH ${stm32l4_cmsis_source_dir})

```

Although the steps to download CMSIS packages might be unnecessary, because `ObKo-stm32cmake` provides utilities for that, the rest is mandatory to make the `find_package(CMSIS ...` work.

The question is: am I missing something regarding the setup, or is the documentation missing the details?

Hish15, 2021-01-20:
It does not have to be that complicated :)

If you take a look at the examples for instance the one [with fetches](https://github.com/ObKo/stm32-cmake/blob/master/examples/fetch-cmsis-hal/CMakeLists.txt).

You don't have to use the fetching functions, here are some ways to handle HAL :
 - You can clone/download from ST's github page to /opt/
 - You can submodule/download STM32Cube<FAMILY> directly into your project
 
 If you put it in /opt/ stm32-cmake will work automagically. If you don't, it's just one constant to give the path to STM32Cube!
 
 Let me now if it makes sense to you, and if you have'nt, have a look at the examples.

atsju, 2021-06-21:
Hello @KKoovalsky.
Thank you for your report. You may have a better view of the repository now after some time elapsed. Are the examples and especially the one pointed by @Hish15 enough ? If not, could you precise what exactly is missing to start with the repo ?

atsju, 2021-07-07:
No activity. Closing. 
Do not hesitate to open the issue again if it has not been answered.

closed 2021-07-07 by atsju