ObKo/stm32-cmake #181: Set CMAKE_CXX_FLAGS_RELEASE ？

plumewind, 2021-01-18:
hello，

Thank you for your contribution.
I'm very comfortable with version 1.

Recently, I started to try your version 2, but I encountered a big problem:
`** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20004ffc
Error: checksum mismatch - attempting binary compare
diff 0 address 0x00000000. Was 0xff instead of 0x7f
diff 1 address 0x00000001. Was 0x4f instead of 0x45
diff 2 address 0x00000002. Was 0x00 instead of 0x4c
diff 3 address 0x00000003. Was 0x20 instead of 0x46
......
diff 126 address 0x0000009b. Was 0x08 instead of 0x00
diff 127 address 0x0000009c. Was 0x95 instead of 0x00
More than 128 errors, the rest are not printed.
** Verify Failed **
shutdown command invoked`

I use the same code, version 1 works.
Then I found that version 2 didn't set the compile flag as version 1 did.
Then I commented out the settings in version 1, and the same problem occurred.
`# SET(CMAKE_C_FLAGS_DEBUG "-Og -g" CACHE INTERNAL "c compiler flags debug")
# SET(CMAKE_CXX_FLAGS_DEBUG "-Og -g" CACHE INTERNAL "cxx compiler flags debug")
# SET(CMAKE_ASM_FLAGS_DEBUG "-g" CACHE INTERNAL "asm compiler flags debug")
# SET(CMAKE_EXE_LINKER_FLAGS_DEBUG "-Xlinker -Map=output.map" CACHE INTERNAL "linker flags debug")

# SET(CMAKE_C_FLAGS_RELEASE "-Os -flto" CACHE INTERNAL "c compiler flags release")
# SET(CMAKE_CXX_FLAGS_RELEASE "-Os -flto" CACHE INTERNAL "cxx compiler flags release")
# SET(CMAKE_ASM_FLAGS_RELEASE "" CACHE INTERNAL "asm compiler flags release")
# SET(CMAKE_EXE_LINKER_FLAGS_RELEASE "-Xlinker -Map=output.map -s -flto" CACHE INTERNAL "linker flags release")`

However, I copy and paste these settings into STM32 cmake / cmake / STM32_ gcc.cmake The same problem still appears, as if what I pasted is useless.

Does anyone know what to do? Thank you very much.

Hish15, 2021-01-20:
Hello @plumewind,

What you call version 2 is actually the "modern" cmake approach.
The idea is, use target to specify requirements on objects, then make those targets work together.

If you follow the explanation on [README.md](https://github.com/ObKo/stm32-cmake/blob/master/README.md),
you will notice that you have a target, with all the cool flags you need : 
`CMSIS::STM32::<FAMILY> (e.g. CMSIS::STM32::F4) - common includes, compiler flags and defines for family`
by linking the target to your main target you will be all set flag-wise.

FYI here is the bit of code that handle the build flags :
https://github.com/ObKo/stm32-cmake/blob/49f8f8a6e38592e072ba8165007444331d49b642/cmake/stm32/utilities.cmake#L22-L27

plumewind, 2021-01-22:
Thanks, it was written in the cmakelist, some like this:
```
# STM32F1-Discovery
add_executable(${CMAKE_PROJECT_NAME}.elf ${PROJECT_SOURCES})
target_link_libraries(${CMAKE_PROJECT_NAME}.elf PUBLIC -lc -lm
    HAL::STM32::F1::RCC
    HAL::STM32::F1::GPIO
    HAL::STM32::F1::CORTEX
    HAL::STM32::F1::DMA
    HAL::STM32::F1::FLASH
    HAL::STM32::F1::I2C
    HAL::STM32::F1::TIM
    HAL::STM32::F1::SPI
    HAL::STM32::F1::PWR
    HAL::STM32::F1::UART
    HAL::STM32::F1::IWDG
    CMSIS::STM32::F103xB
	STM32::NoSys 
)
target_compile_definitions(${CMAKE_PROJECT_NAME}.elf
	PUBLIC -D${STM32_CHIP}
	PUBLIC USE_HAL_DRIVER
	)
```

So, it is mean that the build flags is add secretly？
But, the contrast experiment operation I mentioned above, this step is done before.

In other words, the problems I mentioned will still arise, which may be other problems? Do you have any suggestions?
Thank you very much.

atsju, 2021-01-31:
Hi @plumewind 
There are some default build flags set by cmake (-O0 or -O2 for example) but you can also set your own as @Hish15 mentioned. You can look exactly at the executed build commands in /build folder files.

Now about the problem you try to describe. It is very difficult to understand your problem. 

>  I encountered a big problem
> ** Verify Failed **

This is not precise enough for anyone to help you. I suppose your code compiles but you get a runtime error but you do not say what your code is expected to do or anything that could help us understand. 

If you believe this is an issue with this repo then I suppose you should post a minimal code example (for example a dedicated MINIMAL repository) that reproduces the issue. With a precise description of what are the expected outcomes, a description of the failure and precise steps to reproduce it.

plumewind, 2021-02-04:
Thank you. I'll sort out my questions carefully.

plumewind, 2021-02-25:
hello,

I test a experiment about usart, also get the paoblem.
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.241107
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x08008ee0 msp: 0x20004fe4
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800d250 msp: 0x20004ffc
** Programming Started **
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 64kbytes
Warn : no flash bank found for address 0
Warn : no flash bank found for address 1fc5c
wrote 0 bytes from file USART_TEST.elf in 0.001673s (0.000 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20004ffc
Error: checksum mismatch - attempting binary compare
diff 0 address 0x00000000. Was 0xff instead of 0x7f
diff 1 address 0x00000001. Was 0x4f instead of 0x45
diff 2 address 0x00000002. Was 0x00 instead of 0x4c
diff 3 address 0x00000003. Was 0x20 instead of 0x46
diff 4 address 0x00000004. Was 0x51 instead of 0x01
diff 5 address 0x00000005. Was 0xd2 instead of 0x01
diff 6 address 0x00000006. Was 0x00 instead of 0x01
diff 7 address 0x00000007. Was 0x08 instead of 0x00
diff 8 address 0x00000008. Was 0x7b instead of 0x00
diff 9 address 0x00000009. Was 0x8e instead of 0x00
diff 10 address 0x0000000b. Was 0x08 instead of 0x00
diff 11 address 0x0000000c. Was 0x7d instead of 0x00
diff 12 address 0x0000000d. Was 0x8e instead of 0x00
diff 13 address 0x0000000f. Was 0x08 instead of 0x00
diff 14 address 0x00000010. Was 0x8d instead of 0x02
diff 15 address 0x00000011. Was 0x8e instead of 0x00
diff 16 address 0x00000012. Was 0x00 instead of 0x28
diff 17 address 0x00000013. Was 0x08 instead of 0x00
diff 18 address 0x00000014. Was 0x8f instead of 0x01
diff 19 address 0x00000015. Was 0x8e instead of 0x00
diff 20 address 0x00000017. Was 0x08 instead of 0x00
diff 21 address 0x00000018. Was 0x91 instead of 0xc9
diff 22 address 0x00000019. Was 0x8e instead of 0x80
diff 23 address 0x0000001b. Was 0x08 instead of 0x00
diff 24 address 0x0000001c. Was 0x00 instead of 0x34
diff 25 address 0x00000020. Was 0x00 instead of 0x24
diff 26 address 0x00000021. Was 0x00 instead of 0xcd
diff 27 address 0x00000022. Was 0x00 instead of 0x06
diff 28 address 0x00000025. Was 0x00 instead of 0x02
diff 29 address 0x00000027. Was 0x00 instead of 0x05
diff 30 address 0x00000028. Was 0x00 instead of 0x34
diff 31 address 0x0000002a. Was 0x00 instead of 0x20
diff 32 address 0x0000002c. Was 0x93 instead of 0x03
diff 33 address 0x0000002d. Was 0x8e instead of 0x00
[... 77 lines cut ...]
diff 111 address 0x00000090. Was 0x99 instead of 0x00
diff 112 address 0x00000091. Was 0xd2 instead of 0x00
diff 113 address 0x00000092. Was 0x00 instead of 0x01
diff 114 address 0x00000093. Was 0x08 instead of 0x00
diff 115 address 0x00000094. Was 0x99 instead of 0x00
diff 116 address 0x00000095. Was 0xd2 instead of 0x00
diff 117 address 0x00000097. Was 0x08 instead of 0x00
diff 118 address 0x00000098. Was 0x99 instead of 0x00
diff 119 address 0x00000099. Was 0xd2 instead of 0x00
diff 120 address 0x0000009b. Was 0x08 instead of 0x00
diff 121 address 0x0000009c. Was 0x95 instead of 0x00
diff 122 address 0x0000009d. Was 0x8f instead of 0x00
diff 123 address 0x0000009f. Was 0x08 instead of 0x00
diff 124 address 0x000000a0. Was 0x99 instead of 0x00
diff 125 address 0x000000a1. Was 0xd2 instead of 0x00
diff 126 address 0x000000a3. Was 0x08 instead of 0x00
diff 127 address 0x000000a4. Was 0x99 instead of 0x00
More than 128 errors, the rest are not printed.
** Verify Failed **
shutdown command invoked
```
Here is the project.
[test13 usart.zip](https://github.com/ObKo/stm32-cmake/files/6042483/test13.usart.zip)

Thanks a lot.

plumewind, 2021-02-25:
and, I test it in version 1 by CMakeLists.txt:
[CMakeLists.txt](https://github.com/ObKo/stm32-cmake/files/6042553/CMakeLists.txt)

It is ok.
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.256122
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x08008ee8 msp: 0x20004fe4
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800d250 msp: 0x20004ffc
** Programming Started **
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 64kbytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20004ffc
wrote 35840 bytes from file USART_TEST.elf in 1.972455s (17.744 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20004ffc
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20004ffc
verified 35084 bytes in 0.533316s (64.243 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

plumewind, 2021-03-05:
Hello， @atsju 

I had uploaded the exmaple code.

If you are free any time, can I help you? 

Thank you very much.

atsju, 2021-03-06:
I noticed quite fast you had a linking error because memory addresses didn't fit. You could have noticed that during flashing.

Its `CMSIS::STM32::F103RB` (or what is your chip) and not `CMSIS::STM32::F103xB`. Once this is changed, you will also need to add compilation of DMA to make it build. Should work.

plumewind, 2021-03-12:
Hello!

@atsju  Thank you very much. Actually, I guess it's the problem. But I can't compare V1 with v2.

The two codes I uploaded both use the same chip, stm32f103t8u6. Is memory addresses changed in version V2?

atsju, 2021-03-12:
Sorry I haven't been clear. The stm32-cmake version has evolved and you cannot expect the V1 cmakelist you used to work with V2 without any modification.
You need to replace `CMSIS::STM32::F103xB` with `CMSIS::STM32::F103RB` the `x` is no longer allowed I suppose. 

And an additional remark : if your chip is stm32f103t8u6 why were you using F103xB ? this is an other source of problems waiting to happen... You shall use F103T8.

Hish15, 2021-03-12:
The new approach is way simpler actually. I rewrote your cmakeList, without testing.
To show you the principle of target, I created a target for the hal and a target for your "executable"
you can see how this allow to keep things grouped in a more logical way.

Linking with hal_lib, you have access to the relevant C files, the include dirs, and even flags!

```
# define chip used in this project, this set must define before project definition
# for this project dont use cmake commandline option -DSTM32_CHIP=<chip>

SET(STM32_CHIP STM32F103RB)
SET(STM32_CUBE_F1_PATH /home/software/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/cmake/stm32_gcc.cmake)

cmake_minimum_required(VERSION 3.13)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

PROJECT(USART_TEST C ASM)
set(CMAKE_INCLUDE_CURRENT_DIR TRUE)

find_package(CMSIS COMPONENTS STM32F1 REQUIRED)
find_package(HAL COMPONENTS STM32F1 RCC GPIO CORTEX PWR UART REQUIRED)

add_library(hal_lib INTERFACE)
target_link_libraries(hal_lib PUBLIC
    HAL::STM32::F1::RCC
    HAL::STM32::F1::GPIO
    HAL::STM32::F1::CORTEX
    HAL::STM32::F1::PWR
    HAL::STM32::F1::UART
    CMSIS::STM32::F103RB
    STM32::NoSys
    )

add_executable(${CMAKE_PROJECT_NAME}.elf
    Src/main.c
    Src/stm32f1xx_hal_msp.c
    Src/stm32f1xx_it.c

    Src/hardware/led.c
    Src/hardware/key.c

    Src/system/delay.c
    Src/system/sys.c
    Src/system/usart.c
   )

target_include_directories(${CMAKE_PROJECT_NAME}.elf PRIVATE
    Inc
    Inc/hardware
    Inc/system
    )

target_link_libraries(${CMAKE_PROJECT_NAME}.elf PRIVATE
    -lc -lm #Not sure you need that
    hal_lib)
```

plumewind, 2021-03-13:
Hello,

I've debugged them all successfully.

Thank you very much for your help. Thank you.

The reason is that I still use the configuration of version 1. I don't know that your version 2 has changed so much. Now I have changed the previous configuration.
```
# define chip used in this project, this set must define before project definition
# for this project dont use cmake commandline option -DSTM32_CHIP=<chip>
SET(STM32_CUBE_F1_PATH /home/software/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/cmake/stm32_gcc.cmake)

cmake_minimum_required(VERSION 3.13)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

PROJECT(USART_TEST C ASM)
set(CMAKE_INCLUDE_CURRENT_DIR TRUE)

# set the build type
if(NOT CMAKE_BUILD_TYPE) 
	# set(CMAKEBUILD_TYPE Release)
	 set(CMAKE_BUILD_TYPE Debug)
endif(NOT CMAKE_BUILD_TYPE)

find_package(CMSIS COMPONENTS STM32F1 REQUIRED)
find_package(HAL COMPONENTS STM32F1 RCC GPIO CORTEX PWR UART DMA REQUIRED)

INCLUDE_DIRECTORIES(
    ${CMAKE_CURRENT_SOURCE_DIR}
    ${CMSIS_INCLUDE_DIRS}
    ${STM32HAL_INCLUDE_DIR}
	${CMAKE_CURRENT_SOURCE_DIR}/Inc
	${CMAKE_CURRENT_SOURCE_DIR}/Inc/hardware
	${CMAKE_CURRENT_SOURCE_DIR}/Inc/system
)

SET(PROJECT_SOURCES
	Src/main.c
	Src/stm32f1xx_hal_msp.c
	Src/stm32f1xx_it.c
    	
	Src/hardware/led.c
	Src/hardware/key.c

	Src/system/delay.c
	Src/system/sys.c
	Src/system/usart.c
)

# STM32F1-Discovery
add_executable(${CMAKE_PROJECT_NAME} ${PROJECT_SOURCES})
target_link_libraries(${CMAKE_PROJECT_NAME} PUBLIC -lc -lm
    HAL::STM32::F1::RCC
    HAL::STM32::F1::GPIO
    HAL::STM32::F1::CORTEX
    HAL::STM32::F1::PWR
    HAL::STM32::F1::DMA
    HAL::STM32::F1::UART
    CMSIS::STM32::F103T8
        STM32::NoSys 
)
target_compile_definitions(${CMAKE_PROJECT_NAME}
	PUBLIC -D${STM32_CHIP}
	PUBLIC USE_HAL_DRIVER
	)
```

This is a great project, which has greatly promoted the development of embedded system. It's very cool!

closed 2021-03-13 by plumewind

Hish15, 2021-03-13:
@plumewind, glad it works now!
I agree this project is really enabling for embedded system.

That being said, even if your  exemple works now, I suggest you have a look at my suggestion above :)

 - You do not use the power of targets. When you use target_link_libraries with CMSIS::STM32::F103T8 you no longer need to include any HAL include folder, compilation flags, or source files related to the HAL/CMSIS. You are doing the work twice here.
 - CMAKE_BUILD_TYPE is already Debug by default no needed to handle that.
 - [INCLUDE_DIRECTORIES](https://cmake.org/cmake/help/v3.13/command/include_directories.html) must not be used, prefer target_include_directories => Thats a big deal for the target approach.

Why I insist :
On a big project, you can split layers or modules using cmake's targets. You limit risks isolating things logically :)

Keep the good work!