ObKo/stm32-cmake #185: Add support for using custom startup assembly file

smokeythebandit, 2021-04-07:
I'm using the STM32H7VIT6 and I would like to use RAM_D2, which needs to be initializes prior to me using it. I would like to add a section in the template startup assembly file. Because of this I have a need to use a non-standard startup assembly file. 

Maybe this is already possible and I missed it, in that case; my bad! If not I'd think this would be a great addition to an already very useful project.

Thanks in advance.

Hish15, 2021-04-07:
Hello,

There is a part on the Readme that speaks of the case of a custom linker script.

In your case you can link against  `CMSIS::STM32::<FAMILY>` (`CMSIS::STM32::H7::M7`). This way you will not have the default startup source provided by `CMSIS::STM32::H743xx::M7`, nor the linker script provided by the `CMSIS::STM32::<DEVICE>` target.

atsju, 2021-06-21:
Hello @smokeythebandit,
Has the answer from @Hish15 been able to fix your interrogations ? 
If so, can we close the issue ?

smokeythebandit, 2021-06-24:
Hello, 
Sorry for not closing the issue. 
Used the cmake command: stm32_add_linker_script to load my custom linker script.

closed 2021-06-24 by smokeythebandit