ObKo/stm32-cmake #198: BSP all components library have the same path

xgroleau, 2021-05-14:
All the bsp library have the same source files paths which correspond to the first component since the variable name is the same in all iterations of the for each.  [Here](https://github.com/ObKo/stm32-cmake/blob/86d1dd2d64612b49c4b08f0508423748ddbf88ff/cmake/FindBSP.cmake#L325) is the problem 

You can change `BSP_${BOARD_CANONICAL}_${COMP}_SOURCE` to `BSP_${BOARD_CANONICAL}_${BCOMP}_SOURCE` and it should fix the problem

atsju, 2021-05-15:
Hi @xgroleau,
Thank you for this issue.
Maybe could you provide a pull request for the fix ?

And for my personal interest, could you describe the problem further ? From what I see in the code `BSP_${BOARD_CANONICAL}_${COMP}_SOURCE` is only an intermediate variable. It could be named `hello` and have the same effect. Were am I wrong ?

xgroleau, 2021-05-16:
@atsju Hi, I've created a pull request with the fix, you can see it [here](https://github.com/ObKo/stm32-cmake/pull/199)

closed 2021-06-24 by atsju