ObKo/stm32-cmake #200: [Question] Ram size not the same as product

xgroleau, 2021-05-16:
This is more of a question than a bug, but why is the ram size not the same as the product? For example, the [h735gz has a ram size of 128k in the cmake files](https://github.com/ObKo/stm32-cmake/blob/86d1dd2d64612b49c4b08f0508423748ddbf88ff/cmake/stm32/h7.cmake#L11) while in the product description it has [564 kb of ram](https://www.st.com/en/microcontrollers-microprocessors/stm32h725-735.html#overview) composed in 128 kb of TCM, 432 of system ram and 4 kb of backup. Is there a way to use the available ram? When compiling we use the `--print-memory-usage` so we can see our usage, as of now we almost bust our ram since most of the ram is not available/used.

atsju, 2021-05-16:
Hi @xgroleau 
Indeed it's a good question and I worked in harmonization some month ago so I can try to answer this.
Please have a look at #165 and #164.
I opened every single datasheet of every families and in the case of H7 family there are many different RAM sections where some of them can be remapped or not but all do not have the same properties. 
My approach was following these rules : 
 - the repo should work seamless for most user to start an application
 - follow the ST examples
 - the advanced user should be able to do whatever he wants
 - the limitations should appear clearly

So I looked at ST linker scripts here (all are not present for GCC but IAR is more complete): https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates
And most basic examples use just 128kB so I did exactly the same for this repo. 
I could have used more RAM by making use of remapping but I think this may lead to side effects for some beginner and I prefer that there is a clear error (nor enough RAM here). Moreover the sections are disjoint without remapping and in this case it's impossible to make use of them without explicit source code modification by programmer (it's not transparent use) so I just decided to not create this section.
Now if an advanced user really needs to use more than the 128kB in auto-generated linker script I think best is to write/adapt their own linker script and link against it.

xgroleau, 2021-05-16:
Thanks that answers my question very well thank you! I will write my own linker file and manage sections to my need

closed 2021-05-16 by xgroleau