ObKo/stm32-cmake #211: Bump minimum CMake version to 3.18 to improve find_program()?

gudnimg, 2021-06-13:
With CMake 3.18 there is a `REQUIRED` option for `find_program()`

Example: If the `arm-none-eabi-gcc` is not found, the CMake will stop processing and create an error message.

**CMake >= 3.18**:
```cmake
find_program(CMAKE_C_COMPILER NAMES ${STM32_TARGET_TRIPLET}-gcc PATHS ${TOOLCHAIN_BIN_PATH} REQUIRED)
```
**CMake < 3.18**:
```cmake
find_program(CMAKE_C_COMPILER NAMES ${STM32_TARGET_TRIPLET}-gcc PATHS ${TOOLCHAIN_BIN_PATH})
```

rmspacefish, 2021-06-16:
This might also be worth an own issue.. There are also some calls to find utilities in the `common.cmake` file:

```cmake
find_program(CMAKE_OBJCOPY NAMES ${STM32_TARGET_TRIPLET}-objcopy PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_OBJDUMP NAMES ${STM32_TARGET_TRIPLET}-objdump PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_SIZE NAMES ${STM32_TARGET_TRIPLET}-size PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_DEBUGGER NAMES ${STM32_TARGET_TRIPLET}-gdb PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_CPPFILT NAMES ${STM32_TARGET_TRIPLET}-c++filt PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
```

I am used to add the toolchain binary path to the system path and then simply look for `arm-none-eabi-XYZ` instead of setting a specific variable like `STM32_TOOLCHAIN_PATH` . Then again, the later one allows multiple toolchain (although I guess this is not really relevant for most users)..  I don't know the exact reasoning behind the `NO_DEFAULT_PATH` option, but maybe it might be a good idea to omit that option. Those utilities are not really required, but every toolchain should have them, so maybe `REQUIRED` can be added there as well.

gudnimg, 2021-06-17:
We could also do this to preserve CMake 3.13 compatibility:

```cmake
find_program(CMAKE_C_COMPILER NAMES ${STM32_TARGET_TRIPLET}-gcc PATHS ${TOOLCHAIN_BIN_PATH})

if (NOT CMAKE_C_COMPILER)
    message(FATAL_ERROR "${STM32_TARGET_TRIPLET}-gcc not found at ${TOOLCHAIN_BIN_PATH}.")
endif()
```

gudnimg, 2021-06-17:
Here's an example of what  the warning looks like. I think having a `WARNING` is better than nothing. I'll create an PR for this :) [image]

atsju, 2021-06-20:
> This might also be worth an own issue.. There are also some calls to find utilities in the `common.cmake` file:
> 
> ```cmake
> find_program(CMAKE_OBJCOPY NAMES ${STM32_TARGET_TRIPLET}-objcopy PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
> find_program(CMAKE_OBJDUMP NAMES ${STM32_TARGET_TRIPLET}-objdump PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
> find_program(CMAKE_SIZE NAMES ${STM32_TARGET_TRIPLET}-size PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
> find_program(CMAKE_DEBUGGER NAMES ${STM32_TARGET_TRIPLET}-gdb PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
> find_program(CMAKE_CPPFILT NAMES ${STM32_TARGET_TRIPLET}-c++filt PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
> ```
> 
> I am used to add the toolchain binary path to the system path (temporarily) and then simply look for `arm-none-eabi-XYZ` instead of setting a specific variable like `STM32_TOOLCHAIN_PATH` . CMake caches the path at build generation so the environment stays unpolluted after build generation has been done once. The later one allows multiple toolchains (although I guess this is not really relevant for most users).. I don't know the exact reasoning behind the `NO_DEFAULT_PATH` option, but maybe it might be a good idea to omit that option. Those utilities are not really required, but every toolchain should have them, so maybe `REQUIRED` can be added there as well.

Hello @rmspacefish 
As you wrote yourself, please consider opening a dedicated issue for this. It's not directly linked to original issue and will cause confusion.

atsju, 2021-06-20:
Hello @gudnimg
I think keeping the 3.13 compatibility is better option as it is the default distributed version in many Linux distributions. Your proposal to get a warning instead error seems good but no `arm-none-eabi-gcc` shall create an error. Only `-gdb` is optional and should get a warning. 

Please consider doing a pull request doing the modification and referencing the issue.

gudnimg, 2021-06-21:
@atsju Thanks for the reply. Your comment about 3.13 makes a lot of sense now (I’ve seen other repos also use 3.13 as minimum)

I will submit a PR. It may be today or later this week.  😊

atsju, 2021-06-26:
@gudnimg I looked at your PR #218.
For my understanding, could you explain why you needed arm-none-eabi-gdb ? I mean this is not directly linked to cmake so it's great to have a warning if it's not detected but did you really use the (-NOTFOUND) variable `CMAKE_DEBUGGER` or you needed gdb externally of cmake ?

gudnimg, 2021-06-26:
@atsju I use it externally. Though now that you mention it... does the repo need to find `arm-none-eabi-gdb`? It would be nice if one could use CMake to flash the STM32 chip with custom commands :)

atsju, 2021-06-26:
> does the repo need to find arm-none-eabi-gdb?

This is the point. But you proved that someone will find an use so we can keep it. But maybe it should be `INFO` instead `WARNING`? And then you should `MESSAGE` for each non-mandatory `find_package` not only gdb? 

I let you think about it and update the PR.