ObKo/stm32-cmake #240: Default STM32_CUBE_family_PATH is not being set

BenArtes, 2021-07-22:
**Describe the bug**
After a clean clone of this repository, I installed the STM32CubeF4 drivers to /opt/STM32CubeF4 and tried to build the blinky example. Instead of using the default path I was met with an error indicating that a blank path was being used and that CMSIS could NOT be found:

```bash
-- No STM32_TOOLCHAIN_PATH specified, using default: /usr
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The C compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default  STM32_CUBE_F4_PATH: 
CMake Error at /usr/local/lib/python3.6/dist-packages/cmake/data/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CMSIS (missing: CMSIS_INCLUDE_DIRS CMSIS_SOURCES STM32F4)
  (found version "v1.0.4")
Call Stack (most recent call first):
  /usr/local/lib/python3.6/dist-packages/cmake/data/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/terminus/stm32-cmake/cmake/FindCMSIS.cmake:183 (find_package_handle_standard_args)
  CMakeLists.txt:30 (find_package)
```

In FindCMSIS.cmake lines [74-81](https://github.com/ObKo/stm32-cmake/blob/cebbda690db674885cf0dca2a90dd8f4426d40d0/cmake/FindCMSIS.cmake#L74) there are two attempts to set `STM32_CUBE_${family}_PATH`, first from an ENV variable, then from the default value. Both use cmake's `set(... CACHE PATH)` and I'm wondering if the first attempt to set from the ENV variable is preventing the default from succeeding.

Two different modifications to this file worked, but I'm not sure which is preferred:

### 1) Checking ENV variable before attempting to set STM32_CUBE_${FAMILY}_PATH

```cmake
    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH) AND ($ENV{STM32_CUBE_${FAMILY}_PATH}))
        set(STM32_CUBE_${FAMILY}_PATH $ENV{STM32_CUBE_${FAMILY}_PATH} CACHE PATH "Path to STM32Cube${FAMILY}")
    endif()

    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH /opt/STM32Cube${FAMILY} CACHE PATH "Path to STM32Cube${FAMILY}")
        message(STATUS "Neither STM32_CUBE_${FAMILY}_PATH nor STM32_CMSIS_${FAMILY}_PATH specified using default  STM32_CUBE_${FAMILY}_PATH: ${STM32_CUBE_${FAMILY}_PATH}")
    endif()
```

### 2) Forcing Default Setting using set's FORCE optional

```cmake
    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH $ENV{STM32_CUBE_${FAMILY}_PATH} CACHE PATH "Path to STM32Cube${FAMILY}")
    endif()

    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH /opt/STM32Cube${FAMILY} CACHE PATH "Path to STM32Cube${FAMILY}" FORCE)
        message(STATUS "Neither STM32_CUBE_${FAMILY}_PATH nor STM32_CMSIS_${FAMILY}_PATH specified using default  STM32_CUBE_${FAMILY}_PATH: ${STM32_CUBE_${FAMILY}_PATH}")
    endif()
```

A similar code is located in FindHAL.cmake [Line74-81](https://github.com/ObKo/stm32-cmake/blob/cebbda690db674885cf0dca2a90dd8f4426d40d0/cmake/FindHAL.cmake#L240) and FinaBSP.cmake [Lines 265-272](https://github.com/ObKo/stm32-cmake/blob/cebbda690db674885cf0dca2a90dd8f4426d40d0/cmake/FindBSP.cmake#L265), though these don't seem to cause the same issue.

I'm happy to create a pull request with these changes once I know which files to change and the preferred solution.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Compiler: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
 - stm32-cmake: https://github.com/ObKo/stm32-cmake/commit/cebbda690db674885cf0dca2a90dd8f4426d40d0

atsju, 2021-07-22:
Thank you for reporting this issue. It may be linked to a recent merge of #226. I had no issue running on windows.
Could you please print (message) the value of STM32_CUBE_${FAMILY}_PATH at 3 points (before first if, after first if and at the end) ? I expect path to be something in format `-NOT-FOUND`.

just for notification: @rmspacefish

BenArtes, 2021-07-22:
Message Changes to cmake:
```cmake
    message("Before First If: " ${STM32_CUBE_${FAMILY}_PATH})
    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH $ENV{STM32_CUBE_${FAMILY}_PATH} CACHE PATH "Path to STM32Cube${FAMILY}")
    endif()

    message("After First If: " ${STM32_CUBE_${FAMILY}_PATH})
    if((NOT STM32_CMSIS_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH /opt/STM32Cube${FAMILY} CACHE PATH "Path to STM32Cube${FAMILY}" )
        message(STATUS "Neither STM32_CUBE_${FAMILY}_PATH nor STM32_CMSIS_${FAMILY}_PATH specified using default  STM32_CUBE_${FAMILY}_PATH: ${STM32_CUBE_${FAMILY}_PATH}")
    endif()
    message("End: " ${STM32_CUBE_${FAMILY}_PATH})
```

Messages Without Fixes:
```bash
Before First If: 
After First If: 
-- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default  STM32_CUBE_F4_PATH: 
End: 
```

Messages With Fixes:
```bash
Before First If: 
After First If: 
-- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default  STM32_CUBE_F4_PATH: /opt/STM32CubeF4
End: /opt/STM32CubeF4
```

atsju, 2021-07-22:
Thank you very much. 
Please do the PR with `FORCE` solution. Also have a look at #226 and add the force to all relevant places.

rmspacefish, 2021-07-22:
Hmm, an empty value is set if the environmental variable is not defined. It might be a good idea to only set from ENV if the env variable is actually defined. It might also be a good idea to check whether `/opt/STM32Cube${FAMILY}` exists as a directory as mentioned in #232

atsju, 2021-07-22:
@BenArtes  
I think @rmspacefish is right and it was your proposal number 1 but I did read too fast. 
Could you please use the [defined env](https://cmake.org/cmake/help/v3.16/variable/ENV.html) format if it's not too much asking for this modification. 

Side note about the PR: I think you misconfigured your email address localy. Github doesn't see "you" as author of commit.
[image]

BenArtes, 2021-07-22:
Updated the PR, to use defined envs, and check path for existence.

Also updated local email address.

PR #244 (2021-08-01): Fixes default paths with empty ENVs, and Checks paths for existence.
Closes #240, by checking ENV path exists before setting it
closed 2021-08-01 by Hish15