ObKo/stm32-cmake #241: Blinky Example doesn't compile due to lack of LL Drivers

BenArtes, 2021-07-22:
**Describe the bug**
When attempting to build the `blinky` example I receive the following error.

```bash
-- No STM32_TOOLCHAIN_PATH specified, using default: /usr
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The C compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Neither STM32_CUBE_F4_PATH nor STM32_CMSIS_F4_PATH specified using default  STM32_CUBE_F4_PATH: /opt/STM32CubeF4
-- Found CMSIS: /opt/STM32CubeF4/Drivers/CMSIS/Include;/opt/STM32CubeF4/Drivers/CMSIS/Device/ST/STM32F4xx/Include (found version "5.4.0") found components:  STM32F4 
CMake Error at /home/terminus/stm32-cmake/cmake/FindHAL.cmake:220 (list):
  list sub-command REMOVE_DUPLICATES requires list to be present.
Call Stack (most recent call first):
  CMakeLists.txt:31 (find_package)

```

I presume that the lack of LL drivers in the blinky example causes `  list(REMOVE_DUPLICATES HAL_FIND_COMPONENTS_DRIVERS_LL)` to fail.

I commented out the code as a test and then attempted to build blinky-ll and ran into similar errors related to `HAL_FIND_COMPONENTS_DRIVERS`.

I modified FindHAL.cmake to have checks surrounding the relevant blocks as below and all examples built correctly.

If this is a correct / desirable change I'm happy to create a pull request.

```cmake
if (HAL_FIND_COMPONENTS_DRIVERS)
list(REMOVE_DUPLICATES HAL_FIND_COMPONENTS_DRIVERS)
endif()
if (HAL_FIND_COMPONENTS_DRIVERS_LL)
  list(REMOVE_DUPLICATES HAL_FIND_COMPONENTS_DRIVERS_LL)
endif()

message(STATUS "Search for HAL families: ${HAL_FIND_COMPONENTS_FAMILIES}")
if(HAL_FIND_COMPONENTS_DRIVERS)
message(STATUS "Search for HAL drivers: ${HAL_FIND_COMPONENTS_DRIVERS}")
endif()
if (HAL_FIND_COMPONENTS_DRIVERS_LL)
message(STATUS "Search for HAL LL drivers: ${HAL_FIND_COMPONENTS_DRIVERS_LL}")
endif()

foreach(COMP ${HAL_FIND_COMPONENTS_FAMILIES})
    string(TOUPPER ${COMP} COMP_U)
    
    string(REGEX MATCH "^STM32([A-Z][0-9])([0-9A-Z][0-9][A-Z][0-9A-Z])?_?(M[47])?.*$" COMP_U ${COMP_U})    
    if(CMAKE_MATCH_3)
        set(CORE ${CMAKE_MATCH_3})
        set(CORE_C "::${CORE}")
        set(CORE_U "_${CORE}")
    else()
        unset(CORE)
        unset(CORE_C)
        unset(CORE_U)
    endif()
        
    set(FAMILY ${CMAKE_MATCH_1})
    string(TOLOWER ${FAMILY} FAMILY_L)
    
    if((NOT STM32_HAL_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
        set(STM32_CUBE_${FAMILY}_PATH $ENV{STM32_CUBE_${FAMILY}_PATH} CACHE PATH "Path to STM32Cube${FAMILY}")
    endif()

    if((NOT STM32_HAL_${FAMILY}_PATH) AND (NOT STM32_CUBE_${FAMILY}_PATH))
[... 121 lines cut ...]
            continue()
        endif()
    
        set(HAL_${DRV_COMP}_FOUND TRUE)
        if(HAL_${FAMILY}${CORE_U}_${DRV}_LL_SOURCE AND (NOT (TARGET HAL::STM32::${FAMILY}${CORE_C}::LL_${DRV})))
            add_library(HAL::STM32::${FAMILY}${CORE_C}::LL_${DRV} INTERFACE IMPORTED)
            target_include_directories(HAL::STM32::${FAMILY}${CORE_C}::LL_${DRV} INTERFACE "${HAL_${FAMILY}${CORE_U}_INCLUDE}")
            target_sources(HAL::STM32::${FAMILY}${CORE_C}::LL_${DRV} INTERFACE "${HAL_${FAMILY}${CORE_U}_${DRV}_LL_SOURCE}")
        endif()
    endforeach()
    endif()
    
    set(HAL_${COMP}_FOUND TRUE)
    list(APPEND HAL_INCLUDE_DIRS "${HAL_${FAMILY}${CORE_U}_INCLUDE}")
    list(APPEND HAL_SOURCES "${HAL_${FAMILY}${CORE_U}_SOURCES}")
endforeach()
```

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Compiler: [e.g. arm-none-eabi-gcc +version]
 - stm32-cmake: [v2.0 or 86d1dd2 or ...]

**Additional context**
Add any other context about the problem here.

atsju, 2021-07-22:
Thank you for reporting this issue. 
I confirm blinky is compiling fine so maybe there is something system specific here.

Please do the PR as it will be easier to look at the diffs.

atsju, 2021-07-22:
Ok I see what you mean. We will need to be able to reproduce the issue first.
What version of cmake are you using ? 
What version of HAL are you using ? 
Please confirm you did not modify the blinky example at all ?

BenArtes, 2021-07-22:
Using cmake 3.13.3 in terminal. 

I just tested with cmake 3.19.2 that comes bundled with CLion and the issue does not occur.

HAL is STM32F4 (default).

Blinky was not modified.

BenArtes, 2021-07-22:
This looks like a change in behavior introduced in CMake 3.14: [Release Notes, search for `REMOVE_DUPLICATES`](https://cmake.org/cmake/help/latest/release/3.14.html).

atsju, 2021-07-22:
Well done. I just found same reference and wanted to share link.

atsju, 2021-07-22:
All tests are run an ubuntu 20.04 so after discussion with @Hish15  we will bump minium cmake version to 3.16.
We could fix you issue but because we do not test with cmake 3.13 there could be other masked problems.

BenArtes, 2021-07-22:
Makes sense. Ubuntu 18.04 is actually CMake 3.10. According to [this table](https://gitlab.kitware.com/cmake/community/-/wikis/CMake-Versions-on-Linux-Distros), Pinning at 3.13 really only adds Debian Buster.

atsju, 2021-07-22:
Yes moreover if people were really using 3.13 you should not have been the first person to mention the problem. So I suppose it won't hurt too much.

PR #245 (2021-07-25): update minimum cmake version to 3.16
close #241 
close #242
closed 2021-07-25 by Hish15