ObKo/stm32-cmake #250: Configuring incomplete with fetch example and H7 family (one core)

zisi, 2021-07-28:
**Describe the bug**
I try to run the fetch-cube example but instead of use F4, I use H7. I got this:

```'CMakeLists.txt'
-- No STM32_TOOLCHAIN_PATH specified, using default: /usr
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- The C compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Populating stm32cubeh7
-- Configuring done
-- Generating done
-- Build files have been written to: /home/azisi/Desktop/stm32-cmake/examples/fetch-cube/build/_deps/stm32cubeh7-subbuild
[ 11%] Performing update step for 'stm32cubeh7-populate'
[ 22%] No patch step for 'stm32cubeh7-populate'
[ 33%] No configure step for 'stm32cubeh7-populate'
[ 44%] No build step for 'stm32cubeh7-populate'
[ 55%] No install step for 'stm32cubeh7-populate'
[ 66%] No test step for 'stm32cubeh7-populate'
[ 77%] Completed 'stm32cubeh7-populate'
[100%] Built target stm32cubeh7-populate
-- Found CMSIS: /home/azisi/Desktop/stm32-cmake/examples/fetch-cube/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Include;/home/azisi/Desktop/stm32-cmake/examples/fetch-cube/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Include (found version "5.6.0") found components: STM32H743VI 
-- Configuring done
CMake Error at CMakeLists.txt:14 (add_executable):
  Target "stm32-fetch-h7" links to target "STM32::H7" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:14 (add_executable):
  Target "stm32-fetch-h7" links to target "STM32::H743xx" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Configuring incomplete, errors occurred!
See also "/home/azisi/Desktop/stm32-cmake/tests/hal/build/CMakeFiles/CMakeOutput.log".
See also "/home/azisi/Desktop/stm32-cmake/tests/hal/build/CMakeFiles/CMakeError.log".
```

**Sources to reproduce**
```'CMakeLists.txt'
cmake_minimum_required(VERSION 3.16)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/../../cmake/stm32_gcc.cmake)

project(stm32-fetch-cube C ASM)

stm32_fetch_cube(H7)

find_package(CMSIS COMPONENTS STM32H743VI REQUIRED)

set(PROJECT_SOURCES
    main.c
)

add_executable(stm32-fetch-h7 ${PROJECT_SOURCES})
target_link_libraries(stm32-fetch-h7 CMSIS::STM32::H743VI STM32::NoSys)
stm32_print_size_of_target(stm32-fetch-h7)
```

**Steps to reproduce**
1. mkdir build
2. cd build
3. cmake ..

**Expected behavior**
I expect to build  by using make as step 4.

**Screenshots**
-

**Environment (please complete the following information):**
 - OS: kubuntu 20.04
 - Compiler:  arm-none-eabi-gcc 9.2.1
 - stm32-cmake: e9c22b93fed1f102fbe02a3ccd73737846fc9c99
 - cmake: 3.21
 - HAL/cube/CMSIS: stm32cubeH7 v1.9.0

**Additional context**
-

atsju, 2021-07-28:
@zisi
Please have a look at [readme](https://github.com/ObKo/stm32-cmake/blob/master/README.md) and [mutlicore examples](https://github.com/ObKo/stm32-cmake/tree/master/examples/multi-core). As H7 family contains dual core devices you need to specify the core.

This code is working:
```
stm32_fetch_cube(H7)

find_package(CMSIS COMPONENTS STM32H743VI_M7 REQUIRED)

set(PROJECT_SOURCES
    main.c
)

add_executable(stm32-fetch-h7 ${PROJECT_SOURCES})
target_link_libraries(stm32-fetch-h7 CMSIS::STM32::H743VI::M7 STM32::NoSys)
stm32_print_size_of_target(stm32-fetch-h7)
```

@Hish15  We need to specify the core even on mono-core devices of the H7 family. Is this expected or should we handle it ?
If we handle it I suppose we should handle the `_M7` variant also for mono-core devices? note it would be different of other pure mono-core families...

zisi, 2021-07-29:
Thanks! Also i found a project, https://github.com/Hish15/DustSucker that helps me. Maybe in README could be added some projects that used stm32-cmake.

atsju, 2021-07-29:
We don't really know projects using stm32-cmake but there are already some examples that should be enough to start with.
I plan to add some more/better H7 example as your question comes regularly.