ObKo/stm32-cmake #251: Could no set path of compiler

zisi, 2021-07-28:
**Describe the bug**
I set the compiler by using:

```
set(STM32_TARGET_TRIPLET "arm-none-eabi")
set(STM32_TOOLCHAIN_PATH "${CMAKE_CURRENT_SOURCE_DIR}/tools/gcc-arm-none-eabi-latest")
``` 
but the stm32-cmake uses the usr path to find compiler. Finally i change the stm32_gcc.cmake

```
find_program(CMAKE_C_COMPILER NAMES ${STM32_TARGET_TRIPLET}-gcc PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_CXX_COMPILER NAMES ${STM32_TARGET_TRIPLET}-g++ PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
find_program(CMAKE_ASM_COMPILER NAMES ${STM32_TARGET_TRIPLET}-gcc PATHS ${TOOLCHAIN_BIN_PATH} NO_DEFAULT_PATH)
``` 
in order do not search in default path (usr).

**Sources to reproduce**
-

**Steps to reproduce**
-

**Expected behavior**
-

**Screenshots**
-

**Environment (please complete the following information):**
 - OS: kubuntu 20.004
 - Compiler: -
 - stm32-cmake: e9c22b93fed1f102fbe02a3ccd73737846fc9c99
 - cmake: 3.21
 - HAL/cube/CMSIS: -
 - 
**Additional context**
-

atsju, 2021-07-28:
Thank you for reporting the issue. 
This might be a side effect of #233 
Unfortunately I'm not 100% understanding the effect of `NO_DEFAULT_PATH`.
If you understood the root cause, could you please elaborate a bit more ?

zisi, 2021-07-29:
As i understand by using `NO_DEFAULT_PATH` the searching path is restricted to `${TOOLCHAIN_BIN_PATH}`.

Reference: https://cmake.org/cmake/help/latest/command/find_program.html

atsju, 2021-07-29:
> As i understand by using `NO_DEFAULT_PATH` the searching path is restricted to `${TOOLCHAIN_BIN_PATH}`.

What exactly is the problem with this for you ? In the order of evaluation, `HINTS` is in position 4. What makes you find a wrong gcc (if this is your issue) ?
I mean if we use `NO_DEFAULT_PATH` then other users may complain because they want to pass the toolchain path by other means than hints.

zisi, 2021-07-29:
When i use the cmake module without `NO_DEFAULT_PATH` and by defining the  `set(STM32_TOOLCHAIN_PATH "${CMAKE_CURRENT_SOURCE_DIR}/tools/gcc-arm-none-eabi-latest")` the cmake module search in /usr path and get the system compiler and not that form `STM32_TOOLCHAIN_PATH`.

atsju, 2021-07-29:
Why not remove the arm-none-eabi-gcc installation from your path ? If it is installed I suppose rule 3 will take precedence and `find_program` will take usr path before `HINT`. 

- If you need it in your path then it's because you are using in for an other use and then you can understand that cmake wants to use path variable like this other program
- If you don't need it in your path then remove it and problem solved

This answer is voluntarily quite abrupt but I need to understand your use case as everybody has not the same use case. I don't want to be rude :)

robamu, 2021-08-01:
I read through this quickly but maybe I can add something to the discussion as well. I added this feature so the build system can find `arm-none-eabi-gcc` in the system path, which is really convenient for setting up an environment only once. But 

If this is a needed feature, an option would be to provide the CMake option `STM32_TOOLCHAIN_NO_DEFAULT_PATH`
and then supply `NO_DEFAULT_PATH` to the `find_program` accordingly. This would fix the issue that the user has to edit the `stm32-cmake` files because this option can then be set from an upper `CMakeLists.txt`.

I would set the default option to OFF because I would guess that most users add the path to the binaries to the path temporarily or permanently for convenience reasons. What do you think?

robamu, 2021-08-10:
I added this is #265 . @zisi could you check whether this works for you by setting `STM32_TOOLCHAIN_NO_DEFAULT_PATH` to `ON` from you CMakeLists.txt?

atsju, 2021-08-10:
@zisi Please try  `set(STM32_TOOLCHAIN_PATH "${CMAKE_CURRENT_SOURCE_DIR}/tools/gcc-arm-none-eabi-latest" CACHE)`.

Notice the `CACHE`. 

As you are the only one to reproduce the issue we will need your support before doing any changes.

zisi, 2021-08-10:
Hey! I got 
```
CMake Error at CMakeLists.txt:7 (set):
  set given invalid arguments for CACHE mode.
```

atsju, 2021-08-10:
`set(STM32_TOOLCHAIN_PATH "${CMAKE_CURRENT_SOURCE_DIR}/tools/gcc-arm-none-eabi-latest" CACHE PATH)`

zisi, 2021-08-10:
The same error.

atsju, 2021-08-10:
Sorry. This one should work : 
`set(STM32_TOOLCHAIN_PATH "${CMAKE_CURRENT_SOURCE_DIR}/tools/gcc-arm-none-eabi-latest" CACHE PATH "description")`

Please let me know for a better understandig of the problem even if `HINTS` seems to be the correct solution

zisi, 2021-08-10:
Neither now it works. The problem is, when a system has 2 toolchains, a project one in `STM32_TOOLCHAIN_PATH` and the system one in /usr/bin, by setting inside the project the `STM32_TOOLCHAIN_PATH` it always the project uses the /usr/bin the system one instead of the `STM32_TOOLCHAIN_PATH`. But if i remove the system toolchain the project uses the correct one that is setting in `STM32_TOOLCHAIN_PATH` .

atsju, 2021-08-10:
Strange. Reading https://cmake.org/cmake/help/latest/command/find_program.html In your case I think it's rule 5 that finds the program.
Without CACHE, rule 6 is used but with CACHE I expected rule 2.
Thank you for feedback

PR #266 (2021-08-16): Using HINTS for STM Toolchain Path
This causes toolchains found via the STM32_TOOLCHAIN_PATH variable
to take precedence over a toolchain in the system path.
Also updated README accordingly

Fixes #251  . Closes #265
closed 2021-08-16 by Hish15