ObKo/stm32-cmake #252: Custom linker script with CMSIS

fnadeau, 2021-07-29:
**Describe the bug**
Use of stm32_add_linker_script() when CMSIS is used generate an invalid linker command

**Sources to reproduce**
from master (e9c22b93)

To keep it simple, I patched the blinky example so that it:
- use a custom linker file (see examples/custom-linker-script)
- fetch HAL and CMSIS

**Steps to reproduce**
```
cp examples/custom-linker-script/F407VG.ld examples/blinky/
patch -p1 << EOF
diff --git a/examples/blinky/CMakeLists.txt b/examples/blinky/CMakeLists.txt
index 51191d3..28c7436 100644
--- a/examples/blinky/CMakeLists.txt
+++ b/examples/blinky/CMakeLists.txt
@@ -3,7 +3,8 @@ set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/../../cmake/stm32_gcc.cmake
 
 project(stm32-blinky C ASM)
 set(CMAKE_INCLUDE_CURRENT_DIR TRUE)
-
+stm32_fetch_cmsis(F4 F1 L0)
+stm32_fetch_hal(F4 F1 L0)
 # Configure here which STM32 target(s) to build
 option(BLINKY_F4_EXAMPLE "Compile F4 example" ON)
 option(BLINKY_F1_EXAMPLE "Compile F1 example" OFF)
@@ -49,6 +50,7 @@ if(BLINKY_F4_EXAMPLE)
         CMSIS::STM32::F407VG
         STM32::NoSys
     )
+    stm32_add_linker_script(stm32-blinky-f4 PRIVATE F407VG.ld)
     stm32_print_size_of_target(stm32-blinky-f4)
 endif()
EOF
cmake -S examples/blinky -B build -G Ninja
cmake --build build -v
```

**Expected behavior**
linker should display an error

**Screenshots**
```
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld:/home/fnadeau/git/stm32-cmake/build/F407VG.ld:9: warning: redeclaration of memory region `FLASH'
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld:/home/fnadeau/git/stm32-cmake/build/F407VG.ld:10: warning: redeclaration of memory region `RAM'
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld:/home/fnadeau/git/stm32-cmake/build/F407VG.ld:11: warning: redeclaration of memory region `CCMRAM'
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: warning: /home/fnadeau/git/stm32-cmake/build/F407VG.ld contains output sections; did you forget -T?
```

Linker command that seems to be the issue:
```
/usr/bin/arm-none-eabi-gcc  -T /home/fnadeau/git/stm32-cmake/examples/blinky/F407VG.ld --sysroot=\"/usr/arm-none-eabi\" -mthumb -Wl,--gc-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard /home/fnadeau/git/stm32-cmake/build/F407VG.ld --specs=nosys.specs CMakeFiles/stm32-blinky-f4.dir/blinky.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-hal-f4-src/Src/stm32f4xx_hal_rcc.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-hal-f4-src/Src/stm32f4xx_hal.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-cmsis-f4-src/Source/Templates/system_stm32f4xx.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-hal-f4-src/Src/stm32f4xx_hal_gpio.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-hal-f4-src/Src/stm32f4xx_hal_cortex.c.obj CMakeFiles/stm32-blinky-f4.dir/_deps/stm32-cmsis-f4-src/Source/Templates/gcc/startup_stm32f407xx.s.obj -o stm32-blinky-f4.elf
```

We can see that there is two linker file in it:
- /home/fnadeau/git/stm32-cmake/examples/blinky/F407VG.ld
- /home/fnadeau/git/stm32-cmake/build/F407VG.ld

The first is the copy we made from examples/custom-linker-script and the second is the one generated by stm32-cmake.

**Environment (please complete the following information):**
 - OS: Linux
 - Compiler: arm-none-eabi-gcc 11.1.0
 - stm32-cmake: e9c22b93
 - cmake: 3.21.1
 - HAL/cube/CMSIS: fetched by stm32-cmake

**Additional context**
How can we used CMSIS with custom linker file. Custom linker file is usually required whenever you have a bootloader. (start address is not 0x8000000)

atsju, 2021-07-29:
Thank you for reporting this issue.
This is linked to #237 and I agree it would be better to find a way to warn the user at cmake generation stage and way before link stage. 
However, please have a look at [readme](https://github.com/ObKo/stm32-cmake/blob/master/README.md)

> CMSIS creates the following targets:
> 
> CMSIS::STM32::<FAMILY> (e.g. CMSIS::STM32::F4) - common includes, compiler flags and defines for family
> CMSIS::STM32::<TYPE> (e.g. CMSIS::STM32::F407xx) - common startup source for device type, depends on CMSIS::STM32::<FAMILY>
> CMSIS::STM32::<DEVICE> (e.g. CMSIS::STM32::F407VG) - linker script for device, depends on CMSIS::STM32::<TYPE>

you should use `CMSIS::STM32::F407xx` and this will fix the issue.

fnadeau, 2021-07-29:
Thanks for the quick reply. It does indeed fix my issue.

closed 2021-07-29 by fnadeau