ObKo/stm32-cmake #256: Multiple definitions linking error

zisi, 2021-08-03:
**Describe the bug**
When the `stm32_fetch_cube() ` is used it gets also templates like `system_stm32h7xx.c`.
In case that a modified `system_stm32h7xx.c` is used, an error in linking is appeared related
to multiple definition of functions inside the `system_stm32h7xx.c`.

**Sources to reproduce**
https://gitlab.com/librespacefoundation/satnogs-comms/libsatnogs-comms/-/merge_requests/1/diffs
The project it is under development.

**Steps to reproduce**
1. Under test/CMakeLists.txt in `add_executable(test ...` add `system_stm32h7xx.c`
2. mkdir build
3. cd build
4. cmake ..
5. make

**Expected behavior**

**Screenshots**
-

**Environment (please complete the following information):**
 - OS: kubuntu 20.04
 - Compiler: arm-none-eabi-gcc 9.3.1
 - stm32-cmake: 70a5a0a07f834972c4c69b58dc69b2f824257f23
 - cmake: 3.21.0
 - HAL/cube/CMSIS: stm32cubeH7 v1.9.0

**Additional context**
In order to fix the issue, the below code is used:
```
# Do not use HAL template
set(CMSIS_TEMPLATE "${PROJECT_SOURCE_DIR}/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c")
set(CMSIS_TEMPLATE_PATH "${PROJECT_SOURCE_DIR}/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates")
file(REMOVE ${CMSIS_TEMPLATE})
file(COPY "${PROJECT_SOURCE_DIR}/test/system_stm32h7xx.c"
    DESTINATION ${CMSIS_TEMPLATE_PATH}
)
```
Does it exists another way to not include templates?

atsju, 2021-08-03:
Thank you for reporting. I think you use wrong target.
When you wish to replace some template files you must update target.

From [readme](https://github.com/ObKo/stm32-cmake/blob/master/README.md): 
> CMSIS creates the following targets:
> 
> CMSIS::STM32::<FAMILY> (e.g. CMSIS::STM32::F4) - common includes, compiler flags and defines for family
> CMSIS::STM32::<TYPE> (e.g. CMSIS::STM32::F407xx) - common startup source for device type, depends on CMSIS::STM32::<FAMILY>
> CMSIS::STM32::<DEVICE> (e.g. CMSIS::STM32::F407VG) - linker script for device, depends on CMSIS::STM32::<TYPE>

[image]

Please have a look at #237 and #252 that have same issues with `.ld` files.

closed 2021-08-03 by atsju

zisi, 2021-08-17:
Hey, i use the `CMSIS::STM32::H7::M7` and this commit https://github.com/ObKo/stm32-cmake/commit/1dd8b736734ebe674624ac8e8743f128bae17c66. I add to the build system  `system_stm32h7xx.c` and `startup_stm32h743vitx.s` and i got errors like `error: 'GPIOD' undeclared (first use in this function)`. I can't understand how by using the CMSIS family the building system does not include the CMSIS template like `system_stm32h7xx.c`.

atsju, 2021-08-18:
Thank you for reporting. 
I do not see the question `?` Please clarify the question you have. 

I have seen your history and you are not a noob. You can understand we maintain this repo in our free time (several hours a week). It would realy help if you could do a quality questions or clearly documented bugs. If needed please provide a code

- Github hosted  
- minimalist (ideally it would be based on our examples)
- reproducing the issue

This would really help us to help you. Personally I won't look any deeper for now.

zisi, 2021-09-08:
Hey, sorry for the delay. I prepare an example to reproduce the error. This is the example, https://github.com/zisi/stm32-cmake/tree/linking-error/examples/link-error. To reproduce it, run:

1. mkdir build
2. cd build
3. cmake ..
4. make

The output:
```
[ 92%] Linking C executable stm32-linker-error.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/gcc/startup_stm32h743xx.s.obj:(.isr_vector+0x0): multiple definition of `g_pfnVectors'; CMakeFiles/stm32-linker-error.dir/startup_stm32h743vitx.s.obj:(.isr_vector+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/gcc/startup_stm32h743xx.s.obj: in function `Default_Handler':
(.text.Default_Handler+0x0): multiple definition of `Default_Handler'; CMakeFiles/stm32-linker-error.dir/startup_stm32h743vitx.s.obj:(.text.Default_Handler+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c.obj:(.data.SystemCoreClock+0x0): multiple definition of `SystemCoreClock'; CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj:(.data.SystemCoreClock+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c.obj:(.data.SystemD2Clock+0x0): multiple definition of `SystemD2Clock'; CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj:(.data.SystemD2Clock+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c.obj:(.rodata.D1CorePrescTable+0x0): multiple definition of `D1CorePrescTable'; CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj:(.rodata.D1CorePrescTable+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c.obj: in function `SystemInit':
system_stm32h7xx.c:(.text.SystemInit+0x0): multiple definition of `SystemInit'; CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj:system_stm32h7xx.c:(.text.SystemInit+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/stm32-linker-error.dir/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c.obj: in function `SystemCoreClockUpdate':
system_stm32h7xx.c:(.text.SystemCoreClockUpdate+0x0): multiple definition of `SystemCoreClockUpdate'; CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj:system_stm32h7xx.c:(.text.SystemCoreClockUpdate+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/stm32-linker-error.dir/build.make:272: stm32-linker-error.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/stm32-linker-error.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```
System:

- OS: kubuntu 20.04
- Compiler: arm-none-eabi-gcc 9.2.1

**Expected behavior**
I expect to use the system_stm32h7xx.c and startup_stm32h743vitx.s that are placed inside the project and not the templates which is provided by the stm32_hal.

Thanks!!

atsju, 2021-09-08:
Hello,
as already stated 
```
CMSIS creates the following targets:

CMSIS::STM32:: (e.g. CMSIS::STM32::F4) - common includes, compiler flags and defines for family
CMSIS::STM32:: (e.g. CMSIS::STM32::F407xx) - common startup source for device type, depends on CMSIS::STM32::
CMSIS::STM32:: (e.g. CMSIS::STM32::F407VG) - linker script for device, depends on CMSIS::STM32::
```
So in your example it's normal that with `CMSIS::STM32::H743xx::M7` you use template system and startup files.

Please add a commit to reproduce what you described here https://github.com/ObKo/stm32-cmake/issues/256#issuecomment-900231796

zisi, 2021-09-08:
Hey, add this commit, https://github.com/zisi/stm32-cmake/tree/linking-error and got this:

```
[  8%] Building C object CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj
In file included from /home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:48:
/home/azisi/Documents/stm32-cmake/examples/link-error/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7xx.h:164:3: error: #error "Please select first the target STM32H7xx device used in your application (in stm32h7xx.h file)"
  164 |  #error "Please select first the target STM32H7xx device used in your application (in stm32h7xx.h file)"
      |   ^~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:113:3: error: unknown type name 'uint32_t'; did you mean 'wint_t'?
  113 |   uint32_t SystemCoreClock = 64000000;
      |   ^~~~~~~~
      |   wint_t
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:114:3: error: unknown type name 'uint32_t'; did you mean 'wint_t'?
  114 |   uint32_t SystemD2Clock = 64000000;
      |   ^~~~~~~~
      |   wint_t
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:115:10: error: unknown type name 'uint8_t'
  115 |   const  uint8_t D1CorePrescTable[16] = {0, 0, 0, 0, 1, 2, 3, 4, 1, 2, 3, 4, 6, 7, 8, 9};
      |          ^~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c: In function 'SystemInit':
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:153:6: error: 'FLASH_LATENCY_DEFAULT' undeclared (first use in this function)
  153 |   if(FLASH_LATENCY_DEFAULT  > (READ_BIT((FLASH->ACR), FLASH_ACR_LATENCY)))
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:153:6: note: each undeclared identifier is reported only once for each function it appears in
In file included from /home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:48:
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:153:42: error: 'FLASH' undeclared (first use in this function)
  153 |   if(FLASH_LATENCY_DEFAULT  > (READ_BIT((FLASH->ACR), FLASH_ACR_LATENCY)))
      |                                          ^~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:153:55: error: 'FLASH_ACR_LATENCY' undeclared (first use in this function)
  153 |   if(FLASH_LATENCY_DEFAULT  > (READ_BIT((FLASH->ACR), FLASH_ACR_LATENCY)))
      |                                                       ^~~~~~~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:156:45: error: 'uint32_t' undeclared (first use in this function); did you mean 'wint_t'?
  156 |  MODIFY_REG(FLASH->ACR, FLASH_ACR_LATENCY, (uint32_t)(FLASH_LATENCY_DEFAULT));
      |                                             ^~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:160:3: error: 'RCC' undeclared (first use in this function)
  160 |   RCC->CR |= RCC_CR_HSION;
      |   ^~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:160:14: error: 'RCC_CR_HSION' undeclared (first use in this function)
  160 |   RCC->CR |= RCC_CR_HSION;
      |              ^~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:261:3: error: 'FMC_Bank1_R' undeclared (first use in this function)
  261 |   FMC_Bank1_R->BTCR[0] = 0x000030D2;
      |   ^~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:267:3: error: 'SCB' undeclared (first use in this function)
  267 |   SCB->VTOR = FLASH_BANK1_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
      |   ^~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:267:15: error: 'FLASH_BANK1_BASE' undeclared (first use in this function)
  267 |   SCB->VTOR = FLASH_BANK1_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
      |               ^~~~~~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c: In function 'SystemCoreClockUpdate':
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:313:3: error: unknown type name 'uint32_t'; did you mean 'wint_t'?
  313 |   uint32_t pllp, pllsource, pllm, pllfracen, hsivalue, tmp;
      |   ^~~~~~~~
      |   wint_t
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:314:3: error: unknown type name 'uint32_t'; did you mean 'wint_t'?
  314 |   uint32_t common_system_clock;
      |   ^~~~~~~~
      |   wint_t
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:320:11: error: 'RCC' undeclared (first use in this function)
  320 |   switch (RCC->CFGR & RCC_CFGR_SWS)
      |           ^~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:320:23: error: 'RCC_CFGR_SWS' undeclared (first use in this function)
  320 |   switch (RCC->CFGR & RCC_CFGR_SWS)
      |                       ^~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:322:8: error: 'RCC_CFGR_SWS_HSI' undeclared (first use in this function)
  322 |   case RCC_CFGR_SWS_HSI:  /* HSI used as system clock source */
      |        ^~~~~~~~~~~~~~~~
/home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:323:28: error: 'uint32_t' undeclared (first use in this function); did you mean 'wint_t'?
[... cut ...]
  398 |   SystemD2Clock = (common_system_clock >> ((D1CorePrescTable[(RCC->CDCFGR1 & RCC_CDCFGR1_HPRE)>> RCC_CDCFGR1_HPRE_Pos]) & 0x1FU));
      |                                                                                                  ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/stm32-linker-error.dir/build.make:90: CMakeFiles/stm32-linker-error.dir/system_stm32h7xx.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/stm32-linker-error.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Thanks!

atsju, 2021-09-08:
I have no time to look a this now but have you tried to understand the problem ? 

```
In file included from /home/azisi/Documents/stm32-cmake/examples/link-error/system_stm32h7xx.c:48:
/home/azisi/Documents/stm32-cmake/examples/link-error/build/_deps/stm32cubeh7-src/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7xx.h:164:3: error: #error "Please select first the target STM32H7xx device used in your application (in stm32h7xx.h file)"
  164 |  #error "Please select first the target STM32H7xx device used in your application (in stm32h7xx.h file)"
```

first error seems to be pretty clear. You want to use your own files it's normal that you have more defines and other things to provide yourself. Please open the file and you will see that your are obviously missing a simple and mandatory define.

zisi, 2021-09-09:
Thanks for the comment, i add this definition:
 ```
 add_definitions(-DSTM32H743xx)
```
add it works.