ObKo/stm32-cmake #258: Consider system_stm32xxx.c a "device type-specific" rather than a "familly type-specific"

fnadeau, 2021-08-06:
**Is your feature request related to a problem? Please describe.**
Building application that do not have interrupt vector at offset 0x00 (usually application that require bootloader)

**Describe the solution you'd like**
Documentation to read like this:

> CMSIS consists of three main components:
>
> - Family-specific headers, e.g. stm32f4xx.h
> - Device type-specific startup sources (e.g. startup_stm32f407xx.s **& system_stm32f4xx.c**)
> - Device-specific linker scripts which requires information about memory sizes

As well as system_stm32f4xx.c not be part of compilation when using CMSIS::STM32::F4.

**Describe alternatives you've considered**
Documentation to read like this:

> CMSIS consists of three main components:
>
> - Family-specific headers, e.g. stm32f4xx.h **and system template file (e.g. system_stm32f4xx.c)**
> - Device type-specific startup sources (e.g. startup_stm32f407xx.s)
> - Device-specific linker scripts which requires information about memory sizes

**Additional context**
See this [pull-request](https://github.com/STMicroelectronics/STM32CubeF4/issues/65)
There is currently no way to override the vector table location at build time. The only way to do so is to modify the system_stm32xxx.c file, which when using CMSIS::STM32::F4 is automatically included.

If described solution is deemed acceptable, I'll open a proper PR with the following diff:
```
diff --git a/FindCMSIS.cmake b/FindCMSIS.cmake
index 3821538..62be939 100644
--- a/FindCMSIS.cmake
+++ b/FindCMSIS.cmake
@@ -117,17 +117,6 @@ foreach(COMP ${CMSIS_FIND_COMPONENTS})
     
     set(CMSIS_${COMP}_VERSION ${CMSIS_${FAMILY}${CORE_U}_VERSION})
     set(CMSIS_VERSION ${CMSIS_${COMP}_VERSION})
-        
-    find_file(CMSIS_${FAMILY}${CORE_U}_SOURCE
-        NAMES system_stm32${FAMILY_L}xx.c
-        PATHS "${CMSIS_${FAMILY}${CORE_U}_PATH}/Source/Templates"
-        NO_DEFAULT_PATH
-    )
-    list(APPEND CMSIS_SOURCES "${CMSIS_${FAMILY}${CORE_U}_SOURCE}")
-    
-    if (NOT CMSIS_${FAMILY}${CORE_U}_SOURCE)
-        continue()
-    endif()
 
     if(NOT (TARGET CMSIS::STM32::${FAMILY}${CORE_C}))
         add_library(CMSIS::STM32::${FAMILY}${CORE_C} INTERFACE IMPORTED)
@@ -158,6 +147,17 @@ foreach(COMP ${CMSIS_FIND_COMPONENTS})
             set(DEVICES_FOUND FALSE)
             break()
         endif()
+
+        find_file(CMSIS_${FAMILY}${CORE_U}_SYSTEM
+            NAMES system_stm32${FAMILY_L}xx.c
+            PATHS "${CMSIS_${FAMILY}${CORE_U}_PATH}/Source/Templates"
+            NO_DEFAULT_PATH
+        )
+        list(APPEND CMSIS_SOURCES "${CMSIS_${FAMILY}${CORE_U}_SYSTEM}")
+        if (NOT CMSIS_${FAMILY}${CORE_U}_SYSTEM)
+            set(DEVICES_FOUND FALSE)
+            break()
+        endif()
         
         if(NOT (TARGET CMSIS::STM32::${TYPE}${CORE_C}))
             add_library(CMSIS::STM32::${TYPE}${CORE_C} INTERFACE IMPORTED)
```
This solution seems to allow override of template system_stm32xxx.c in the same manner than startup_stm32xxx.s is overwritten.

atsju, 2021-08-07:
Thank you for reporting this issue. [STM32CubeF4#65](https://github.com/STMicroelectronics/STM32CubeF4/issues/65) explains the root well. ST has not been very efficient in this case...
However I feel like this is more an ST issue and not a tool issue. `system_stm32f4xx.c` is not meant to be modified by user and ST should fix this so that it stays that way.

Your fix proposal is not correct. I suppose you copied it from how startup files are found but if you look closely this is done in a subloop. Your `break` statement would really break findCMSIS :wink:

As a first approach I would just recommend you not using target `CMSIS::STM32::F4`. I mean this target already only searches for 1 path and 1 file so modifying it to only search for a path makes not much sense. So maybe it would be better to not depend on this target and simply add the needed files and path from CMSIS plus your own file wherever they are located in your own pure cmake target.

fnadeau, 2021-08-09:
It does seems like something ST should fix, given that their release cycle is what it is, I have little hope for that to happen in a near future.
 - given that there has been two releases of STM32CubeF4 since the bug was opened
 - given that not only do they need to fix it in STM32CubeF4 but all others as well and need a coherent approach to the problem

system_stm32f4xx.c is a template, so is startup_stm32f4xxxx.s, and template are meant to be modified.

PR #136 and #137 already covers something similar, maybe better as they search local path before including template. However, they don't apply to current top of master branch.

Given patch has not been tested on all example(works in one scenario), so yes it most definitively will break some cases, that what PR and unit test are for ;) It is more or less a copy of the find loop for startup_stm32xxx.s, which has the same break, wouldn't that break has the same effect has what you describe? My idea was that if startup_stm32xxx.s is only include with CMSIS::STM32::F407VG, same logic could be use for system_stm32xxx.c.

Alternative approach would imply:
 - knowing the path of STM32_CMSIS path (that is what makes findCMSIS usefull)
 - adding the following:
```
        target_include_directories(${TARGET} PRIVATE STM32CubeF4/Drivers/CMSIS/Core/Include/)
        target_include_directories(${TARGET} PRIVATE STM32CubeF4/Drivers/STM32G0xx_HAL_Driver/Inc)
```
is that correct? (path needs adjustment)

Given that this issue will likely be closed, would you consider an updated PR for #136  and/or #137 ?

atsju, 2021-08-09:
Thank you for your feedback.

> system_stm32f4xx.c is a template, so is startup_stm32f4xxxx.s, and template are meant to be modified.

Yes correct. Although system_stm32f4xx.c had nothing provided for modification until last releases.

> It is more or less a copy of the find loop for startup_stm32xxx.s, which has the same break, wouldn't that break has the same effect has what you describe?

No. `Startup` is per device and `system` is per family. `Break` would break inner VS outer loop. So it has not the same effect.

About alternative solution:
```cmake
target_include_directories(${TARGET} PRIVATE STM32CubeF4/Drivers/CMSIS/Core/Include/)
```
is the only one related to `findCMSIS`. The other line is related to `findHAL`.

> Given that this issue will likely be closed, would you consider an updated PR for #136 and/or #137 ?

I don't think it would be acceptable in current state because it masks automatic bahaviors to users which change depending on file names and places

**My recommandation**
As a short term solution for you I would remove use of `CMSIS::STM32::F4` and search for files and include path manually.
Why I think stm32-cmake may not need a change is that if `CMSIS::STM32::F4` doesn't search for `system_stm32f4xx.c` and `startup_stm32f4xxxx.s` then it will only add the path `Drivers/CMSIS/Core/Include/` and in this case this target becomes more or less useless and we fall back to my initial "short term solution".

> Given that this issue will likely be closed

We will discuss it internally and close with a conclusion only once we agree with @Hish15 (then we will most likely close #136 and #137 too). Or if you close because you think my proposal is beast approach (for all users, not only for your use case seeing my solution as a workaround).

fnadeau, 2021-08-09:
I tried you're suggestion: it implies that HAL and BSP cannot be used. Both tries to link to CMSIS::STM32::${FAMILY}${CORE_C}. So if in the main project I have find_package(HAL COMPONENTS STM32xxxx),  I also need find_package(CMSIS COMPONENTS STM32xxxx) which in turn tries to build the system_stm32xxx.c template file.

Did I miss something from your suggestion?

atsju, 2021-08-09:
> Did I miss something from your suggestion?

I will need to test on my side. I didn't know HAL and BSP need CMSIS so maybe **I** missed something. If this is the case, a modification may be welcome but we will need to have a look on what would be the required behavior before doing the modification.

Please note this will take some time as we have only limited personal time to threat all PR and issues.

atsju, 2021-08-09:
please have a look to #264 and made any feedback :)

STM32::F4 does more than just including 1 path. is searches for include path and some pure cmsis necessary files.

And now I agree, no template files should be added in STM32::F4. This is the job for STM32::F407xx (like explained in readme).
If you are using STM32::F4 you will need to provide startup an system files yourself!

fnadeau, 2021-08-09:
I like it. Quick test on my setup and #264 works like a charm.

Definitively a better solution. Thanks for looking into this.

PR #264 (2021-08-15): target CMSIS::<family> will not include any template files
fix #258
closed 2021-08-15 by atsju